Freetz/freetz #1: Labor-Support for 3370, 3390 and others.

Himan2001, 2016-02-19:
It is possible to integrate the latest "compile-testing" Labor-Support for the 3370/3390 and other models right now ? - It can be very helpful to fix bugs with the latest versions for this models right now before a final version comes out. The VDSL2-Vectoring Support for this models is much better as from the old 6.30er Final Versions.

olistudent, 2016-02-20:
Hi. We accept patches otherwise you will have to wait until a developer adds support. As there are many labor versions available at the moment and they do change very often it takes much time to support them all...

er13, 2017-08-08:
We have no plans to support labor firmwares for boxes other than the flagship ones (as of now 7490, 7590, 7580).

closed 2017-08-08 by er13