Freetz/freetz #2: Unresolved symbols for 7490 since edcff8854c1658843ba54c10e38734befe2544bd

tkurbad, 2016-07-08:
Hi,

since edcff8854c1658843ba54c10e38734befe2544bd I'm seeing unresolved symbols for kernel modules:

unresolved symbol avm_pa_add_rtp_session in file /lib/modules/3.10.73/kernel/drivers/rtp/krtp.ko
unresolved symbol avm_pa_flush_rtp_session in file /lib/modules/3.10.73/kernel/drivers/rtp/krtp.ko
unresolved symbol yield_trigger_linux_ipi in file /lib/modules/3.10.73/kernel/drivers/rtp/krtp.ko
unresolved symbol simplemempool_alloc in file /lib/modules/3.10.73/kernel/drivers/isdn/avm_dect/avm_dect.ko
unresolved symbol simplemempool_alloc_exit in file /lib/modules/3.10.73/kernel/drivers/isdn/avm_dect/avm_dect.ko
unresolved symbol __simplemempool_alloc_init in file /lib/modules/3.10.73/kernel/drivers/isdn/avm_dect/avm_dect.ko
unresolved symbol simplemempool_free in file /lib/modules/3.10.73/kernel/drivers/isdn/avm_dect/avm_dect.ko
unresolved symbol yield_trigger_linux_ipi in file /lib/modules/3.10.73/kernel/drivers/isdn/avm_dect/avm_dect.ko
unresolved symbol simplemempool_alloc in file /lib/modules/3.10.73/kernel/drivers/isdn/isdn_fon5/zzz/isdn_fbox_fon5.ko
unresolved symbol simplemempool_alloc_exit in file /lib/modules/3.10.73/kernel/drivers/isdn/isdn_fon5/zzz/isdn_fbox_fon5.ko
unresolved symbol __simplemempool_alloc_init in file /lib/modules/3.10.73/kernel/drivers/isdn/isdn_fon5/zzz/isdn_fbox_fon5.ko
unresolved symbol simplemempool_free in file /lib/modules/3.10.73/kernel/drivers/isdn/isdn_fon5/zzz/isdn_fbox_fon5.ko
unresolved symbol ifx_mps_ack in file /lib/modules/3.10.73/kernel/drivers/char/ubik2/pcmlink.ko
unresolved symbol ifx_mps_en in file /lib/modules/3.10.73/kernel/drivers/char/ubik2/pcmlink.ko
unresolved symbol ifx_mps_trigger in file /lib/modules/3.10.73/kernel/drivers/char/ubik2/pcmlink.ko
unresolved symbol simplemempool_alloc in file /lib/modules/3.10.73/kernel/drivers/char/ubik2/pcmlink.ko
unresolved symbol simplemempool_alloc_exit in file /lib/modules/3.10.73/kernel/drivers/char/ubik2/pcmlink.ko
unresolved symbol __simplemempool_alloc_init in file /lib/modules/3.10.73/kernel/drivers/char/ubik2/pcmlink.ko
unresolved symbol simplemempool_free in file /lib/modules/3.10.73/kernel/drivers/char/ubik2/pcmlink.ko
unresolved symbol yield_trigger_linux_ipi in file /lib/modules/3.10.73/kernel/drivers/char/ubik2/pcmlink.ko
unresolved symbol simplemempool_alloc in file /lib/modules/3.10.73/kernel/drivers/isdn/capi_codec/capi_codec.ko
unresolved symbol simplemempool_alloc_exit in file /lib/modules/3.10.73/kernel/drivers/isdn/capi_codec/capi_codec.ko
unresolved symbol __simplemempool_alloc_init in file /lib/modules/3.10.73/kernel/drivers/isdn/capi_codec/capi_codec.ko
unresolved symbol simplemempool_free in file /lib/modules/3.10.73/kernel/drivers/isdn/capi_codec/capi_codec.ko
unresolved symbol yield_trigger_linux_ipi in file /lib/modules/3.10.73/kernel/drivers/isdn/capi_codec/capi_codec.ko
unresolved symbol yield_trigger_linux_ipi in file /lib/modules/3.10.73/kernel/drivers/net/nlaudio/ulpcmlink.ko
unresolved symbol yield_trigger_linux_ipi in file /lib/modules/3.10.73/kernel/drivers/net/nlaudio/nlaudio.ko

Perhaps, AVM's kernel config changed?

Best,
Torsten

tkurbad, 2016-07-08:
Nevermind, I just saw that this only matters if I'd replace the kernel....

closed 2016-07-08 by tkurbad

olistudent, 2016-07-08:
We do need a new open source package from AVM. It is released a few weeks after the Firmware normally.