Freetz/freetz #4: Build for 7170 fails at "applying webmenu patch"

JBBgameich, 2017-01-11:
The build for FritzBox 7170 fails, building for other boxes works. 
I hope this sort of Bug report helps, here is the error message:

[image]

`	applying patch file ./patches/scripts/190-add_freetz_webmenu.sh
applying webmenu patch
	patching build/modified/filesystem/usr/www/all/html/de/menus/menu2.html
~/freetz/tools/freetz_functions: Zeile 122:  7334 Speicherzugriffsfehler  sed -i $extended_regexp -e "$1" "$2"
ERROR: modsed failed editing build/modified/filesystem/usr/www/all/html/de/menus/menu2.html (bad sed syntax?)
Makefile:310: die Regel für Ziel „firmware-nocompile“ scheiterte
`

JBBgameich, 2017-05-13:
This is fixed now, I don't know which commit has fixed my problem, but anyway, building works as before again.

closed 2017-05-13 by JBBgameich

er13, 2017-08-08:
Speicherzugriffsfehler (German for segmentation fault) is an indication for a hardware or a software problem on your build host. It has nothing to do with Freetz itself.