Freetz/freetz #5: AVM-Forwarding broken in Fritz!OS-6.8x (7490)

tkurbad, 2017-02-14:
Hi,

I just upgraded my box to 6.80 and came to realize that my port forwarding to 443 on the box itself (placed in an earlier version via AVM-Forwarding) was gone.

After re-adding the forwarding, it is again gone from the list as soon as the box is rebooted.

I've opened the web-interface of the box via 444, so I guess AVM now deliberately closes port 443 if it's not in use (from the original firmware's perspective).

Any chances of overcoming this situation?

Thanks,
Torsten

PeterPawn, 2017-02-14:
Look at the IPPF site (README.md contains the link) and locate the thread, where exactly this question was discussed only a short time ago (and it was answered there, that means, it's worth the reading).

tkurbad, 2017-02-15:
Ok, should someone else have this problem and no time to read through a bunch of threads on the forum, here's the solution:

Port forwardings from port 443 have to go into the `voip_forwardrules = ".....",` section of /var/flash/ar7.cfg.

If I find the time, I'll try to write a patch for AVM-Forwarding that takes this into account.

Best,
Torsten

closed 2017-02-15 by tkurbad