Freetz/freetz #6: Support for 7581

ictinc, 2017-03-20:
Hi there,
Wil the fritz box 7581 be supported any time soon?
Is there anything​I can do to help speed this up?

Kind regards, ictinc

olistudent, 2017-03-20:
Hi ictinc,

support depends on the hardware (mips -> good; arm,intel,... -> bad). Then we need serial logs from the console and somebody willing to test the images. I doubt that there will be support in the near future...

As we do not get test boxes from AVM no developer is owning one of these new boxes. This makes supporting new images difficult.

Regards,
Oliver

kriegaex, 2017-03-20:
I want to mention that in the past the project got private hardware donations which enabled us to support new FB types. So if you know someone who has a leftover 7581, ...

olistudent, 2017-03-20:
Hi Alexander,

in IPPF I found postings which read like Broadcom chipset and ARM cpu. Perhaps the 7581 will be a model that is only sold in Netherlands. So I'm pretty sure Freetz will never support this box. :-(

kriegaex, 2017-03-20:
Of course, @olistudent, if there are technical reasons not to support it then a hardware donation is not useful in this case. I just made a general statement. We both know that in the years of my development inactivity in the project I pretty much lost touch with all things technical.

kounsokiri, 2017-04-09:
Hi there, i have a 7581 and i'm willing to make some serial logs if i knew how to...

kounsokiri, 2017-04-09:
I'm compiling the Trunk version now for the 7580 cause i think the 7581 is just a watered down version of the 7580...

PeterPawn, 2017-04-09:
The CPU architecture is completely different ... 7580 uses a GRX350 chipset (MIPS).

If you really want to build your own version of a 7581 firmware, you should request the open source archive for this model first (ask AVM).

er13, 2017-08-08:
7581 is an ARM based box. Up to now only MIPS based boxes are supported in Freetz. Supporting 7581 requires a lot of effort - Freetz build system, compiler/binutils/uClibc, patches to them, ARM specific patches for the packages, etc. To be honest, as of now we have no motivation to put any effort into it.

jschwartzenberg, 2017-10-12:
XS4ALL is also shipping 5490s now due to DECT issues with the newer models (according to the helpdesk). I do not have one myself, but possibly it would be easier to support the 5490.

I could imagine the ARM boxes have a better chance with OpenWRT.

er13, 2018-04-26:
No plans to support ARM boxes as of now...

closed 2018-04-26 by er13