Freetz/freetz #7: Support for 6360

Phil795, 2017-05-12:
Hi,

is it possible to use freetz on a 6360?

JBBgameich, 2017-05-12:
If your box is available in the 'make menuconfig menu', it's supported. You'll have ti choose it from the list of supported boxes whole building. If it's not in that list, it's most likely not supported.

er13, 2017-08-08:
We have no plans to support 6360 in Freetz - an [EOM and almost EOS](https://avm.de/service/eos-liste/fritzbox/) cable box with no firmware image publicly available.
Besides that, there is no known non-hackish way to flash the modified firmware.

closed 2017-08-08 by er13

programminghoch10, 2018-11-06:
This box is in EOM and EOS over a year now.
Because of AVMs certificate mistake I think a lot of people have this box lying around with old certificates and don't know what to do with it. (me) 
There is no official image, but there is an 06.51 image going around, which can be normally flashed via ADAM2 using push_firmware (I don't believe that is hackish).
I have modded the image myself to get at least telnet access to the box.
So I'm interested now in how hard it is to mod a box with existing firmware and the possibility to "ask" the box anything using telnet commands.
I would try myself, but I don't have enough knowledge and experience in programming and modding to do so.