Freetz/freetz #8: Compile fail

ds2k5, 2017-07-05:
Hello,
have a Ubuntu 16.04 LTS.

git clone https://github.com/Freetz/freetz
cd freetz
make dirclean
make clean
make menuconfig
make

The compile fails

configure: WARNING: unrecognized options: --disable-nls
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether make sets $(MAKE)... (cached) yes
checking how to print strings... printf
checking for style of include used by make... GNU
checking for gcc... gcc
checking whether the C compiler works... no
configure: error: in `/home/freetz/freetz/source/host-tools/fakeroot-1.21/build/biarch':
configure: error: C compiler cannot create executables
See `config.log' for more details
tools/make/fakeroot/fakeroot.mk:56: recipe for target '/home/freetz/freetz/source/host-tools/fakeroot-1.21/build/biarch/.configured' failed
make: *** [/home/freetz/freetz/source/host-tools/fakeroot-1.21/build/biarch/.configured] Error 77

.config
[_config.tar.gz](https://github.com/Freetz/freetz/files/1124592/_config.tar.gz)

Installed Packages
[packages.tar.gz](https://github.com/Freetz/freetz/files/1124596/packages.tar.gz)

What did i wrong ?

Thanks

ds2k5, 2017-07-06:
Solution:

sudo apt-get install libc6-dev-i386 gcc-multilib

closed 2017-07-06 by ds2k5

olistudent, 2017-07-06:
Hi ds2k5,

I think these packages are in the prerequisites? http://freetz.org/wiki/help/howtos/common/install

Regards,
Oliver