Freetz/freetz #10: Issue on 7270v2, reboot loop

jturello-ulink, 2017-08-23:
Hi,
I've compiled freetz for 7270v2, with advanced user setting and a few changes like turning on "bridge-utils" and "busybox".
Everything went good (make -> success message -> image file created) until I uploaded the image to the Fritz!Box. Since then, the device reboot in loop: led power on, then all led turn red, etc. Seems bricked...
What could happened? Is there any way to restore the device?

Thanks

My computer info:
Toshiba R700 Laptop under Linux Mint 18.2 
Linux 4.10.0-32-generic #36~16.04.1-Ubuntu SMP Wed Aug 9 09:19:02 UTC 2017 x86_64 x86_64 GNU/Linux

PeterPawn, 2017-08-23:
These old devices (the 7270v2 starts in 2008 and its successor 7270v3 in May 2009) often have problems with capacitors on the PCB (visible bulges of the vent on top of them due to electrolyte problems) ... this leads to different problems. Usually the Power LED should blink for at least five seconds, if the bootloader was able to initialize the device. Your description sounds like a defect already in this bootloader.

Some references (in German) for the "known problem":
https://www.heise.de/ct/hotline/Fritzbox-durch-Elko-Tausch-wiederbeleben-2392678.html
https://www.elko-verkauf.de/shop-bereich/product/view/100/652.html
https://www.youtube.com/watch?v=hwCsPkdGEbo

Usually these old devices are not worth the effort to repair them (a 7412 device - without ISDN, but with a more powerful processor - is cheap on eBay) ... if you're not a "maker". But my opinion is solely based on your description of the error and the mentioned LED display behavior. Take an own look onto the PCB and try to find "bloated" capacitors - if they are replaced, most devices are fit again.

Any software error is easy to fix (even without a Windows system), if the bootloader (and the tiny FTP server, which is provided there) is still accessible.

Everything you need is an original firmware image (your Freetz image was built from such an image), where you may extract the "kernel.image" file. This file has to be written to "mtd1" using the FTP server - there are many how-to instructions in the internet and I'm sure, some of them are in English too.

chlorophyll-zz, 2018-03-10:
Here you will find the recovery Image,  to restore the bootloader with a Windows System.
http://web.archive.org/web/http://download.avm.de:80/fritz.box/fritzbox.fon_wlan_7270_v2/x_misc/deutsch/

PeterPawn, 2018-03-10:
This recovery program needs a full-functioning boot-loader, too.

It's nothing more than a special FTP client program, which uses some well-known commands to communicate with the device and writes a new firmware image (which is contained as payload in the executable file).

If the boot-loader is working, it shows a short blinking sequence (about 5 seconds with 2 Hz) immediately after the device gets powered up. It marks the period, while the FTP server from EVA (this is the boot-loader by AVM) is listening for an incoming connection. 

And only in this time, a FTP client may connect - if the boot-loader isn't working (as reported in #1 - the "all LEDs turned red" info is unambigious), the recovery program can not repair anything.

closed 2018-04-26 by er13