Freetz/freetz #39: Missing /dev/net/tun

stblassitude, 2018-05-11:
In r14683 and with the config attached below, I have a working OpenVPN setup on my 7390, but only after I manually `mkdir /dev/net; mknod /dev/net/tun c 10 200`. I'm not quite certain how to proceed; should the device node be available by default, or would the OpenVPN package be responsible for creating it?

I had to enable a number of removal patches to get the image size small enough.

```
vagrant@default:~/freetz$ grep -v '^#' .config | grep -v '^$' 
FREETZ_HAVE_DOT_CONFIG=y
FREETZ_USER_LEVEL_ADVANCED=y
FREETZ_SHOW_ADVANCED=y
FREETZ_TYPE_7390=y
FREETZ_TYPE_LANG_DE=y
FREETZ_TYPE_FIRMWARE_06_8X=y
FREETZ_TYPE_FIRMWARE_FINAL=y
FREETZ_TYPE_LANGUAGE="de"
FREETZ_TARGET_IPV6_SUPPORT=y
FREETZ_REMOVE_MEDIASRV=y
FREETZ_REMOVE_NAS=y
FREETZ_REMOVE_BRANDING_1und1=y
FREETZ_REMOVE_BRANDING_ewetel=y
FREETZ_REMOVE_BRANDING_otwo=y
FREETZ_REMOVE_CAPIOVERTCP=y
FREETZ_REMOVE_DTRACE=y
FREETZ_REMOVE_FTPD=y
FREETZ_REMOVE_NTFS=y
FREETZ_REMOVE_USBHOST=y
FREETZ_REMOVE_PRINTSERV=y
FREETZ_REMOVE_PRINTSERV_MODULE=y
FREETZ_REMOVE_SAMBA=y
FREETZ_REMOVE_AVM_E2FSPROGS=y
FREETZ_PATCH_FREETZMOUNT=y
FREETZ_USBSTORAGE_AUTOMOUNT=y
FREETZ_AUTOMOUNT_EXT3=y
FREETZ_AUTOMOUNT_FAT=y
FREETZ_AUTOMOUNT_NTFS=y
FREETZ_ADD_ETCSERVICES=y
FREETZ_RESTORE_DEBUG_CFG_SUPPORT=y
FREETZ_AVMDAEMON_DISABLE_NTP=y
FREETZ_PACKAGE_DROPBEAR=y
FREETZ_PACKAGE_DROPBEAR_DISABLE_HOST_LOOKUP=y
FREETZ_PACKAGE_HASERL=y
FREETZ_PACKAGE_INETD=y
FREETZ_PACKAGE_MOD=y
FREETZ_PACKAGE_MOD_ETCSERVICES=y
FREETZ_PACKAGE_MODCGI=y
FREETZ_PACKAGE_NTFS=y
FREETZ_PACKAGE_OPENVPN=y
FREETZ_PACKAGE_OPENVPN_VERSION_2_4=y
FREETZ_PACKAGE_OPENVPN_OPENSSL=y
FREETZ_PACKAGE_OPENVPN_WITH_BLOWFISH=y
FREETZ_PACKAGE_OPENVPN_WITH_LZO=y
FREETZ_PACKAGE_OPENVPN_ENABLE_SMALL=y
FREETZ_BOX_CERTIFICATE_SUPPORT_AVAILABLE=y
FREETZ_PACKAGE_IPTABLES_IS_SELECTABLE=y
FREETZ_PACKAGE_AVM_FORWARDING=y
FREETZ_PACKAGE_OPENVPN_CGI=y
FREETZ_LIB_libcrypto=y
FREETZ_LIB_libssl=y
FREETZ_OPENSSL_VERSION_PROMPT=y
FREETZ_OPENSSL_VERSION_1=y
FREETZ_OPENSSL_SHLIB_VERSION="1.0.0"
FREETZ_OPENSSL_LIBCRYPTO_EXTRA_LIBS="-ldl"
FREETZ_OPENSSL_SMALL_FOOTPRINT=y
FREETZ_OPENSSL_CONFIG_DIR="/mod/etc/ssl"
FREETZ_LIB_liblzo2=y
FREETZ_LIB_libntfs=y
FREETZ_LIB_ld_uClibc=y
FREETZ_LIB_libcrypt=y
FREETZ_LIB_libdl=y
FREETZ_LIB_libm=y
FREETZ_LIB_libpthread=y
[... 548 lines cut ...]
FREETZ_KERNEL_VERSION_2_6_32_MAX=y
FREETZ_KERNEL_VERSION_3_10_MAX=y
FREETZ_KERNEL_LAYOUT="iks"
FREETZ_MODULES_KVER="2.6.28.10"
FREETZ_KERNEL_PATCHES="2.6.28.10"
FREETZ_AVM_UCLIBC_0_9_33=y
FREETZ_AVM_UCLIBC_NPTL_ENABLED=y
FREETZ_AVM_SOURCE_7390_06_80=y
FREETZ_AVM_SOURCE_ID="7390_06.80"
FREETZ_AVM_SOURCE_FOR_SYSTEM_TYPE_IKS=y
FREETZ_GCC_ABI="32"
FREETZ_GCC_ARCH="mips32r2"
FREETZ_GCC_TUNE="24kc"
FREETZ_GCC_FLOAT_ABI="soft"
FREETZ_REPLACE_KERNEL_AVAILABLE=y
FREETZ_REPLACE_KERNEL_EXPERIMENTAL=y
FREETZ_REPLACE_MODULE_AVAILABLE=y
FREETZ_TYPE_PREFIX="7390"
FREETZ_TYPE_PREFIX_SERIES_SUBDIR="06_8X"
FREETZ_INSTALL_BASE=y
FREETZ_REPLACE_BUSYBOX=y
FREETZ_CIFS_SUPPORT_AVAILABLE=y
FREETZ_NFS_SUPPORT_AVAILABLE=y
FREETZ_NFS_SUPPORT_AVAILABLE_AS_MODULE=y
```

er13, 2018-05-23:
udev is responsible for creating it. Please disable automatic start of openvpn and provide the output of the following commands right after rebooting the box:
```
dmesg | grep -E "/dev(/net)?/tun"
find /dev -name tun | xargs ls -l
```
and then the output of the 2nd line right after starting openvpn.

er13, 2018-07-08:
ping