Freetz/freetz #40: Option for custom firmware, e.g. inhaus

Gizmo-Ger, 2018-05-16:
Would it be possible to add the option to specify a custom firmware? Right now one can only choose between release and labor. It would be nice to have the option selecting a custom firmware.

PeterPawn, 2018-05-16:
```
 .config - Freetz Configuration
 > Override options ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
  +------------------------------------------------------------------------------------------------------ Override firmware source ------------------------------------------------------------------------------------------------------+
  | FREETZ_DL_OVERRIDE:                                                                                                                                                                                                                  |
  |                                                                                                                                                                                                                                      |
  | Enable this to override automatically generated entries.                                                                                                                                                                             |
  |                                                                                                                                                                                                                                      |
  |                                                                                                                                                                                                                                      |
  | Symbol: FREETZ_DL_OVERRIDE [=y]                                                                                                                                                                                                      |
  | Type  : boolean                                                                                                                                                                                                                      |
  | Prompt: Override firmware source                                                                                                                                                                                                     |
  |   Location:                                                                                                                                                                                                                          |
  |     -> Override options                                                                                                                                                                                                              |
  |   Defined at config/.cache.in:57514                                                                                                                                                                                                  |
  |   Selected by: FREETZ_TYPE_CUSTOM [=n] && <choice> && FREETZ_SHOW_EXPERT [=y]                                                                                                                                                        |
  |                                                                                                                                                                                                                                      |
```

... gibt es schon.

er13, 2018-05-16:
Such an "option" is already available:
 1. "Level of user competence" => Expert
 2. "Override options/Override firmware source" => [*]
 3. Set "Override options/{Firmware site, Firmware source, MD5 checksum for downloaded Image}" to the appropriate/desired values.

closed 2018-05-16 by er13