Freetz/freetz #42: 7590 International image

choco, 2018-06-20:
Hi, I'm trying to build an image for the 7590 international version. I basically just changed the weburl pointing to the german Labour to the international labour (which is 06.98-56436)
I can produce an image but FritzBox refuses to flash it... any help would be freatly appreciated!

PeterPawn, 2018-06-21:
Use the boot-loader to install your first image - load the image into RAM and let it start; it will install itself into flash memory. You may have a look on https://github.com/PeterPawn/YourFritz/blob/master/eva_tools/eva_to_memory or https://github.com/PeterPawn/YourFritz/blob/master/eva_tools/EVA-FTP-Client.ps1#L244 to get an idea of the needed steps (or you could use the scripts, too).

As long as the running FRITZ!OS version does not contain the public parts of the RSA key, which was used to sign the own firmware (or if the own image isn't signed at all), FRITZ!OS refuses to install it via GUI upload.

choco, 2018-06-21:
Thank you for you help.

I managed to flash the image, I'll just report here the steps since all the documentation about this seems to be only available in German.

1. Create image for you firmware using freetz, I'll call it image-step1.image
2. Convert image to memory format `bash image2ram < image-step1.image > image-step2.image
3. Shutdown FritzBox, connect to it using LAN cable, set static ip for router and for yourself (for example Fritz_IP: 192.168.1.1 and My_IP: 192.168.1.2)
4. Execute `bash eva_discover INTERFACE=eth0 FROM=My_IP TO=Fritz_IP HOLD=0`
5. Start FritzBox and wait for command above to terminate, if successful will return EVA_FOUND=1
6. Execute `bash eva_to_memory image-step2.image FritzIP`
7. Wait for image upload and then wait for flashing.

closed 2018-06-21 by choco

PeterPawn, 2018-06-21:
Congratulations on your success.

Correct summary (for the most parts) ... but there are additional possible ways to prepare the network configuration and it's not always necessary to change the local IPv4 address. 

The 2nd step may be skipped completey, due to Freetz is able to create the needed file itself (see the addendum on my first answer above).

Any further reader could (or better *should*) consider, to enter the commands for step 3 and 4 as a single line (best combined with a check of "$EVA_FOUND" to the value "1"), because there's only a very limited time (5 to 10 seconds) afterwards, if the discovery script was called with "HOLD=0". 

If this time passes without a FTP connection attempt, the box continues its boot process.

---

There are _detailed_ descriptions in German language on the IPPF board - because the devices of AVM are mainly used in Germany.

As far as I can see, it's always possible to translate these descriptions (using Google Translate or your own, favorite translation offer) to English - the sense should be recognizable and usually any further questions will be answered with very short delays, too.

But additional discussions/explanations should go on at the IPPF bulletin board, it's no problem to ask any question in English there. Here the meaning of "question" is limited (by myself at least) to an attempt to clarify misunderstandings and will not expand to a request (or demand) to repeat the same description once again in English. The interested audience is too limited, than that a detailed explanation in English is really needed (or is worth its efforts).