Freetz/freetz #43: 4020 not booting

programminghoch10, 2018-07-02:
The 4020 is not booting after successfully compiling and uploading the image.

[Bootlog.txt](https://github.com/Freetz/freetz/files/2154683/Bootlog.txt)

This is the bootlog from my 4020 captured at the serial console.
If needed I can attach the env too.

as stated in line 633 its redirecting kernel messages to /dev/debug so I'm trying to get this file too...

er13, 2018-08-08:
Could you please provide your .config?

Are you able to reach AVM/Freetz web-if? Does the pcpd-crash happen after you actively try to configure something in AVM web-if or just after doing nothing except for waiting for a little bit more than 3 minutes?

Have you tried to disconnect all cables before booting the box? I.e. to check if it only happens when some devices are connected.

programminghoch10, 2018-08-08:
So I tried being more professional this time...
I'm tring to help solving the Ticket #2772
http://freetz.org/ticket/2772
As stated in the last comment, the power led blinks and it reboots after a while.
I thought that isn't enough to debug this problem.

So this time I documented everything (I hope):
Here is the . config file:
[.config.txt](https://github.com/Freetz/freetz/files/2272190/default.config.txt)
And here is the freetz compile log:
[freetz compile.txt](https://github.com/Freetz/freetz/files/2272192/freetz.compile.txt)
The [upload](https://github.com/Freetz/freetz/files/2272195/upload.txt) also went as it should be.

[bootlog.txt](https://github.com/Freetz/freetz/files/2272225/bootlog.txt)
So here is the bootlog of the first try.
And because you said connected devices could be manipulating the outcome, I only connected power and the serial connection during boot (nothing beeing transmitted to the box) . Also I did reset the box to factory defaults before uploading the image.

Before booting I captured the env and found this line:
crash                 [0]5daaf03a,5b3a9453,1[1]0,0,0[2]11955f02,5b39f98d,1[3]0,0,0
This didn't change though

Second try I connected LAN2
In the time the box didn't respond to anything after boot (no ping and obviously no http request)
Other than that the boot went the same (as far as I can see). Even the timings were similar and the crash entry in the env is the same.
[bootlog2.txt](https://github.com/Freetz/freetz/files/2272265/bootlog2.txt)

So the pcpd seems to be crashing on its own, because I'm not able to access any Web-IF.

er13, 2018-08-09:
Thx a lot!

The new bootlogs are a little bit more complete than the previous one and are full of `can't resolve symbol '__ctype_b_loc'`. Looks like an uClibc-issue (wrong uClibc .config parameters, probably locale related).

I'll try to take a look into the issue later this day.

er13, 2018-08-09:
Could you please update to the latest revision and check if it solves the problem?

Please start with a fresh checkout. You will also have to build your own toolchain as the available download toolchain is not compatible with 4020.

Thx for testing!

programminghoch10, 2018-08-09:
I needed to set up a new VM because my previous VM didn't have enough space to build my own toolchain... After some troubles I managed to get the make running, but it now says that it cant determine the version
"ERROR: cannot determine version"
Is this related to the 4020 issue or is my freetz misconfigured in any way?

[freetz compile.txt](https://github.com/Freetz/freetz/files/2276091/freetz.compile.txt)

f-666, 2018-08-10:
Es scheint dir die Datei .version im Freetz Hauptverzeichnis zu fehlen. Eventuell ist beim Auschecken was schief gegangen.

programminghoch10, 2018-08-10:
Okay dann werde ich das nochmal versuchen

er13, 2018-08-10:
Die Analyse von f-666 ist richtig. .build-prerequisites fehlt laut Log auch. Bitte sicherstellen, dass der Checkout vollständig ist (am besten einfach neu auschecken) und erneut versuchen.

Ansonsten der Hinweis: es bedarf keines erneuten Aufsetzens einer VM. Du kannst einfach eine neue virtuelle Festplatte, die groß genug ist, der existierenden VM hinzufügen, diese partitionieren und den Build-Prozess dann unterhalb des Mount-Points der neuen Partition stattfinden lassen.

programminghoch10, 2018-08-10:
Das Image hat sich jetzt ohne probleme kompliliert.
[freetz compile.txt](https://github.com/Freetz/freetz/files/2277923/freetz.compile.txt)
(Ich habe den build toolchain weggelassen, das wäre zu lange)

Die Box scheint ohne Probleme zu starten.
Ich hatte LAN2 verbunden und beide Web-IF sind erreichbar.

[bootlog.txt](https://github.com/Freetz/freetz/files/2277916/bootlog.txt)

Nach einigen Minuten habe ich dann angefangen die Web-IF zu durchforsten und ein paar Einstellungen vorzunehmen. Soweit scheint alles zu klappen.

programminghoch10, 2018-08-11:
Was kommt jetzt?

Ich würde mich bereit erklären verschiedene Konfigurationen zu testen um den "EXPERIMENTAL" Status loszuwerden.
Ich würde beginnen zu testen welche removal patches nicht funktionieren.

Ich habe auch noch eine 4040 für die man Support in Angriff nehmen könnte, Ich weiß das ist ein größerer Task da die Architektur nicht MIPS sondern ARM ist. Also wenn jemand mal viel Zeit hat dann würde ich da gerne helfen dies umzusetzen.

programminghoch10, 2018-08-11:
Ich habe alle removal patches aktiviert und bei jedem fehler den jeweiligen patch deaktiviert.
Zusammengefasst: 
[removal patches.txt](https://github.com/Freetz/freetz/files/2280108/removal.patches.txt)
nachdem ich alle mit fehler deaktiviert habe hat es erfolgreich funktioniert mit dieser config:
[.config.txt](https://github.com/Freetz/freetz/files/2280111/default.config.txt)

er13, 2018-08-11:
Removal-Patches funktionieren seit Fritz!OS 6.5x auch bei den anderen Boxen nicht. Ist also kein 4020-spezifisches Problem und müsste generell in Angriff genommen werden, wozu ich allerdings keine Lust habe.

Was Du allerdings noch testen könntest, ist, ob [diese Änderung](https://github.com/Freetz/freetz/commit/abc62a7c53b702ee0ee78df2442fe1de223610ff) wirklich zwingend notwendig ist. Diese wurde damals nur auf Verdacht eingebaut.

Könntest Du also bitte alle Removal-Patches wieder deaktivieren und die oben verlinkte Änderung in Deinem lokalen Checkout wieder rückgängig machen. Es ist kein kompletter Neubau notwendig, einfach mal `make` in dem bisherigen Checkout aufrufen.

programminghoch10, 2018-08-11:
Die Box bootet ohne die Änderung und beide Web-IF sind erreichbar.
[bootlog.txt](https://github.com/Freetz/freetz/files/2280182/bootlog.txt)
[.config.txt](https://github.com/Freetz/freetz/files/2280183/default.config.txt)

closed 2018-08-11 by programminghoch10

er13, 2018-08-12:
Danke fürs Testen und fürs Bootlogs zur Verfügung stellen!

Bootlogs waren entscheidend - haben sofort auf den Fehler hingewiesen.

Wäre nicht schlecht, wenn Du noch die Bilder und eine kurze Anleitung an das Ticket anhängen könntest, wie Du die serielle Console angelötet hast, um die anderen Interessierten zu motivieren ;-)