Freetz/freetz #44: Does FRITZ!Box 7590 support PXE netboot?

bittner, 2018-07-05:
I'm wondering if the [FRITZ!Box 7590](https://avm.de/produkte/fritzbox/fritzbox-7590/) supports PXE for network booting (with TFTP and DHCP with "next-server" option). I can't seem to find any information on this in the specs, the official forum and elsewhere.

If it doesn't suport PXE out-of-the-box does this project add this functionality to the box?

derWeihnachtsmann, 2018-07-06:
This Project will add functionality for it. 
You can install dnsmasq standalone (no more client-listung in the AVM-GUI, since DHCP is full managed by dnsmasq then) 
..or as a proxy (provide PXE-boot option only and do not respond to dhcp-questions). 

I searched a google guide for you https://blog.n0dy.radio/2014/09/14/network-booting-with-dnsmasq-in-proxy-mode/

er13, 2018-07-08:
This is a bug tracking system. For questions like yours please use [IP Phone Forum](https://www.ip-phone-forum.de/forums/freetz.525/)

closed 2018-07-08 by er13

derWeihnachtsmann, 2018-08-05:
Edit: There are still all Members in FritzUI visible, even when your DHCP Server is on different IP.