Freetz/freetz #50: 6490 Supported?

gelsas, 2018-08-29:
Is the 6490 (Vodafone branded) Box Supported?

PeterPawn, 2018-08-29:
No, the original firmware (which will be modified by Freetz only) does not support the "kdg" branding.

gelsas, 2018-08-29:
Is there another solution Like freetz for my Fritzbox 6490 with the kdg Branding ?

PeterPawn, 2018-08-29:
Doubtless ... but:
- this isn't the right place for a discussion of anything other than a Freetz problem
- there are bulletin boards, where these questions are welcome - e.g. ip-phone-forum.de
- even in another place, you should consider to tell a recipient/reader much more 'bout your real goals - no-one here can realize, what your expectations are and why you want to modify the original firmware ... e.g. if you want to make the FRITZ!Box play music, it could be tricky due to the lack of a speaker

closed 2018-08-29 by er13