Freetz/freetz #52: Use previous driver version functionality

jekkos, 2018-09-21:
Hi there,
I ran in this firmware modification repo by accident. In fact I read about the fritzbox 'use previous driver' functionality in the firmware. In belgium modems with specific fritz os firmware versions are allowed to go online. This mechanism seems to be implemented in fritzbox (as it is able to identify itself with a version string at will).

Do you have any idea how this works technically? Like is there a command that I can enter in my cli to mimic this functionality? Some help would be greatly appreciated.

jekkos, 2018-09-21:
For example can this value be overridden by patching the vdsl firmware blob? Or is it configured through the vdsl_cpe_control binary? Thanks.

er13, 2018-10-22:
Not sure what exactly you want.

If you just want to pretend to be using some specific Fritz!OS version without actually using the DSL driver from it then it might be enough to change the Fritz!OS version string in the following files from the unpacked firmware image:

- /etc/.version
- /etc/version
- /etc/init.d/rc.conf

and possibly
 

- /etc/.revision
-  /etc/.subversion

I however never tested it myself. So everything is on your own risk without any guarantee / liability or whatever!

I also don't know how the Belgian providers restrict the access to the internet - by the Fritz!OS version or by the DSL driver version. If the latter, then it's obviously not enough to fake the Fritz!OS version.

--

If you really want to use the old DSL driver then just enable it in the AVM web interface.

--

p.s. I'm closing this ticket as this is a bug tracking system and not the support system. If you need some support please use [IP Phone Forum](https://www.ip-phone-forum.de/forums/avm.361/). Even though the most threads are in German, you can still ask in English.

closed 2018-10-22 by er13

jekkos, 2018-10-26:
Thanks for your answer and reference to the forum. In this case it's the dsl version so indeed, changing fritz os might not really help.

jekkos, 2019-02-07:
Ok think I found what I was looking for. Attach `dsl_control -D1` to the kernel module and then all diagnostics are shown. Also I see what line inventory parameters are set here. Once I start driver in debug mode it applies a custom firmware string (the same as OpenWrt does).

jekkos, 2019-02-07:
Also on the 'ceritfiied' fritzbox I found a vdsl.cfg file that is loaded by the daemon to tweak SNR params. It needs to be applied using `-A` argument.