Freetz/freetz #53: Minidlna does not discover m2ts and mpg files

donacarr, 2018-10-11:
The m2ts files are AVCHD format and mpg files are MPEG-2 based
Minidlna 1.2.1
Freetz Revisition 14887

Let me know if further details are needed.
Thank you

er13, 2018-10-17:
Does playing these files work with minidlna installed on your Linux machine (some more or less recent Linux distribution)? If no, then it's a minidlna problem and should be reported to minidlna developer and not Freetz.

Freetz changes nothing in this regard in minidlna code. The only reason I could imagine is an old ffmpeg version used by Freetz, which doesn't support these "formats".

er13, 2018-10-17:
Note also this thread: https://sourceforge.net/p/minidlna/discussion/879957/thread/eda430d9/

er13, 2018-10-25:
Ping

donacarr, 2018-10-25:
Thank you for the answer but I need some time to double check.

donacarr, 2018-11-01:
I looked the log file and I found basically two types of error:
scanner.c:490: warn: Unsuccessful getting details for /var/media/ftp/uStor05/Riprese/2011-12.m2ts
metadata.c:697: warn: Opening /var/media/ftp/uStor05/Riprese/video1.mpg failed! [Invalid data found when processing input]

But I don't have a linux box to try minidlna.
Maybe I should open a thread here https://sourceforge.net/p/minidlna/discussion/.
If you want to close the issue...up to you,
Thank you.

er13, 2018-11-02:
I just want to clarify if it's a Freetz issue (i.e. Freetz does something in a wrong way or uses an old library, e.g. ffmpeg) or it's a minidlna issue (i.e. the support for these files is not available in minidlna itself regardless of the platform it is started on). If the latter then the minidlna developers should take of the issue, they know their code much better.

donacarr, 2018-11-02:
ok...then let me try minidlna on a virtualbox image of ubuntu ...I am just downloading it...I will let you know. thank you.

donacarr, 2018-11-02:
I completed to install minidlna 1.2.1 on ubuntu 14 as indicated here https://www.htpcguides.com/install-latest-readymedia-minidlna-ubuntu/

And the two kind of files are found by minidlna...so I think it is a bug in freetz.
Let me know if you need some details...

er13, 2018-11-02:
Please provide your .config.

Are
```
FREETZ_PACKAGE_FFMPEG_DEMUXER_mpegvideo
FREETZ_PACKAGE_FFMPEG_DEMUXER_mpegps
FREETZ_PACKAGE_FFMPEG_DEMUXER_mpegts
```
enabled in it?

donacarr, 2018-11-03:
Here is the file.
However only FREETZ_PACKAGE_FFMPEG_DEMUXER_mpegvideo is enabled...
I did not change the config...just using the default one....
[.config.zip](https://github.com/Freetz/freetz/files/2545029/default.config.zip)

Waiting your feedback.
Thank you.

er13, 2018-11-03:
Please also enable everything under
```
Packages/Packages/FFmpeg 1.2.12/Configuration/--- Parsers ---/
```

donacarr, 2018-11-03:
It works now.. thank you.
Just another question... I noticed that upgrading from 6.83 to 6.84 the USB devices now have a different label...something like uStor05 or uStor15. Before fritzbox showed  the correct label assigned during the formatting (for instance MyPassport ) ... Do you know how to restore the old behaviour?

er13, 2018-11-03:
In Freetz web-If : `Freetz/Settings/Freetzmount/Mount points naming scheme` => `Partition label`

donacarr, 2018-11-04:
thank you ...it works....my brother has revision 14237 (18 months older than me) and he complains that fritz sometime when restarts change the directory at which attach the filesystem. In the event logs there is something like _Partition mounted as SanDisk-UltraFit-01_ but sometime he sees _Partition mounted as SanDisk-UltraFit-05_  ... I don't know if it is related to the device.. please look the file in attach. It shows that label was simply _Sandisk_ .......Any clue?  could my revision fix that?
[Sandisk ext4.zip](https://github.com/Freetz/freetz/files/2545683/Sandisk.ext4.zip)

er13, 2018-11-04:
It looks like your brother uses `Mount points naming scheme` called `Vendor-Product as prefix` (Vendor=Sandisk, Product=UltraFit). The number at the end depends on the order the USB devices are mounted in. As the number changes I assume he has multiple USB devices connected. The mount order cannot unfortunately be controlled, this is where the mount by `Partition label` comes in handy.

So he should just switch to the same method as you, to `Partition label`. Note if he has multiple devices connected the partition labels must be unique otherwise just one of them will be mounted.

p.s. I'm closing the ticket as the new questions have nothing to do with the issue reported initially. If you have questions and need some support please use [IP Phone Forum](https://www.ip-phone-forum.de/forums/avm.361/) for that. Even though it's main language is German you can still ask questions in English.

closed 2018-11-04 by er13