Freetz/freetz #54: Aufruf von "fwmod"

PeterPawn, 2018-10-30:
Bei speziellen Einstellungen in der Konfiguration:
```shell
FREETZ_FWMOD_SKIP_UNPACK=y
FREETZ_FWMOD_SKIP_MODIFY=y
FREETZ_FWMOD_SKIP_PACK=y
# FREETZ_FWMOD_USBROOT is not set
# FREETZ_FWMOD_NFSROOT is not set
```
(wenn man nur die Freetz-Toolchain und ausgewählte Pakete erzeugen will, aber kein Image) wird trotzdem "fwmod" aufgerufen, was dann mangels Aktion mit dem Help-Screen und einem Fehlercode reagiert (und in der Folge mit einem Abbruch des "make"-Laufs).

er13, 2018-10-30:
Im menuconfig wird das doch aber deutlich als [Invalid fwmod configuration](https://github.com/Freetz/freetz/blob/master/config/ui/fwmod.in#L8) gekennzeichnet.

PeterPawn, 2018-10-30:
Das wird bei mir nicht als Kommentar angezeigt im Menü, wenn ich die Einstellungen vornehme:
[image]
Außerdem wäre es m.E. einfacher, dann den Aufruf von "fwmod" nicht auszuführen.

er13, 2018-10-30:
Stand jetzt ist es kein "gewollter Anwendungsfall". Aber es spricht auch nicht wirklich was dagegen, auf den Aufruf von fwmod bei diesen Einstellungen zu verzichten. Ich schaue es mir an. Allerdings nicht heute.

closed 2018-10-31 by mhei