Freetz/freetz #57: 6490 Callmon Kernelpanic

h45rd, 2018-11-08:
Hi,

First off I want to express my sincere gratitude to everyone that worked on freetzing the 6490, especially @f-666. Great job :)

Prereqs:
The following is concerning the latest dev branch (14986) of freetz, using Fritz!OS 7.x as base image.
Clean checkout and build, new offline 6490 with factory defaults loaded.

While fiddling around I stumbled upon a kernel panic regarding callmon.
As soon as I add any of my voip numbers, the avm watchdog will throw the following kernel panic at 125s on the next reboot and the box will keep bootlooping. Only happens when callmon is added to the image. Doesn't matter if callmon is actually running or set to start automatically or not.

```
[...]
[  125.313182] Pid: 0, comm: swapper Tainted: P         C  2.6.39.3 #1
[  125.313345] Call Trace:
[  125.313429]  [<c1267a20>] ? AVM_WATCHDOG_set_timeout+0xa0/0xa0
[  125.313605]  [<c146f209>] panic+0x38/0x14a
[  125.313728]  [<c1267a20>] ? AVM_WATCHDOG_set_timeout+0xa0/0xa0
[  125.313876]  [<c1267a38>] AVM_WATCHDOG_timer_init_ctrl_handler+0x18/0x20
[  125.314047]  [<c1043087>] run_timer_softirq+0xa7/0x210
[  125.314179]  [<c103d31f>] __do_softirq+0x8f/0x190
[  125.314324]  [<c103d290>] ? remote_softirq_receive+0x20/0x20
[  125.314464]  <IRQ>  [<c103d6d5>] ? irq_exit+0x95/0xa0
[  125.314601]  [<c1018e6e>] ? smp_apic_timer_interrupt+0x6e/0xf0
[  125.314750]  [<c147578e>] ? apic_timer_interrupt+0x2a/0x30
[  125.314890]  [<c100940b>] ? mwait_idle+0x5b/0xa0
[  125.315008]  [<c10015df>] ? cpu_idle+0x3f/0x80
[  125.315134]  [<c145a65a>] ? rest_init+0x7b/0x81
[  125.315251]  [<c16426f4>] ? start_kernel+0x2be/0x2c3
[  125.315377]  [<c16420ba>] ? i386_start_kernel+0xba/0xc2
[  125.315507] Kernel panic - not syncing: ar7wdt_hw_reboot: init sequence hangs !
[  125.315511] 
[  125.315737] Pid: 0, comm: swapper Tainted: P         C  2.6.39.3 #1
[  125.315898] Call Trace:
[  125.315968]  [<c1267a20>] ? AVM_WATCHDOG_set_timeout+0xa0/0xa0
[  125.316114]  [<c146f233>] panic+0x62/0x14a
[  125.316220]  [<c1267a20>] ? AVM_WATCHDOG_set_timeout+0xa0/0xa0
[  125.316367]  [<c1267a38>] AVM_WATCHDOG_timer_init_ctrl_handler+0x18/0x20
[  125.316535]  [<c1043087>] run_timer_softirq+0xa7/0x210
[  125.316673]  [<c103d31f>] __do_softirq+0x8f/0x190
[  125.316834]  [<c103d290>] ? remote_softirq_receive+0x20/0x20
[  125.316973]  <IRQ>  [<c103d6d5>] ? irq_exit+0x95/0xa0
[  125.317170]  [<c1018e6e>] ? smp_apic_timer_interrupt+0x6e/0xf0
[  125.317322]  [<c147578e>] ? apic_timer_interrupt+0x2a/0x30
[  125.317462]  [<c100940b>] ? mwait_idle+0x5b/0xa0
[  125.317599]  [<c10015df>] ? cpu_idle+0x3f/0x80
[  125.317740]  [<c145a65a>] ? rest_init+0x7b/0x81
[  125.317885]  [<c16426f4>] ? start_kernel+0x2be/0x2c3
[  125.318043]  [<c16420ba>] ? i386_start_kernel+0xba/0xc2
[  125.343872] 
[  125.343879] 
[  125.343882] ------------------- Last part of Panic-log Content: -------------------
```

During a regular boot the watchdog should be done at around 91s:
`[   91.511427] AVM_WATCHDOG: System Init Ueberwachung abgeschlossen (84590 ms noch verfuegbar)`

Any advice on how to debug this further?
I am not knowledgeable enough regarding the inner workings of AVMs image to draw any meaningful conclusions from what I know so far.

Cheers,
h45rd

P.S.: Was ist hier eigentlich best practice RE: Sprache. Deutsch oder Englisch?

er13, 2018-11-08:
What is callmon? [This tool](https://www.telinta.com/solutions/callmon/), the package callmonitor included in Freetz or something else?

f-666, 2018-11-08:
Bitte die .config Datei anhängen.
Tritt der Fehler auch ohne _callmon_ auf?

h45rd, 2018-11-08:
Uff.. sorry für die Verwirrung. 
Ja. Ich meine das Freetz Package Callmonitor.

Der Fehler tritt nur auf, wenn Callmonitor mitgebaut wird und im Image ist und erst sobald eine Telefonnummer konfiguriert wird.
Ob der Callmonitor Daemon läuft oder nicht ist dabei egal. Sobald er im Image ist **und** eine Nummer konfiguriert ist, gibt es den o.g. Crash.

Die Config von meinem Minimalbauversuch habe ich gerade nicht mehr zur Hand. Das Problem trat aber -  wenn ich mich richtig erinnere - schon mit der default Freetz Config, welcher lediglich Callmonitor hinzugefügt wurde, auf.
Ich verifiziere das aber heute Abend und gebe nochmal Rückmeldung.

Hier meine "Alltagsconfig", mit der es auf jeden Fall knallt. Beim Durchschauen und Schreiben kommt mir gerade die Idee, dass es ggfs. auch an der Mitauswahl von @PeterPawn's _FREETZ_PACKAGE_DECRYPT_FRITZOS_CFG=y_ in Verbindung mit dem Callmonitor liegen könnte. Denn ohne die Funktion habe ich nie mit Callmonitor gebaut.
 
<details>
 <summary>.config</summary>

```
#
# Automatically generated file; DO NOT EDIT.
# Freetz Configuration
#
FREETZ_HAVE_DOT_CONFIG=y

#
# User competence ------------------------
#
# FREETZ_USER_LEVEL_BEGINNER is not set
# FREETZ_USER_LEVEL_ADVANCED is not set
FREETZ_USER_LEVEL_EXPERT=y
FREETZ_SHOW_ADVANCED=y
FREETZ_SHOW_EXPERT=y

#
# Hardware/Firmware ----------------------
#
# FREETZ_TYPE_300IP is not set
# FREETZ_TYPE_W501V is not set
# FREETZ_TYPE_2170 is not set

#
# WLAN
#
# FREETZ_TYPE_3020 is not set
# FREETZ_TYPE_3030 is not set
# FREETZ_TYPE_3130 is not set
# FREETZ_TYPE_3131 is not set
# FREETZ_TYPE_3170 is not set
# FREETZ_TYPE_3270_V2 is not set
# FREETZ_TYPE_3270_V3 is not set
# FREETZ_TYPE_3272 is not set
# FREETZ_TYPE_3370 is not set
# FREETZ_TYPE_3390 is not set
# FREETZ_TYPE_3490 is not set

#
# WAN
#
# FREETZ_TYPE_4020 is not set
# FREETZ_TYPE_4040 is not set

#
# Fon
#
# FREETZ_TYPE_FON is not set
# FREETZ_TYPE_5010 is not set
# FREETZ_TYPE_5050 is not set
# FREETZ_TYPE_5124 is not set
# FREETZ_TYPE_5140 is not set

#
# Cable
#
[... 2324 lines cut ...]
FREETZ_KERNEL_VERSION_2_6_32_MIN=y
FREETZ_KERNEL_VERSION_2_6_39_MIN=y
FREETZ_KERNEL_VERSION_2_6_39_MAX=y
FREETZ_KERNEL_VERSION_3_10_MAX=y
FREETZ_KERNEL_VERSION_4_4_MAX=y
FREETZ_AVM_UCLIBC_1_0_14=y
FREETZ_AVM_UCLIBC_NPTL_ENABLED=y
FREETZ_AVM_UCLIBC_XLOCALE_ENABLED=y
FREETZ_AVM_SOURCE_6590_07_00=y
FREETZ_AVM_SOURCE_ID="6590_07.00"
FREETZ_AVM_SOURCE_FOR_SYSTEM_TYPE_PUMA6=y
FREETZ_REPLACE_KERNEL_AVAILABLE=y
FREETZ_REPLACE_KERNEL_EXPERIMENTAL=y
FREETZ_REPLACE_MODULE_AVAILABLE=y
FREETZ_TYPE_PREFIX="6490"
FREETZ_TYPE_PREFIX_SERIES_SUBDIR="07_0X"
FREETZ_INSTALL_BASE=y
FREETZ_REPLACE_BUSYBOX=y
# FREETZ_CIFS_SUPPORT_NEEDS_REPLACE_KERNEL is not set
FREETZ_CIFS_SUPPORT_AVAILABLE=y
FREETZ_NFS_SUPPORT_NEEDS_REPLACE_KERNEL=y
# FREETZ_NFS_SUPPORT_AVAILABLE_AS_MODULE is not set
# FREETZ_NFS_SUPPORT_AVAILABLE is not set
```
</details>

h45rd, 2018-11-09:
Aktuellen trunk neu ausgecheckt. .config wie folgt alles default, bis auf:
Experte -> 6490 -> Fritz!OS 7.x. Dazu Callmonitor und Decrypt Pakete gewählt.
<details>
 <summary>expand</summary>

```
#
# Automatically generated file; DO NOT EDIT.
# Freetz Configuration
#
FREETZ_HAVE_DOT_CONFIG=y

#
# User competence ------------------------
#
# FREETZ_USER_LEVEL_BEGINNER is not set
# FREETZ_USER_LEVEL_ADVANCED is not set
FREETZ_USER_LEVEL_EXPERT=y
FREETZ_SHOW_ADVANCED=y
FREETZ_SHOW_EXPERT=y

#
# Hardware/Firmware ----------------------
#
# FREETZ_TYPE_300IP is not set
# FREETZ_TYPE_W501V is not set
# FREETZ_TYPE_2170 is not set

#
# WLAN
#
# FREETZ_TYPE_3020 is not set
# FREETZ_TYPE_3030 is not set
# FREETZ_TYPE_3130 is not set
# FREETZ_TYPE_3131 is not set
# FREETZ_TYPE_3170 is not set
# FREETZ_TYPE_3270_V2 is not set
# FREETZ_TYPE_3270_V3 is not set
# FREETZ_TYPE_3272 is not set
# FREETZ_TYPE_3370 is not set
# FREETZ_TYPE_3390 is not set
# FREETZ_TYPE_3490 is not set

#
# WAN
#
# FREETZ_TYPE_4020 is not set
# FREETZ_TYPE_4040 is not set

#
# Fon
#
# FREETZ_TYPE_FON is not set
# FREETZ_TYPE_5010 is not set
# FREETZ_TYPE_5050 is not set
# FREETZ_TYPE_5124 is not set
# FREETZ_TYPE_5140 is not set

#
# Cable
#
FREETZ_TYPE_6490=y
# FREETZ_TYPE_6590 is not set

#
# LTE
#
# FREETZ_TYPE_6810 is not set
# FREETZ_TYPE_6820 is not set
# FREETZ_TYPE_6840 is not set
[... 2229 lines cut ...]
FREETZ_KERNEL_VERSION_3_10_MAX=y
FREETZ_KERNEL_VERSION_4_4_MAX=y
FREETZ_AVM_UCLIBC_1_0_14=y
FREETZ_AVM_UCLIBC_NPTL_ENABLED=y
FREETZ_AVM_UCLIBC_XLOCALE_ENABLED=y
FREETZ_AVM_SOURCE_6590_07_00=y
FREETZ_AVM_SOURCE_ID="6590_07.00"
FREETZ_AVM_SOURCE_FOR_SYSTEM_TYPE_PUMA6=y
FREETZ_REPLACE_KERNEL_AVAILABLE=y
FREETZ_REPLACE_KERNEL_EXPERIMENTAL=y
FREETZ_REPLACE_MODULE_AVAILABLE=y
FREETZ_TYPE_PREFIX="6490"
FREETZ_TYPE_PREFIX_SERIES_SUBDIR="07_0X"
FREETZ_INSTALL_BASE=y
FREETZ_REPLACE_BUSYBOX=y
# FREETZ_CIFS_SUPPORT_NEEDS_REPLACE_KERNEL is not set
FREETZ_CIFS_SUPPORT_AVAILABLE=y
FREETZ_NFS_SUPPORT_NEEDS_REPLACE_KERNEL=y
# FREETZ_NFS_SUPPORT_AVAILABLE_AS_MODULE is not set
# FREETZ_NFS_SUPPORT_AVAILABLE is not set
```
</details>

Gleicher Bootloop auch mit dem Minimalimage.
Kann das jemand reproduzieren oder liegt bei mir ggfs. etwas Anderes im Argen?

msebald, 2019-02-07:
Ich habe auch genau dieses Problem. Da ich den Callmonitor verwende um Anrufe an openHAB zu melden, würde ich auch ungerne darauf verzichten.

@h45rd Gibt es denn ein Update? Das letzte Post war ja Anfang Dezember.

Danke und viele Grüße

h45rd, 2019-02-07:
Habe mich leider noch nicht weiter damit befasst. Zeitmangel und Fehlen einer "Bastelbox". Mittlerweile alle im Produktiveinsatz.. da bekomme ich eine auf den Deckel, wenn ich mal für ein paar Stunden spielen will :)

Das Debugging wird zusätzlich dadurch erschwert, dass der Fehler scheinbar nur bei bestimmten Konfigurationen auftritt. Oder nutzt einfach kaum noch jemand den Callmonitor?

@msebald 
Vielleicht finden wir ja den Übeltäter, wenn wir unsere Konfigurationen vergleichen.

Hier die callmonitor.diff:
```
export CALLMONITOR_ENABLED='yes'
export CALLMONITOR_EXPERT='yes'
export CALLMONITOR_PHONEBOOKS='avm callers cache'
export CALLMONITOR_READ_FONBUCH='yes'
export CALLMONITOR_REVERSE_PROVIDER='49:telefonbuch 43:telefonbuch_at 41:search_ch 39:pronto_it 32:wittegids_be other:'
export CALLMONITOR_USERNAME='callmon'
```
Habe am ehesten die Rückwärtssuche (entweder im internen Telefonbuch oder im Web) in Verdacht.
Keine Ahnung, warum die beim Initialisieren anspringen könnte/sollte, aber bei der Vorwahlensuche war das ja auch der Fall.

An der Fritz Konfig selbst wäre ggfs. außergewöhnlich, dass ich 14 Rufnummern konfiguriert und ein recht großes Telefonbuch habe.

[hermann72pbs Analyse](https://www.ip-phone-forum.de/threads/callmonitor-f%C3%BChrt-zum-dauerreboot-seit-der-firmwareversion-7.301550/) legt nahe, dass man im Callmonitor Source schauen sollte, was genau beim Initialisieren (nicht Starten) des Callmonitors je nach gesetzten Optionen passiert und während der Bootphase "hängen" könnte.

Falls Du Zeit und den Nerv hast: schon versucht per Ausschlussverfahren zu ermitteln, welche Funktion den watchdog triggert?
Ich würde so vorgehen: alle Einstellungen zurücksetzen und nur eine Rufnummer konfigurieren. Dann jeweils eine Option im Callmonitor setzen, Reboot. Und das so lange, bis es wieder zum Bootloop kommt.
Dann die entsprechende Stelle im Source suchen..

msebald, 2019-08-01:
@h45rd , sorry, dass ich mich nicht mehr gemeldet habe. War und bin dann auch eingespannt gewesen und ein Testdevice habe ich natürlich auch keines. War bis heute auf 6.92. Die Config vom Callmonitor habe ich natürlich auch nicht rausgesucht.

Wie dem auch sei. Habe heute bce5b22 mit Firmware 7.12 für die Fritzbox 7590 kompiliert und aufgespielt. Kein Absturz, nichts. Habe auch mal einen Testanruf gemacht, der im Callmonitor eine Aktion ausgelöst hat, die auch funktioniert hat. Auch hier kein Absturz. Die Box läuft nun seit ~1 Stunde. Sieht soweit erstmal gut aus.

Sollte es doch wieder zu Abstürzen bzw Reboots kommen, werde ich direkt hier rein schreiben. Auch könnte ich natürlich die Config vom Callmonitor noch raussuchen.

msebald, 2019-08-02:
Heute Nacht, also ca 10 Stunden nach Upgrade auf bce5b22 und 7.12 hat sich die 7590 aufgehängt bzw hat neu gestartet. Ursache unbekannt, k.A., ob ich im Nachhinein hier noch etwas feststellen kann.

Mal gucken, wie's weiter geht. Stable sollte sie schon sein, und ein Reboot alle ~10 Stunden wäre nicht hinnehmbar.