Freetz/freetz #58: libgpg compile error

programminghoch10, 2018-11-08:
And the story goes on...
This time the libgpg fails to compile.
[config.txt](https://github.com/Freetz/freetz/files/2562322/config.txt)
[compile.txt](https://github.com/Freetz/freetz/files/2562323/compile.txt)

er13, 2018-11-09:
Please test if https://github.com/Freetz/freetz/commit/bdc1b147fc11f31dcace0b77ea97417e89e565f9 solves the problem, I cannot reproduce the exact error from your log.

programminghoch10, 2018-11-09:
Seems like its working, as I said its a huge project what im creating.
It nearly uses all packages for my usb rootfs.
It's compiling over 2 hours now without errors, so I guess it's ok.
I'll just reopen this if the same problem still occurs.
Thanks!

closed 2018-11-09 by programminghoch10

programminghoch10, 2018-11-09:
It will never work without problems...
This time it says "Unexpected config area found, extraction aborted"
I found this on the Internet, but it says it was fixed :/
https://www.ip-phone-forum.de/threads/freetz-trunk-7580_06-88-beta-mesh-bauen-revision-14326.295836/
.[config.txt](https://github.com/Freetz/freetz/files/2566136/config.txt)
[compile.txt](https://github.com/Freetz/freetz/files/2566137/compile.txt)

er13, 2018-11-09:
> This time it says "Unexpected config area found, extraction aborted"
Don't forget ARM support is HIGHLY experimental. Kernel config area has not yet been worked on.

As temporary workaround try changing [this line](https://github.com/Freetz/freetz/blob/master/make/linux/kernel.mk#L138) from
```
ifeq ($(strip $(FREETZ_KERNEL_VERSION_3_10_MIN)),y)
```
to
```
ifeq ($(strip $(FREETZ_KERNEL_VERSION_3_10)),y)
```

Don't however try to use the replace-kernel option after that. It definitely won't work.

programminghoch10, 2018-11-10:
I changed the line and run it again,
It now says:
make/linux/kernel.mk:164: recipe for target 'source/kernel/ref-ipq40xx-4040_07.01/linux-4.4/arch/arm/boot/vmlinux.eva_pad' failed