Freetz/freetz #60: MPTCP

hoshymuc, 2018-12-19:
Hi there, 

Has anybody got Freetz working with MPTCP?  I see in the repo "CONFIG_DEFAULT_MPTCP_PM="default"" in the kernel build which I understand to be disabled?

Would love to use my 7580 with VDSL, Cable(external modem), and LTE in tripple aggregation as I have a fast and geo-local hosted VM ready for action :)

PeterPawn, 2018-12-19:
Sounds nearly impossible ... FRITZ!OS uses an own WAN-side IP stack from AVM and the "normal" multi-path TCP implementation will not function here - at least I would expect this so.

AVM itself seems to implement something in this place for the LTE/DSL combined models ... maybe it gets to be used in the future in other models, too.

hoshymuc, 2018-12-19:
Thanks..  that’s unfortunate but not entirely unexpected :)

—
Ashley Evans
+49 176 7073 4051

olistudent, 2018-12-22:
Hi. I tried to remove your phone number from last comment. Hopefully the history is deleted now. I will close this issue because of PeterPawn's remarks.

closed 2018-12-22 by olistudent