Freetz/freetz #61: Bump Labor 20.12.2018

BlackStar999, 2018-12-22:
@@ -601,11 +601,11 @@ menu "Override options"
 		default "FRITZ.Box_7490.113.07.01.image"                                if FREETZ_TYPE_FIRMWARE_FINAL && FREETZ_TYPE_7490 && FREETZ_TYPE_LANG_DE && FREETZ_TYPE_FIRMWARE_07_0X
 		default "FRITZ.Box_7490.en-de-es-it-fr-pl.113.06.31.image"              if FREETZ_TYPE_FIRMWARE_FINAL && FREETZ_TYPE_7490 && FREETZ_TYPE_LANG_EN && FREETZ_TYPE_FIRMWARE_06_2X
 		default "FRITZ.Box_7490.en-de-es-it-fr-pl.113.06.53.image"              if FREETZ_TYPE_FIRMWARE_FINAL && FREETZ_TYPE_7490 && FREETZ_TYPE_LANG_EN && FREETZ_TYPE_FIRMWARE_06_5X
 		default "FRITZ.Box_7490.en-de-es-it-fr-pl.113.06.84.image"              if FREETZ_TYPE_FIRMWARE_FINAL && FREETZ_TYPE_7490 && FREETZ_TYPE_LANG_EN && FREETZ_TYPE_FIRMWARE_06_8X
 		default "FRITZ.Box_7490.en-de-es-it-fr-pl.113.07.01.image"              if FREETZ_TYPE_FIRMWARE_FINAL && FREETZ_TYPE_7490 && FREETZ_TYPE_LANG_EN && FREETZ_TYPE_FIRMWARE_07_0X
-		default "FRITZ.Box_7490_Labor.113.07.08-64272.image"                    if FREETZ_TYPE_FIRMWARE_LABOR_BETA && FREETZ_TYPE_7490
+		default "FRITZ.Box_7490_Labor.113.07.08-64370.image"                    if FREETZ_TYPE_FIRMWARE_LABOR_BETA && FREETZ_TYPE_7490
 		#
 		default "FRITZ.Box_7530.164.07.02.image"                                if FREETZ_TYPE_FIRMWARE_FINAL && FREETZ_TYPE_7530 && FREETZ_TYPE_LANG_DE && FREETZ_TYPE_FIRMWARE_07_0X
 		default "FRITZ.Box_7530.en-de-es-it-fr-pl-nl.164.07.02.image"           if FREETZ_TYPE_FIRMWARE_FINAL && FREETZ_TYPE_7530 && FREETZ_TYPE_LANG_EN && FREETZ_TYPE_FIRMWARE_07_0X
 		#
 		default "FRITZ.Box_7560.149.06.53.image"                                if FREETZ_TYPE_FIRMWARE_FINAL && FREETZ_TYPE_7560 && FREETZ_TYPE_LANG_DE && FREETZ_TYPE_FIRMWARE_06_5X
@@ -866,22 +866,22 @@ menu "Override options"
 		default "f64d17124ba354427a1710d12b35e00f"  if FREETZ_TYPE_W501V
 
 	config FREETZ_DL_SOURCE_CONTAINER
 		string "Downloaded container file"              if FREETZ_DL_OVERRIDE
 		#
-		default "fritzbox-7490-labor-64272.zip"         if FREETZ_TYPE_FIRMWARE_LABOR_BETA && FREETZ_TYPE_7490
-		default "fritzbox-7590-labor-64273.zip"         if FREETZ_TYPE_FIRMWARE_LABOR_BETA && FREETZ_TYPE_7590
+		default "fritzbox-7490-labor-64370.zip"         if FREETZ_TYPE_FIRMWARE_LABOR_BETA && FREETZ_TYPE_7490
+		default "fritzbox-7590-labor-64369.zip"         if FREETZ_TYPE_FIRMWARE_LABOR_BETA && FREETZ_TYPE_7590
 		#
 		default "fritzbox-6840-labor-61107.zip"         if FREETZ_TYPE_FIRMWARE_LABOR_MAINTENANCE && FREETZ_TYPE_6840
 		#
 		default ""
 
 	config FREETZ_DL_SOURCE_CONTAINER_MD5
 		string "MD5 checksum for container"             if FREETZ_DL_OVERRIDE
 		#
-		default "8d3a306ae634ec2edc060764dc9fa513"      if FREETZ_TYPE_FIRMWARE_LABOR_BETA && FREETZ_TYPE_7490
-		default "15a8aae14d1d1f089018941565417e25"      if FREETZ_TYPE_FIRMWARE_LABOR_BETA && FREETZ_TYPE_7590
+		default "06c6e2256aff1faf29226718e41b7d9e"      if FREETZ_TYPE_FIRMWARE_LABOR_BETA && FREETZ_TYPE_7490
+		default "5973577421270a8094c3453b7d8edfef"      if FREETZ_TYPE_FIRMWARE_LABOR_BETA && FREETZ_TYPE_7590
 		#
 		default "deb859e52537a38c25184c699011a4b8"      if FREETZ_TYPE_FIRMWARE_LABOR_MAINTENANCE && FREETZ_TYPE_6840
 		#
 		default ""
 

@@ -253,11 +253,11 @@ Currently supported boxes and firmwares
 	* 113.07.01 rev61077
 	* 113.06.31 rev32500 (International, Belgian release)
 	* 113.06.53 rev38152 (International, Belgian release)
 	* 113.06.84 rev48533 (International)
 	* 113.07.01 rev61484 (International)
-	* 113.07.08 rev64272 (Labor)
+	* 113.07.08 rev64370 (Labor)
 
 ====== Fritz!Box Fon WLAN (75xx) ======
 
 * Fritz!Box Fon WLAN 7530 (EXPERIMENTAL)
 	* 164.07.02 rev62311
@@ -284,11 +284,11 @@ Currently supported boxes and firmwares
 	* 154.06.86 rev45533
 	* 154.06.92 rev47566
 	* 154.07.01 rev61378
 	* 154.06.84 rev57902 (International, Italian release)
 	* 154.07.01 rev61989 (International)
-	* 154.07.08 rev64273 (Labor)
+	* 154.07.08 rev64369 (Labor)
 
 ====== T-Com derivates ======
 
 * Eumex 300IP
 	* Alien Fritz!Box Fon

BlackStar999, 2018-12-22:
Hoffe hier richtig eingestellt

PeterPawn, 2018-12-22:
Besser ist es, Du erstellst Dir einen Fork des Freetz-Repos, nimmst darin die notwendigen Änderungen vor und dann macht man daraus wieder einen "Pull Request", der - im besten Falle - nur noch "per Knopfdruck" übernommen werden muß.

Es wird sich vermutlich niemand hinstellen und aus dem Text im ersten Beitrag von Hand ein "diff"-File wiederherstellen, was dann auch erst einmal auf die (eigene) lokale Kopie des Repos angewendet werden müßte, bevor man diese Änderungen dann nach "origin" (oder "upstream", je nachdem, wie man selbst zum "oiginalen" Repo steht, wobei der Name "origin" eine Standardvorgabe von "git" ist und "upstream" auch nur einer "Konvention" entspricht) publizieren kann.

Das bessere Vorgehen ist (grob) z.B. hier beschrieben: https://guides.github.com/activities/forking/ - da muß dann jemand mit Schreibrechten tatsächlich nur noch einen Button im GitHub-Webinterface drücken, um einen Pull-Request (unverändert) zu übernehmen und außerdem kann man im Rahmen des PR dann noch Änderungen "anfordern" bzw. sich über Sinn und Unsinn des gesamten PRs austauschen ... und zwar so, daß daran "die Öffentlichkeit" auch teilnehmen kann.

olistudent, 2018-12-22:
Hallo BlackStar999. Versuchst du das bitte mal so wie von Peter beschrieben. Ich mache dann hier mal zu...

closed 2018-12-22 by olistudent