Freetz/freetz #63: Tvheadend on 6490

stareye72, 2019-01-07:
Hii,
i would like to have tvheadend for 6490 so to use the tuner, i could do package but no idea how to implement i am not  developer.
Thx in advance

Regards

stareye

derWeihnachtsmann, 2019-01-08:
Next you will need Kernel drivers for the TVcard you have. 
as the 6490 uses a MIPS CPU, this could be tricky to find/compile

stareye72, 2019-01-08:
The 6490 has own tuner and is arm not mips

derWeihnachtsmann, 2019-01-08:
I don't know what 6490 you have but mine is MIPS:

`root@fritz:/var/mod/root# cat /proc/cpuinfo
system type             : VR9
machine                 : Unknown
processor               : 0
cpu model               : MIPS 34Kc V5.6
BogoMIPS                : 331.77`

and which tuner does the 6490 have? DVB-S/T/C?

Please dont tell me you want to use the Tuner from the 6490Cable.. xD

stareye72, 2019-01-08:
dvb-c

stareye72, 2019-01-08:
maybe is enough to have the package could word as repeater for 6490 , you don't have to have raspberry pi.

PeterPawn, 2019-01-08:
The FRITZ!Box 6490 Cable uses a Puma6 SoC from Intel - it provides an ARM(v7) core (where the eCM (cable modem) functions are implemented) and an x86 (dual-)core, where all other functions (incl. DVB-C streaming) are running. 

Four tuners are available for DVB-C streaming, as long as this functionality isn't completely disabled, what's possible with newer firmware versions. 

These tuners are offered via a SAT>IP server, which was extended/updated in the latest firmware, too ... now it's usable with more SAT>IP clients, even with some Enigma2-based STBs. The SAT>IP server is discoverable via SSDP protocol (UPnP).

I'd bet, neither memory (the box has gotten 256 MB RAM at all, but that's splitted and each "architecture" can use only 128 MB RAM on its own) nor processor power of the two x86 cores are enough to make someone happy, while running Tvheadend on a FRITZ!Box 6490 Cable.

f-666, 2019-01-08:
The DVB-C tuners of the 6490 unfortunately do not provide standard Linux DVB device files. The user fesc has currently the best knowledge about what AVM is doing in this area on the box.

Neither RAM not CPU power should be a problem. I have VDR running on a 7490 with a USB DVB-C tuner for streaming and direct recording to an USB hard disk, and it works without problems. The 6490 should be more powerful.

PeterPawn, 2019-01-08:
You have *one* tuner (I'd assume, that your "a" means a single one) running on your 7490 and this device provides 256 MB of RAM.

When you're running (without hassle) four (or at least two) tuners in parallel from a 7430 device (this one has all needed components, from USB to the same VR9 chipset as processor) and it's still usable for its genuine tasks (router, smart-home, DECT base, IP-based telephony), I'll become impressed - but only then. 

And there's a difference (at least in my opinion) between "it works" (and I did not write, that it's impossible at all) and "I'm happy with this solution." ... simply remember the latency problems while streaming more than one HD channel simultaneously (maybe, these problems were nowadays solved or mitigated already).

From https://docs.tvheadend.org/basic_requirements/:

>  It will happily run in less than 1GB of RAM, and many people run it successfully on original Raspberry Pis with perhaps only 256MB of usable *free* memory. 

Maybe it's running on 128 MB of RAM (overall, but the FRITZ!OS system needs some memory by itself) with four tuners, too, while the original functions are still usable ... than I'll lose my bet. But that's some kind of "future-music" and needs a proof first.

olistudent, 2019-03-17:
This is a issue tracker, not a discussion forum. Please discuss feature requests in IPPF.

closed 2019-03-17 by olistudent