Freetz/freetz #70: voipd never use openvpn tun0 (7490 Fritz!OS 6.84)

Greg57070, 2019-02-04:
Hello,

I'm trying to setup a SIP connexion from Fritzbox to my Asterisk SIP Server using an OpenVPN tunnel.

I'm facing a strange issue : The VPN is up, routing works well (from Fritzbox and clients behind fritzbox).

With a tcpdump, voipd is trying to connect to my server through internet connection (not using tunnel)
A traceroute from Fritzbox to my Asterisk show right path (using tun0).

Any idea why voipd don't use the tunnel ?

Thank you all for the good work, keep up with it !
Best regards,
Gregory

feedzapper, 2019-02-04:
I'ts not a bug.
You need to set the right Interface in the "voip.cfg" for the number you are routing over the tunnel.
Change   sipiface = sipiface_tunnel;  in the  voip.cfg for the specific number.
Maybe you need also to do a "voipd -R" command after setting this to got registered.
Because the voipd daemon is started first on clean boot and the openvpn tunnel just AFTER the voipd daemon. The voipd daemon does not know from the openvpn tunnel...

Greg57070, 2019-02-04:
Thanks a lot for your help !
It's working now !

Best regards,
Gregory

closed 2019-02-04 by Greg57070