Freetz/freetz #73: Show trunk-version in menuconfig

WileC, 2019-02-16:
it would be nice, to see the actual version of the git pull in menuconfig

fda77, 2019-02-23:
Please add also the date of the hash
Very important is how long the short-hash is.  Some people like the full hash, other will complain when it is only 7, 8 or 9 chars long.  To avoid this, please make it configurabe.

er13, 2019-03-24:
Please define "the latest pulled"? Do you mean origin/HEAD, HEAD or something else?

WileC, 2019-03-24:
i think, like in #124, to see that in status line.

PeterPawn, 2019-03-25:
To simplify such a task, it's in my opinion a good idea to update [Kconfig](https://github.com/torvalds/linux/tree/master/scripts/kconfig) to a newer version. 

These newer versions have new macro capabilities and a new builtin function `shell` to generate config file contents (on the fly) from executed commands. 

You can get an impression on this topic from this article: https://lwn.net/Articles/756963/

Implementing this with an older version is surely possible, too ... but it's harder to find a solution, that doesn't tamper with other files under source control.

closed 2019-05-05 by er13