Freetz/freetz #76: get_ip ipv6

WileC, 2019-02-17:
extract IPv6 address, not only IPv4

fda77, 2019-02-17:
merge es die aus freetz-ng rüber ^^

f-666, 2019-02-18:
Sehr gerne. Wirklich.
Wo kann ich es finden?

fda77, 2019-02-20:
Bis die verwaltungspolitischen Dinge geklärt sind im reop auf https://boxmatrix.info/wiki/SVN-Server > Develop > svn
oder auch https://boxmatrix.info/wiki/Freetz_News - wird aber noch nicht automatisch für alle repos aktualisert

WileC, 2019-02-20:
stimmt.. auf dem freetz-org repo ist noch das alte get_ip...

WileC, 2019-02-22:
hab's mir mal angesehen.. also irgendwo wird beim freetz-ng die get_ip gepatcht beim build, aber ich finde nicht wirklich das patch-skript. Kann man nicht einfach die "get_ip" komplett ändern? Ohne das ganze über ein patch-skript zu lösen?

fda77, 2019-02-23:
Ne, da wird nichts beim build gepatcht

WileC, 2019-02-23:
müsste aber, da bei freetz-ng und freetz die ./make/mod/files/root/usr/bin/get_ip jeweils die gleiche datei ist, erst im ./build/modified/firmware/usr/bin/get_ip unterscheiden sie sich und es gibt in der "neueren" zwei "case $1" Abschnitte...

PeterPawn, 2019-02-23:
@WileC:
Hast Du beim Vergleich vielleicht den Stand in "freetz-ng/freetz-ng" (bis vor 2 Stunden gab es da ja noch einen Fork von "freetz") benutzt und dabei nicht mitbekommen, daß der mit dem Stand bis zum 17.02.2019 nichts mehr zu tun hatte?

fda77, 2019-02-23:
Wie mein Co-Kommentator schrieb, hast du wohl falsch verglichen. Um das in Zukunft auch bei anderen zu verhindern hatte ich das heut morgen etwas deutlicher hervorgehoben

PeterPawn, 2019-02-23:
> Um das in Zukunft auch bei anderen zu verhindern hatte ich das heut morgen etwas deutlicher hervorgehoben.
Diese Aufopferung im Interesse der Freetz-**Benutzer** (also der bewußte Verzicht auf einen eigenen Freetz-Fork im "freetz-ng"-Account, damit nur niemand das verwechseln kann) verdient in meinen Augen Bewunderung: :clap:

fda77, 2019-02-23:
Im Gegensatz zu mancher PPersonen
Dein Kommentar ist übrigens genau wie diese Person die ich meine

PeterPawn, 2019-02-24:
@WileC:
Wenn Du die Änderungen des "freetz-ng/freetz-ng"-Repos (bis einschließlich 15.02.2019) mit dem aktuellen Stand aus "freetz/freetz" vergleichen willst, verwende einfach diese URL dafür:

https://github.com/Freetz/freetz/compare/master...freetz-ng:978aea2a713c51d1864a558cc144cf9945b814af

Wenn Du den Stand der Datei mit den Änderungen bzgl. IPv6 sehen willst, findest Du ihn hier: 

https://github.com/Freetz/freetz/blob/b8ff421841654d51c7b563b9cdde70c070834d2d/make/mod/files/root/usr/bin/get_ip

Von dort kannst Du Dir über die Historie auch wieder die Änderungen (geziel an dieser Datei) anzeigen lassen.

Das dann einfach selbst zu ändern und als eigenen PR einzubringen, ist keine große Sache und dabei kannst Du dann vielleicht gleich noch die Funktion "helpmsg()" dahingehend anpassen, daß darin deutlich klargestellt wird, daß die Abfragen von IPv6-Adresse und -Präfix automatisch auch die "Methode" namens "--dsld" auswählen ... zumindest wenn man es so implementiert, wie es unter dem o.a. Link nachzulesen ist.

WileC, 2019-03-01:
Alternativ kann die IPv6-Adresse mittels **showdsldstat | sed -rn 's/.* IPv6: address ([^/]*).*/\1/p'** ausgegeben oder umgeleitet werden.

fda77, 2019-03-01:
@WileC: Es gibt 1000 Möglichkeiten die IP zu ermitteln. Schon witzig dass du exakt die gleiche Lösung wie ich gefunden hast

WileC, 2019-03-01:
Zufälle gibts, die glaubt man kaum... wäre schön gewesen, wenn man das get_ip-Skript "portieren" könnte...

fda77, 2019-03-01:
Den Status meiner Bemühungen siehst du hier: https://github.com/Freetz/freetz/issues/89

WileC, 2019-03-04:
PR unter #106

f-666, 2019-03-06:
erledigt

closed 2019-03-06 by f-666