Freetz/freetz #86: onlinechanged.cgi

WileC, 2019-02-22:
Also mir ist aufgefallen, wenn man mit einer Fritzbox 3490/6820lte sowohl IPv4 als auch IPv6 als Zugang nach Außen hat, wird bei einem reconnect zweimal die onlinechanged.cgi ausgeführt: einmal bei der Verbindung über IPv4 und nochmals beim [onlinechangedv6] ... das selbe skript. Vielleicht könnte man ja zukünftig das irgendwie aufteilen, in einen IPv4, IPv6-Teil, in den man die expliziten Anweisungen bei der jeweiligen Verbindung eintragen kann?

fda77, 2019-02-23:
onlinechanged unterscheidet zwischen den ip-versionen. Falls du was unter "allgemein" eingetragen hast, wird es halt bei beiden ausgeführt. Frag am bestin im schönen Forum wie das geht
(übrigens sind .cgi Datei teil Weboberfläche)

WileC, 2019-02-23:
ja, aber es gibt doch nur die drei Teile start, online, offline ... wenn ich was in "online" eintrage, wird der teil bei einem IPv4 als auch bei einem IPv6 connect ausgeführt...

fda77, 2019-02-23:
Ach, du benutzt das Paket "onlinechanged-cgi"! Oben schriebst du von einer .cgi [Datei] die ausgeführt wird
Das Package ist dann wohl noch nicht auf ipv6 vorbereitet, weiss ich nicht, hab es nie benutzt. Müsste dann halt jemand erweitern
Ich lege die onlinechanged Script-Datein einfach per ssh und ganz ohne das Package an, da gibt es extra v6 events

WileC, 2019-02-24:
Und wo legst du sie genau ab, dass die Skripte nur bei IPv4 oder nur bei IPv6 ausgeführt werden?

fda77, 2019-02-24:
Skripte in Verzeichnis /tmp/flash/onlinechanged/ , Dateiname ist egal, +x nicht vergessen
Wie die Events genau heissen muss du irgendwo nachlesen, hab seit 2 Jahren kein DualStack mehr. start/stop ists aber nicht!

WileC, 2019-02-25:
Stimmt... eher dann **online** / **offline** ... aber vielleicht kannst du ja nochmal nachsehen ... bei mir sieht's im **syslog** so aus:
`Feb 18 08:30:14 router user.notice ONLINECHANGED[646]: [online] approved
Feb 18 08:30:14 router user.notice ONLINECHANGED[646]: [online] executing /etc/onlinechanged/00-get_ip
Feb 18 08:30:14 router user.notice ONLINECHANGED[646]: [online] executing /etc/onlinechanged/20-opendd
Feb 18 08:30:14 router user.notice ONLINECHANGED[646]: [online] executing /etc/onlinechanged/webdav_net
Feb 18 08:30:14 router kern.warn kernel: [1373777.540000][0][0]system-load 3 loadavg 1.94 1.77 1.68 - 109 tasks:18 % curr:avmike(3 %)max:avmike(3 %, pid:639) pgstat: sum=62887 free=21091 slab=5135 alloc=1103/s fault=3013/s ai_sys:0.07/s 0x805cbd80 pndisc_constructor+0x
Feb 18 08:30:14 router user.notice ONLINECHANGED[646]: [online]  * webdav_net online status_onlinev4
Feb 18 08:30:14 router kern.info kernel: [1373777.760000][1]kdsld: flushing internet sessions
Feb 18 08:30:15 router kern.warn kernel: [1373778.260000][1][1]system-load 2 loadavg 1.94 1.77 1.68 - 113 tasks:3 % curr:luacgi(1 %)max:luacgi(1 %, pid:674) pgstat: sum=63025 free=20857 slab=5127 alloc=61/s fault=199/s ai_sys:0.82/s 0x8161c7cc find_option+0x2c/0xc0 [kd
Feb 18 08:30:15 router user.notice ONLINECHANGED[677]: [onlineipv6] waiting
Feb 18 08:30:15 router user.notice ONLINECHANGED[646]: [online] executing /tmp/flash/onlinechanged/onlinechanged-cgi
Feb 18 08:30:15 router user.notice onlinechanged: +++ DYNDNS-Update
Feb 18 08:30:15 router kern.err kernel: [1373778.910000][0][ppe_setup_qos_write] avmwanqueuesetrate 1 1000000
Feb 18 08:30:18 router user.notice curl: good 93.230.209.7,2003:00c1:37ff:0851:ca0e:14ff:fe18:1f10
Feb 18 08:30:18 router user.notice curl: good 93.230.209.7,2003:00c1:37ff:0851:ca0e:14ff:fe18:1f10
Feb 18 08:30:18 router user.notice ONLINECHANGED[646]: [online] finished
Feb 18 08:30:19 router user.notice ONLINECHANGED[677]: [onlineipv6] approved
Feb 18 08:30:19 router user.notice ONLINECHANGED[677]: [onlineipv6] executing /etc/onlinechanged/00-get_ip
Feb 18 08:30:19 router user.notice ONLINECHANGED[677]: [onlineipv6] executing /etc/onlinechanged/20-opendd
Feb 18 08:30:19 router user.notice ONLINECHANGED[677]: [onlineipv6] executing /etc/onlinechanged/webdav_net
Feb 18 08:30:19 router user.notice ONLINECHANGED[677]: [onlineipv6]  * webdav_net onlineipv6 status_onlinev4v6
Feb 18 08:30:20 router user.notice ONLINECHANGED[677]: [onlineipv6] executing /tmp/flash/onlinechanged/onlinechanged-cgi
Feb 18 08:30:20 router daemon.info chronyd[760]: chronyd version 3.2 starting (+CMDMON +NTP +REFCLOCK +RTC +PRIVDROP -SCFILTER -SECHASH -SIGND +ASYNCDNS +IPV6 -DEBUG)
Feb 18 08:30:20 router daemon.warn chronyd[760]: commandkey directive is no longer supported
Feb 18 08:30:20 router user.notice onlinechanged: +++ DYNDNS-Update
Feb 18 08:30:20 router daemon.info chronyd[760]: Frequency 0.369 +/- 0.021 ppm read from /var/tmp/chrony.drift
Feb 18 08:30:22 router user.notice curl: good 93.230.209.7,2003:00c1:37ff:0851:ca0e:14ff:fe18:1f10
Feb 18 08:30:22 router user.notice curl: good 93.230.209.7,2003:00c1:37ff:0851:ca0e:14ff:fe18:1f10
Feb 18 08:30:22 router user.notice ONLINECHANGED[677]: [onlineipv6] finished
Feb 18 08:30:24 router daemon.info chronyd[760]: Selected source 2003:2:2:140:194:25:134:196
`

Also wird einmal **online** und einmal **onlineipv6** gestartet... Doch beide führen das Verzeichnis /etc/onlinechanged gleichermaßen aus?! ...

fda77, 2019-02-25:
Ja genau. Anhand des Parameters der aufgerufenen Scripte sieht man den auslösenden Event, zb "onlineipv6" für ipv6.  Das prüft man dann mit if oder case
Wenn du onlinechanged-cgi reparieren willst, in /etc/onlinechanged/ hat das package bestimmt eine Datei. Dort muss man die "ipv6" irgendwie dazumachen

WileC, 2019-03-01:
also habs "ermittelt": der "onlinechanged" übergibt als Parameter 1 ($1) folgende Werte:
online, onlineipv6, offline, offlineipv6. 

Durch die Case-Abfrage "online*)" wird natürlich einmal der Status "online" und dann nochmal der Status "onlineipv6" ausgeführt... daher habe ich das insoweit "online)" abgeändert, dass der Skriptteil lediglich bei einer IPv4-Verbindung ausgeführt wird...

fda77, 2019-03-01:
Prima, dann weisst du ja nun wie du onlinechanged-cgi anpassen musst

PeterPawn, 2019-03-01:
> einmal der Status "online" und dann nochmal der Status "onlineipv6" ausgeführt

Das war sogar "Absicht" bei der letzten Änderung dort:

https://github.com/Freetz/freetz/commits/master/make/onlinechanged-cgi/files/root/etc/default.onlinechanged/onlinechanged-cgi

Diese Datei ist ja nur ein **Vorschlag**, was der Benutzer selbst für die verschiedenen Aktionen des `onlinechanged` von AVM (denn dort klinkt sich Freetz nur ein und durchsucht mehr Verzeichnisse nach passenden Dateien, als AVM es tut ... darunter auch das Verzeichnis, wo die Datei vom "onlinechanged-cgi" gespeichert wird) ausführen lassen könnte. 

Bei der letzten Änderung wurden leider die "neuen" Aktionen von AVM-Seite (die ja auch erst dazukamen, als die IPv6-Unterstützung Einzug hielt) nicht in den Kommentaren für die https://github.com/Freetz/freetz/blob/master/make/onlinechanged-cgi/files/root/etc/default.onlinechanged/onlinechanged.def noch ergänzt.

Das wäre doch eine nette, kleine Aufgabe für jemanden, der sich mal als "contributor" versuchen will. Das kann man alles auch innerhalb des GitHub-GUI ändern in dieser Datei und daraus dann einen PR machen (nachdem man das mal schnell selbst getestet hat, daß die Änderung syntaktisch korrekt war - oder man verläßt sich bei so einer minimalen Änderung auch auf den "reviewer") ... dafür braucht es nicht mal das "große Besteck".

WileC, 2019-03-01:
und wie kann ich nun diese Datei umändern? was brauch ich dazu?

PeterPawn, 2019-03-01:
Meinst Du jetzt als Änderung am Paket (also Fork, Ändern, PR) oder doch eher, wie Du das für Deine Box machst?

WileC, 2019-03-01:
naja.. ich ändere das bei mir .. oder hier online.. und jeder der ein **git pull** macht, kann das dann auschecken.

PeterPawn, 2019-03-01:
Verstehe ich gerade nicht so richtig ... willst Du jetzt Deine Datei ändern oder meinen Vorschlag aufgreifen und den Kommentar in der CGI-Seite (für andere) so anpassen, daß darin nicht länger nur "start, online, offline" als mögliche Parameter stehen, sondern auch die beiden IPv6-Aufrufe ("onlineipv6" und "offlineipv6"), die AVM später in die Firmware eingebaut hat?

Bei letzteren gehst Du einfach in Deinen GitHub-Account (wo Du ja schon bist, wenn Du hier schreibst), klickst auf das "Fork" oben rechts in der "freetz/freetz"-Seite und kriegst damit Deine eigene Kopie des Freetz-Repos.

Darin kannst Du jetzt erst einmal einen neuen Branch anlegen (über den Button, wo normalerweise "Branch: master" steht - da kann man auch einen neuen anlegen lassen) und in diesem kannst Du dann (alles immer noch mit dem GitHub-GUI) die Datei "onlinechanged.def" entsprechend editieren.

Bist Du damit fertig, kannst Du per Mausklick aus Deinen Branch einen Pull-Request generieren, den dann nur noch jemand (auch wieder mit einem Mausklick) in das Master-Repo als Korrektur übernehmen kann. Wurde das dann integriert, kannst Du den Branch wieder löschen, Deinen Fork aktualisieren (da kommt dann ja die Änderung wieder mit) und hast dann für die nächste Änderung das Repo schon in Deinem Account.

WileC, 2019-03-02:
Und wie kann ich von freetz **nur** den **master-branch** forken .. ohne die ganzen "stable-branches" ??

PeterPawn, 2019-03-02:
Das geht - anders als bei SVN - bei Git nicht (jedenfalls wüßte ich nicht, wie - denn es geht wahrscheinlich schon irgendwie, wenn auch mit hohem zusätzlichen Aufwand) - daher plädierte ich auch ursprünglich dafür, daß man die ganzen alten Branches aus dem SVN nicht hierher synchronisiert, sondern in ein gesondertes Repository als Archiv - sie sollen ja nicht einfach "weg sein".

Nur bringen sie in der "täglichen Arbeit" eben nichts ... man ignoriert sie am besten einfach. Sollte sich jemand zur "maintenance" im Repo entschließen und tatsächlich ein Archiv anlegen für die alten Branches, kann man diese allerdings auch wieder aus dem "Stamm-Repo" entfernen und sogar die dafür notwendigen Objekte im "working tree" mittels "prune" wieder entfernen lassen. 

Das sorgt am Ende auch dafür, daß diese Branches die Datenmenge beim Klonen (was ja nun häufig vorkommt - sowohl bei "contributors" als auch beim Freetz-Benutzer, der mit einem "sauberen System" startet) nicht unnötig aufblähen. 

Wobei Git hier auch schon um Längen effektiver ist bei der Nutzung des lokalen Speichers (und beim Übertragen von Daten) als SVN ... dort bringt ein "frischer" Klon aller Branches knapp 700 MB auf die Waage, wenn ich mich richtig erinnere und bei einem aktuellen Git-Klon des Freetz-Repos sind es gerade mal 76 MB.

kriegaex, 2019-03-02:
Leute, lernt endlich Git! Peter hat es ja schon beschrieben: Ein Klon des gesamten Repos ist immer noch einfacher und schneller, als bei jedem Branch-Wechsel eine komplette Kopie des Arbeitsverzeichnisses von einem SVN-Server zu holen. Einmal geklont, fressen die Branches doch kein Brot mehr, man kann aber jederzeit etwas nachschlagen, vergleichen, branchen, mergen, rebasen, committen etc., ohne auch nur ein einziges Byte über die Leitung zu schicken. Nur für Fetch/Pull und Push braucht man überhaupt eine Netzwerkverbindung. Das bedeutet: Branchwechsel, Merges etc. dauern in Git Millisekunden bis Sekunden, nicht wie oft bei SVN Sekunden bis Minuten, und zwar bei jedem einzelnen SVN Update oder Commit. Was Ihr glaubt, hier an vorweggenommener Optimierung vornehmen zu müssen durch Löschen alter Branches, ist schlichtweg unnötig. Projekthistorie ist bei Git kein unnötiger Ballast wie bei SVN sondern eine wertvolle Wissensquelle.

Ich weiß, ich bin emotional bei dem Thema, aber Ihr seid es ja auch. Eure Ängste und Vorbehalte sind nämlich genau das: emotional, aber rational vollkommen unbegründet. Lernt Git, erkennt, daß Eure SVN-Gewohnheiten irrelevant und Eure Ängste unbegründet sind und genießt einfach ein superschnelles, effizientes SCM-System.

PeterPawn, 2019-03-02:
> Löschen alter Branches, ist schlichtweg unnötig.

Ich will nur noch mal festhalten, daß das (vermutlich) kein Plädoyer dafür ist, JEDEN Branch auf ewig zu behalten (hoffe ich zumindest, wenn nicht, wird @kriegaex es gleich noch einmal klarstellen).

Ein Branch hat genau so lange eine "Daseinsberechtigung", wie er tatsächlich eine Verzweigung repräsentiert. Gibt es in einem "Branch" andere Commits (das kann schon dadurch der Fall sein, daß bei einem "Backport" eines neuen Commits auf einen früheren Stand ja eine abweichende Historie von diesem Stand an entsteht), ist der immer "_N_ commits ahead of ...".

Ein Branch, der per Pull-Request (insgesamt) in das "Stamm-Repo" übernommen wurde, kann (muss/sollte) durchaus auch wieder gelöscht werden ... er repräsentiert nämlich nach dem Rebase für das eigene Repo (wo dann die Änderungen aus dem Stamm-Repo "eingepreist" werden) nicht länger einen abweichenden Stand.

Bei der Frage, ob man die alten SVN-Branches nun übernehmen mußte oder nicht, geht/ging es mir auch nicht wirklich um Datenmengen u.ä. ... eher um die Übersicht. 

Soweit ich weiß (gerne lasse ich mich korrigieren), sind festzustellende Abweichungen zwischen "master" und den alten Branches (vor 2014 sagen wir mal) darin begründet, daß tatsächlich ab und an mal eine wichtige Änderung noch auf einen früheren Stand (wo der Branch dann ansetzte) portiert wurde. 

Wenn ich nichts wirklich Wichtiges in den letzten Jahren übersehen habe, gab es solche Backports aber auch schon seit Jahren nicht mehr. Der letzte Commit in Freetz 2.0 ist zwar aus 11/2015, aber wenn man die letzten drei abzieht, ergibt sich schon wieder ein anderes Bild und das sagt dann, daß selbst dieser (neueste) Branch fast 5 Jahre hinterher ist.

Mithin suggerieren diese Branches nur, daß es weiterhin "gepflegte" Zweige für die verschiedenen Stände gäbe und man - meinetwegen mit einem Update für die aktuelle OpenSSL-Version, das man auf diesen Stand portiert - problemlos auch mit einem dieser älteren Branches arbeiten könnte. 

Das genau geht aber (zumindest mit gutem Gewissen) schon lange nicht mehr ... daher mein Standpunkt: Archivieren und vergessen. Der Aspekt des "Lernens" aus alternativen Lösungen in anderen Branches, spielt m.E. hier keine wirklich wichtige Rolle, weil es diese Alternativen schlicht nicht gibt  ... die Commits, mit denen ein Branch im "alten Freetz" dem "master" voraus ist, sollten sich alle auch in ebendiesem "master" wiederfinden lassen - halt nur aufgrund der abweichenden Historie mit anderen Hash-Werten und daher "nicht zu erkennen".

Wenn man **echte** Branches im Projekt hätte (spinnen wir mal ein "Freetz 3.0" als Release/Tag herbei) und es kommt ein PR herein, dessen Inhalt auf diesen älteren Punkt portiert werden sollte (dann wird aus dem "Ich war hier.", was da in den "working tree" geritzt wurde beim Taggen, eine andere Historie ab diesem Punkt und damit ein "branch"), dann muß der eben sowohl mit dem "master" als auch mit dem älteren Branch gemischt werden - aber auch der Branch für diesen PR hat sich damit dann "erledigt" und ist in die Historie (in beiden Branches) eingegangen. Solche Branches (in eigenen Repos) aufzuheben, macht also nur in eher seltenen Fällen einen Sinn, zumal man sie i.d.R. ohnehin nach jeder Änderung am "parent" wieder mit "rebase" auf den aktuellen Stand bringen müßte.

Im Git verschwinden mit dem Löschen eines Branches auch die Commits in diesem nicht automatisch ... sie sind nur nicht mehr erreichbar, weil sie in keinem Verlauf mehr verzeichnet sind.

PS: Beim Korrekturlesen habe ich dann bemerkt, daß "branch" und "tag" und "parent repository" oder "project main repository" (aka "origin" bzw. "upstream" bei einem Fork) nicht so leicht auseinanderzuhalten sind im Deutschen. 

Ein "tag" ist per se eben nur die "Wegmarkierung" oder ein sprechender Name für einen bestimmten Commit (anstelle des Hash-Wertes, mit dem man wirklich jeden Commit "finden" kann) ... an einem solchen kann sich der Weg aber auch verzweigen und es geht in verschiedene Richtungen weiter. 

Der "branch" ist dann eigentlich nicht der Punkt, an dem die Verzweigung ansetzt, sondern der "Endpunkt", der bei dieser alternativen Form der Realität erreicht wurde - also der "aktuelle Stand" in diesem Zweig. 

Ein Branch im "project repository" sollte schon etwas mehr Sinn haben, als nur die für einen Fix notwendigen Änderungen zu repräsentieren ... dafür ist ein Branch in einem privaten Repository aber praktisch wieder unumgänglich, wenn man dem Maintainer nicht das Hantieren mit "patch" und Diff-Files zumuten will.

Ein Beispiel für einen sinnvollen Branch im Freetz-Repo (als echter "feature branch") wäre es eben gewesen, wenn man die Unterstützung der anderen Plattformen (ARM, ATOM) in solchermaßen getrennten Zweigen entwickelt hätte ... da macht dann ein solcher Branch ebenso Sinn, wie bei regelmäßigen Releases (die bis zu weiteren, alternativen Änderungen ab diesem Punkt ja auch wieder nur "tags" sind, halt mit "annotations" und mit ein paar zusätzlichen Funktionen im GitHub), für die später noch Fehlerkorrekturen vorgenommen werden.

Das mit dem "staging" im Repo von @f-666 sieht ja schon gut aus ... ich würde mir so einen "staging"-Branch auch im Freetz-Repo wünschen für "neue Sachen". Für die "Freetz-Benutzer" sollte m.E. der "master"-Branch da sein, der aber auch keine ungetesteten Änderungen enthalten sollte. 

[... cut ...]
Ende des Jahres kommt z.B. auch auf Freetz die komplette Umstellung auf OpenSSL 1.1.1 zu, denn der Support für 1.0.2 endet am 31.12.2019 - der ganze damit verbundene Aufwand (keine Ahnung, wieviele Pakete OpenSSL 1.1.x noch gar nicht unterstützen .- das API hat sich deutlich verändert) ist enorm und die Änderungen garantiert umfangreich. Das sollte man auch besser in einem "Freetz-Branch" angehen und anschließend auf einen Schlag und nicht tröpfchenweise in den "master"-Zweig integrieren ... um mal ein Beispiel für weitreichende Änderungen, die in einer "Benutzer-Version" erst einmal noch nichts zu suchen haben, zu nennen.

kriegaex, 2019-03-03:
> Ich will nur noch mal festhalten, daß das (vermutlich) kein Plädoyer dafür ist, JEDEN Branch auf ewig zu behalten (hoffe ich zumindest, wenn nicht, wird @kriegaex es gleich noch einmal [...]

Danke für die Klarstellung, Peter. Ja, in dieser pauschal verkürzten Form war meine Aussage mißverständlich und sogar falsch, wenn man sie auf alle Branches bezieht.

Da ich ja für viele, kurzlebige (s.o.) Story bzw. Feature Branches plädiere, würde man ja irgendwann in der IDE oder an der Konsole den Wald vor lauter Bäumen nicht mehr sehen, wenn man sich die Branches auflisten läßt. Was ich meine, sind Branches, die noch "unmerged" sind, d.h. die noch von der Hauptlinie (einfaches Feature Branch Modell) oder vom Entwicklungs-Branch (GitFlow) abweichen, also noch nicht gemergte Commits beinhalten und nicht obsolet sind. Maintenance Branches, falls man so etwas braucht, sind auch ein Beispiel.

Story Branches sollten nach Übernahme in die höher integrative Entwicklungsline (s.o.) bzw. kurz danach gelöscht werden, sobald klar ist, daß es keine Probleme mehr gibt und man vermutlich nichts mehr an diesem Feature oder Bugfix machen wird.

Eine Sache mache ich aber immer in meinen Coaching-Projekten: Im Upstream-Repo wird automatisch durch einen Push-Hook der letzte Stand des zu löschenden Branches getaggt, und zwar in etwa so, daß z.B. ein Feature Branch zum Jira-Ticket `FR-123` Namens `feature/FR-123` vor dem Löschen getaggt wird als `old_branches/feature/FR-123`. Wieso machen wir das so? Falls man irgendwann man nachschauen möchte, was genau zum Ticket `FR-123` committet wurde, weil man es nicht mehr versteht, weil man doch noch einen Fehler gefunden hat oder weshalb auch immer, kann man sehr einfach den gelöschten Branch wiederbeleben, indem man das Tag auscheckt, an der Stelle den Branch wieder anlegt, ihn auffrischt vom Master her und weiter arbeitet. Selbst wenn man nicht arbeitet sondern nur lesen und verstehen will, kann man sich leicht die Historie des gelöschten Branches anschauen, einen Diff machen usw.

_Rant of the day:_ Wie man mit Git allgemein sinnvoll arbeitet und dann im zweiten Schritt so ein Feature Branch Modell aufsetzt, bringe ich meinen Coachees normalerweise in zwei Git Workshops bei. Der erste dreht sich um Git-Grundlagen und beinhaltet auch einen kleinen Deep Dive, damit man versteht, was Git hinter den Kulissen macht und wieso das so viel besser ist als das, was SVN u.a. machen, der zweite Workshop erklärt den Story/Feature Branch Workflow für das Projekt und die Alternative "full-blown GitFlow". Diese Workshops hatte ich vor vielen Jahren bereits dem Freetz-Team angeboten, aber wegen damaliger Widerstände gegen Git - Motto: "Was ich nicht verstehe, davor habe ich Angst, also fasse ich es erst gar nicht an." Oder: "Was der Bauer nicht kennt, ..." - kam das nie zustande. Meine Kunden zahlen hohe Preise für solche Entwickler-Workshops, für Freetz hätte ich es kostenlos gemacht. Statt die Chance zu nutzen, Git **wirklich** zu verstehen, etwas für die eigene Entwicklerkarriere zu tun und nebenbei auch in Freetz einfacher und sauberer zu arbeiten, verlief die Initiative wegen Desinteresse, teils sogar Widerstand, im Sande.

> Ein Branch, der per Pull-Request (insgesamt) in das "Stamm-Repo" übernommen wurde, kann (muss/sollte) durchaus auch wieder gelöscht werden ... er repräsentiert nämlich nach dem Rebase für das eigene [...]

Einfache Antwort: Übernahme ja, Projekthistorie ist mir immer sehr wichtig. Nicht mehr gepflegte Branches zumachen und dann taggen, wie oben beschrieben, um sie auffindbar zu machen und nicht dem Git Garbage Collector zum Opfer fallen zu lassen. Die Größe des Repos ist mir nicht so wichtig. Sie hält sich auch bei umfangreichen Historien normalerweise stark in Grenzen, solange man nicht Binaries committet. Es gibt auch Tools, um solche (nicht mehr benötigten, weil bereinigten) Binaries aus dem Git Object Store zu löschen, aber das nur nebenbei.

> Mithin suggerieren diese Branches nur, daß es weiterhin "gepflegte" Zweige für die verschiedenen Stände gäbe und man - meinetwegen mit einem Update für die aktuelle OpenSSL-Version, das man auf diesen Stand portiert - problemlos auch mit einem dieser älteren Branches arbeiten könnte.

Wenn dem so ist, dann taggen und weg damit.

> Wenn man **echte** Branches im Projekt hätte (spinnen wir mal ein "Freetz 3.0" als Release/Tag herbei) und es kommt ein PR herein, (...)

Genau dafür braucht man Tags, aber meine Teams arbeiten witziger Weise kaum mit Release Branches, dafür exzessiv mit Story Branches. Die sind in meinen Augen viel wichtiger im Alltag. Übrigens rebasen wir auch selten sondern nutzen in der Regel einen echten Merge vom Hauptzweig, um die untergeordneten Branches aufzufrischen, da der Rebase die Commit-Historie neu schreibt und bei bereits gepushten Branches ist das ein No-Go. Gepusht wird in meinen Projekten ständig, da Entwickler in der Regel gemeinsam an Stories/Features arbeiten, nicht als Einzelkämpfer. Wie gesagt, ich mag keine Code Ownership.

> Im Git verschwinden mit dem Löschen eines Branches auch die Commits in diesem nicht automatisch ... sie sind nur nicht mehr erreichbar, weil sie in keinem Verlauf mehr verzeichnet sind.

Nicht mehr erreichbar ist doof, daher das Taggen vor dem Löschen, s.o.

Auf den Rest antworte ich nicht mehr im Einzelnen, weil es Energieverschwendung wäre. Sollte das Freetz-Team - wer auch immer das gerade ist - irgendwann mal von sich aus auf mich zu kommen, die zwei o.g. Git Wor [... cut ...] ilnehmen und man es dann hinterher auch so nutzt, wie ich es dort beschreiben würde, wäre ich auch heute noch (bzw. wieder) dazu bereit, die Zeit zu opfern und Euch das zu zeigen und zu erklären. So wie jetzt an Freetz gearbeitet wird, egal ob von nur einem oder 20 Maintainers/Contributors, habe ich keine Lust, mich weiter aktiv in solche Diskussionen einzubringen, denn egal, wieviel ich hier schreibe, es ist immer verkürzt und damit teilweise un-/mißverständlich, es bekommt nicht jeder mit etc.

closed 2019-03-04 by WileC