Freetz/freetz #88: Download xz-5.2.4 failed

WileC, 2019-02-23:
`--2019-02-23 18:13:08--  http://tukaani.org/xz/xz-5.2.4.tar.xz
Auflösen des Hostnamens tukaani.org (tukaani.org) … 84.34.147.45
Verbindungsaufbau zu tukaani.org (tukaani.org)|84.34.147.45|:80 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 301 Moved Permanently
Platz: https://tukaani.org/xz/xz-5.2.4.tar.xz [folgend]
--2019-02-23 18:13:08--  https://tukaani.org/xz/xz-5.2.4.tar.xz
Verbindungsaufbau zu tukaani.org (tukaani.org)|84.34.147.45|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 302 Found
Platz: https://sourceforge.net/projects/lzmautils/files/xz-5.2.4.tar.xz/download [folgend]
--2019-02-23 18:13:08--  https://sourceforge.net/projects/lzmautils/files/xz-5.2.4.tar.xz/download
Auflösen des Hostnamens sourceforge.net (sourceforge.net) … 216.105.38.13
Verbindungsaufbau zu sourceforge.net (sourceforge.net)|216.105.38.13|:443 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 302 Found
Platz: https://downloads.sourceforge.net/project/lzmautils/xz-5.2.4.tar.xz?r=&ts=1550941991&use_mirror=netcologne [folgend]
--2019-02-23 18:13:11--  https://downloads.sourceforge.net/project/lzmautils/xz-5.2.4.tar.xz?r=&ts=1550941991&use_mirror=netcologne
Auflösen des Hostnamens downloads.sourceforge.net (downloads.sourceforge.net) … 216.105.38.13
Verbindungsaufbau zu downloads.sourceforge.net (downloads.sourceforge.net)|216.105.38.13|:443 … verbunden.
Es ist nicht möglich, eine SSL-Verbindung herzustellen.
Download failed - "http://tukaani.org/xz/xz-5.2.4.tar.xz"  ->  error code 4

--2019-02-23 18:13:47--  http://freetz.wirsind.info/xz-5.2.4.tar.xz
Auflösen des Hostnamens freetz.wirsind.info (freetz.wirsind.info) … 188.165.115.52
Verbindungsaufbau zu freetz.wirsind.info (freetz.wirsind.info)|188.165.115.52|:80 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 404 Not Found
2019-02-23 18:13:47 FEHLER 404: Not Found.

Download failed - "http://freetz.wirsind.info/xz-5.2.4.tar.xz"  ->  error code 8

--2019-02-23 18:13:47--  http://freetz.magenbrot.net/xz-5.2.4.tar.xz
Auflösen des Hostnamens freetz.magenbrot.net (freetz.magenbrot.net) … 31.172.113.113
Verbindungsaufbau zu freetz.magenbrot.net (freetz.magenbrot.net)|31.172.113.113|:80 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 404 Not Found
2019-02-23 18:13:47 FEHLER 404: Not Found.

Download failed - "http://freetz.magenbrot.net/xz-5.2.4.tar.xz"  ->  error code 8

--2019-02-23 18:13:47--  http://freetz.3dfxatwork.de/xz-5.2.4.tar.xz
Auflösen des Hostnamens freetz.3dfxatwork.de (freetz.3dfxatwork.de) … 31.172.113.113
Verbindungsaufbau zu freetz.3dfxatwork.de (freetz.3dfxatwork.de)|31.172.113.113|:80 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 404 Not Found
2019-02-23 18:13:47 FEHLER 404: Not Found.

Download failed - "http://freetz.3dfxatwork.de/xz-5.2.4.tar.xz"  ->  error code 8
tools/make/lzma2-host/lzma2-host.mk:14: recipe for target 'dl/xz-5.2.4.tar.xz' failed
make: *** [dl/xz-5.2.4.tar.xz] Error 1`

f-666, 2019-02-24:
Ich vermute ein Problem auf dem Buildhost oder temporär bei Sourceforge:
`Es ist nicht möglich, eine SSL-Verbindung herzustellen.`

fda77, 2019-02-24:
Trag den boxmatrix oder deb mirror unter "User-defined download site" ein, die Datei liegt dort

WileC, 2019-02-24:
Und **was genau** müsste man wo eintragen?

WileC, 2019-02-24:
und ich meine der HTTP-Request 404 not found heisst, dass die Datei dort nicht mehr liegt, bzw. in der wget url so nicht vorhanden ist?!

fda77, 2019-02-24:
SF hat Probleme. Die Datei ist auf keinem freetz.org Mirror (404).
Es ist auch aktuelle niemand da der auf dem Mirror Schreibrechte hat.
Wann jemand da sein wird weiss man nicht

Wo? Der Eintrag im menuconfig heisst wie das in den Gänsefüsschen oben
Was
https://boxmatrix.info/hosted/freetz
ODER
http://freetz.digital-eliteboard.com/Teamserver/dl
Auf den einen hat Hippie Schreibzugriff, auf den anderen Gismotro. Ich auf keinen

Oder einfacher: Einen Fork nehmen wo die schon hinterlegt sind ;-)

olistudent, 2019-03-11:
Datei liegt jetzt auf dem Mirror.

closed 2019-03-11 by olistudent