Freetz/freetz #89: Freetz-NG mergen

fda77, 2019-02-24:
Da Unwahheiten darüber verbeitet werden wie das nun wirklich ist, möchte ich es hier nochmal erläutern:
 - Unter den Freetz Developern gab es Streit, die Wege trennten sich
 - Das letzte offizielle Statement der Freetz Developer via Mailingliste mir gegenüber ist: Wir brauchen nichts von dir, wir wollen nicht von dir
 - Solange das so ist sollte,  nicht von mir aus Freetz-NG hierher gemergt werden, ausser ich mache es selbst einen push request 
Umgekehrt merge ich auch nichts ungefragt aus Freetz oder sonst einem Fork

Diese Situation find ich nicht gut, und hab in den letzten Jahren versucht dies zu ändern.
Zuletzt mit Freetz-NG wo ich erfolglos versucht ein paar Aktive zusammenzubringen.
Man sollte dennoch darauf hinarbeitem dass es in Zukunft nur 1 "master" für Freetz gibt und alles zusammengeführt wird. Wie weiss ich nicht, falls jemand ne Idee hat her damit.

PS: Durch meinen Stalker bin ich nun aber nicht mehr im IPPF und mein Account dort ist inaktiv und steht zum löschen.
Wer möchte kann im Digital-Elite-Board im Freetz Bereich vorbeischauen

f-666, 2019-02-24:
Da ich nun Schreibrechte auf den freetz-master habe, kann ich PRs mergen.
Es scheint außer mir keiner etwas sonst im Repository zu tun.
Ich weiß nicht, wer die früheren Freetz Developer überhaupt sind.
Bisher lässt man mich gewähren.

Ich habe nur einen kleinen Teil der Commits im Freetz-NG Zweig angesehen und mir auch die Zeit genommen, sie im Detail zu durchdringen. Es waren aber einige coole Änderungen darunter (vielleicht auch mehr, wie gesagt, ich hatte keine Zeit).

Falls du PRs machst, werde ich sie mergen. Da ich, so frisch dabei, nichts kaputt machen will, würde ich mich besser fühlen, wenn wir mit den konservativeren Dingen anfangen würden.

Da ich Freetz gut finde und selbst benutze, fände ich auch ein einziges, zentrales Repo gut.
Mit GIT habe ich kein Problem, diese Stelle hier scheint mir geeignet.

fda77, 2019-02-24:
Also ich hätte ja gern ein offizielles Statement, um genau zu sein von den Admins hier.
Sonst sagt der eine wieder dies und das Gegenteil wird 30 Minuten später gemacht.

Git(Hub) find ich nicht so besonders, ist aber kein Problem für mich. (Davon unabhängig werde ich weiter zum SVN synchonisieren)
Bin aktuell bei r15222, komplett mergen geht leider nicht mehr da, inkompatibel.  Einzelne Commits sind jenachdem auch nicht möglich, wegen anderer Änderungen von mir vorher

Ich hatte Oliver vor ein paar Tagen schon vorgeschlagen, einen zusätzlichen Branch hier anzulegen. Dort würde ich mein Kram reinpacken und aus anderen Branches dazu "rebasen" was mir gefällt. (rebase erhält die Timeline im SVN)
Auf diesen einen Branch hätte nur ich Schreibzugriff

Was in anderen Branches passiert, kann mir somit egal sein. Den "master" könnte jemand (mir egal wer) entweder aus Branches zusammenmischen oder auch direkt reinmergen wie bisher

Bislang bekam ich noch keine Antwort von Oliver

MasterRoCcO, 2019-02-24:
In deiner r15222 Version lauft 831-extend-dsl_line_settings-speed.sh nicht meht

kriegaex, 2019-02-25:
> Also ich hätte ja gern ein offizielles Statement, um genau zu sein von den Admins hier.

Es gibt hier bislang 5 Entwickler mit Schreibzugriff. 4 davon haben auch die Admin-Rolle (der fünfte vielleicht versehentlich nicht, keine Ahnung), wir sind also nicht sonderlich restriktiv und vertrauen einander. So gesehen, ist es schwer, ein "offizielles" Statement zu bekommen, wir sind alle Individuen und es gibt keinen Abstimmungsprozeß. Ich selbst bin ja nicht mal aktiv als Entwickler, wie jeder weiß. Ich antworte auch nur, damit es jemand tut, und weil ich mich gut mit Git auskenne. Dein Ruf soll nicht ungehört verhallen.

> Sonst sagt der eine wieder dies und das Gegenteil wird 30 Minuten später gemacht.

Ich weiß nicht, mit wem Dir da was genau passiert ist. Hört sich schlimm an.

> Git(Hub) find ich nicht so besonders, ist aber kein Problem für mich. (Davon unabhängig werde ich weiter zum SVN synchonisieren)

Kannst Du zu Deinem Privatvergnügen gern machen. Ich empfehle trotzdem, mehr über Git zu lernen, wenn Du hier kollaborieren möchtest und Wert darauf legst, daß Früchte Deiner kreativen Arbeit mit ins Projekt kommen. Ich empfehle dieses Buch, das ich vor vielen Jahren mal in zwei Tagen komplett gelesen habe und das kostenlos online zu lesen ist:

[Scott Chacon - Pro Git](https://git-scm.com/book/de/v2)

Es gibt aber auch wesentlich kürzere Tutorials für den schnelleren (und oberflächlicheren) Einstieg.

> Bin aktuell bei r15222, komplett mergen geht leider nicht mehr da, inkompatibel. Einzelne Commits sind jenachdem auch nicht möglich, wegen anderer Änderungen von mir vorher

So läuft das eben manchmal bei Einzelkämpfer-Forks.

> Ich hatte Oliver vor ein paar Tagen schon vorgeschlagen, einen zusätzlichen Branch hier anzulegen. Dort würde ich mein Kram reinpacken und aus anderen Branches dazu "rebasen" was mir gefällt. (rebase erhält die Timeline im SVN)
>
> Auf diesen einen Branch hätte nur ich Schreibzugriff

Das wird nicht passieren. Branches mit beschränktem Zugriff sind zwar technisch möglich, aber bisher nicht gelebte Praxis in diesem Projekt oder in irgendeinem anderen, das ich kenne und welches mit Git arbeitet. Das ist klassische "SVN-Denke".

Es ist viel einfacher: Gehe auf die Hauptseite dieses Repositories, klicke auf "Fork" - und voilà! Du hast Dein eigenes, privates Repository, in dem Du beliebige Branches anlegen, mergen und rebasen kannst nach Herzenslust. Keiner kann Dir ins Handwerk pfuschen. Von dort aus kannst Du jederzeit Pull Requests gegen dieses Repository anlegen und losschicken. Ich denke, das ist genau, was Du möchtest. Nur solltest Du dafür, wie gesagt, mit Git(Hub) sauber zu arbeiten lernen, falls Du es nicht sowieso schon kannst (keine Ahnung, ehrlich). Ich empfehle dringend, sich von SVN zu verabschieden. Aber wie gesagt: Dein Privatvergnügen.

> Was in anderen Branches passiert, kann mir somit egal sein. Den "master" könnte jemand (mir egal wer) entweder aus Branches zusammenmischen oder auch direkt reinmergen wie bisher

Kann es Dir mit dem Fork auch. Dafür, daß ein PR gegen den aktuellen _master_ oder irgendeinen anderen in Freetz upstream (also hier) existierenden Branch sauber merge-bar ist, bist Du als Autor des PR verantwortlich, d.h. Du würdest Deinen Fork auch immer wieder aktualisieren und somit auf dem neuesten Stand halten. Die SVN "Merge Hell" will hier sicher keiner, und mit Git hat man die auch nicht, wenn man die Grundlagen versteht und sauber arbeitet.

> Bislang bekam ich noch keine Antwort von Oliver

Vermutlich ist er beschäftigt, so wie ich auch meistens.

Sag Bescheid, ob der Vorschlag annehmbar ist. Ich persönlich würde Beiträge zur Weiterentwicklung von Freetz, egal von wem sie stammen, gut finden, ohne Deine jemals gesehen zu haben. Die Aussage gilt allgemein. Was ich doof finde, sind Forks, die nicht reine technische Git-Forks mit dem Ziel, das Upstream-Repo mit PRs zu versorgen, sind, sondern eine Abspaltung von Freetz bedeuten. Aber das hier ist offene Software und daher kannst Du das machen, wenn es Dir beliebt. Erwarte nur nicht, daß es Jubelstürme hier auslöst.

fda77, 2019-02-25:
@MasterRoCcO: Bitte nicht hier, am besten im DEB. Hab auch mal Issues bei -ng aktiviert

@kriegaex 
Danke für die Antwort, klingt doch schonal gut!
Da es bislang keinen Grund gab mir zu misstrauen und wir uns schon lang kennen, schlage ich vor dass du mich als auch Admin einlädst. Umgekehrt weiss ich wo ich dran bin.

Zur Repo-Diskussion will ich nicht viel mehr als oben schon schreiben
> Git(Hub) find ich nicht so besonders, ist aber kein Problem für mich.
Ich commite momentan auf git(hub) und synchronisiere auf svn. Benutzer bevorzugen svn, weil man leicht erkennen kann welche Revision neuer ist. So einfach ist das ^^

Ich bezeichne mich selbst als git AnFänger. Hab zwar alles irgendwie hinbekommen was ich vorhatte, schliesse aber nicht aus dass mir mal was dummes passiert
Wie schnell man ein Buch lies hat keine Bedeutung, mit manchen ellenlangen Forenposts bin ich schon in 2 Sekunde fertig

Vorgehen: Also du bist komplett gegen Branches? Tags/Branches sind nur zum festhalten eines bestimmten Standes?
Entwicklungen sollen also in den Forks stattfinden? Es wird nur der "master" verwendet um zu mergen? (oder gleich committen?)
Soweit ich das mit git verstanden habe, ist das aber nicht das normale Vorgehen, sondern Branches wären das!

> So läuft das eben manchmal bei Einzelkämpfer-Forks.
Ich hätte gesagt bei schlechtem Management °_°

> Das wird nicht passieren. Branches mit beschränktem Zugriff sind zwar technisch möglich, aber bisher nicht gelebte Praxis in diesem Projekt oder in irgendeinem anderen, das ich kenne und welches mit Git arbeitet. Das ist klassische "SVN-Denke".
Doch muss es unbedingt
 - der master sollte gegen direkte commits  schreibtgeschützt sein. Das verhindert versehentliche "commits" - bei den paar die es hier gab ist das nämlich bereits geschehen!
 - in den ehemaligen Tags braucht niemand irgendwas zu ändern, nur-Lesezugriff verhindert Fehler
 - und wie gesagt für jeden Entwickler (der es braucht) einen eigenen Branch von dem man in den master rebast. 
   So kann man mit "checkout" einfach zwischen den Ständen wechseln. Für Forks bräuchte man getrennte Verzeichnisse, was es unnötig umständlich macht
Und vielleicht sollten das diejenigen unter sich ausmachen, die überhaupt comitten/pushen/mergen/rebasen/sqashen?

> .. Du .. bereits einen Fork hast. .. Insofern hast Du doch bereits, was Du möchtest. Oder mißverstehe ich etwas?
Schau mal was ich ganz oben schrieb:
> Diese Situation find ich nicht gut .. in Zukunft nur 1 "master"
Vielleicht wäre es einfach zu verstehen wenn man die deutsche Übersetzungen verwendet:
 - fork: sich aufspalten (wie Flüsse)
 - Branch: Zweig treiben (also neue Features/Fähigkeiten)

kriegaex, 2019-02-26:
Gerade keine Zeit zum Antworten, sorry. Melde mich aber, ist auf Wiedervorlage.

kriegaex, 2019-03-02:
Entschuldige bitte, vor dem Samstagmorgen (Ortszeit bei mir) war an eine Antwort nicht zu denken.

> Danke für die Antwort, klingt doch schonal gut!
> Da es bislang keinen Grund gab mir zu misstrauen und wir uns schon lang kennen, schlage ich vor dass du mich als auch Admin einlädst. Umgekehrt weiss ich wo ich dran bin.

Ich bin ja schon lange inaktiv als Entwickler und lese auch nichts im IPPF, lese auch hier kaum Commits oder Tickets und bin schon gar nicht auf dem aktuellen Stand, was Pseudonyme und "wer macht was" betrifft. Jetzt habe ich verstanden, daß Du Cuma bist (warst) und "Freetz-NG" verantwortest. Freetz-NG ist ein inzwischen auf GitHub gelöschter Fork von Freetz, wenn ich das richtig verstehe, d.h. es gibt ihn nur noch in Deinem SVN. Das macht die Zusammenarbeit schwer, ebenso das Vergleichen. Ich mache Dir also für Deinen speziellen Fall zwei alternative Angebot, denn Du möchtest ja offenbar bei Freetz mitmachen.

### Alternative A: Einstellung von Freetz-NG als eigenständiges Projekt

Nach eigener Aussage hast Du erfolglos versucht, Mitstreiter zu finden. Zusammen ist man stärker, also überleg Dir ob Du inzwischen teamfähig geworden bist und hier mitmachen willst. Falls ja, nimmst Du Freetz-NG vom Netz, machst entsprechende Ankündigungen im IPPF etc., forkst dieses Repository und spielst alle Dir wichtigen Änderungen von Freetz-NG als Commits ein, entweder im Master oder in Branches, wie es Dir beliebt. Wichtig ist, daß sie durch PRs nach Freetz übernommen werden können (regelmäßigen Rebase nicht vergessen). Wie in meiner vorigen Antwort beschrieben, hast Du so bis auf weiteres volle Kontrolle und Autonomie, was Freetz-NG-Inhalte betrifft.

Falls das gut klappt und es nicht wieder zu inhaltlichen und persönlichen Streitereien kommt, löschst Du, wenn alles Dir Wichtige nach Freetz übernommen wurde, den Fork, der vorher mal Freetz-NG war. Alle sind happy und Du bekommst Schreibrechte auf das Upstream-Repo (dieses hier).

### Alternative B: Weiterführung von Freetz-NG als eigenständiges Projekt

Wenn Du glaubst, Freetz weiterhin Konkurrenz machen zu müssen, erhältst Du keinen Schreibzugriff hier. Einzelne (oder auch zahlreiche) PRs sind dennoch willkommen, Zusammenarbeit ist weiterhin möglich. Aber hier Schreibrechte zu haben, auf Deinem Fork (der ja Git-technisch gar keiner ist, weil Du separat auf SVN arbeitest) jedoch weiterhin Alleinherrscher zu sein und aktiv eine (bessere?) Alternative zu Freetz zu bewerben, verträgt sich nicht miteinander.

> > Ich commite momentan auf git(hub) und synchronisiere auf svn. Benutzer bevorzugen svn, weil man leicht erkennen kann welche Revision neuer ist. So einfach ist das ^^

Und was soll das bringen? Die Git-Historie ist ein azyklischer, gerichteter Graph, da kann man viel besser nachvollziehen, was wovon abstammt, deshalb ist ja im Normalfall auch das Mergen leichter. Wenn ich in SVN im Branch A Commit 21 und im Branch B Commit 22 mache, was hilft mir die Info, welcher Commit neuer ist? Wenn ich das wissen will, schaue ich mir in Git die Zeitstempel an. Das ist also ein Nicht-Argument. Wer deswegen auf SVN bleibt, ein Git in **allen** Belangen unterlegenes System, hat Git nicht verstanden.

> Wie schnell man ein Buch lies hat keine Bedeutung, mit manchen ellenlangen Forenposts bin ich schon in 2 Sekunde fertig

Ich habe nicht behauptet, daß es eine Bedeutung hat. Ich wollte nur ausdrücken, daß das Buch so spannend war, daß ich es zwei Tage lang nicht mehr aus der Hand legen konnte. Es sollte Dir nur eventuell vorhandene Ängste wegen trockener IT-Lektüre nehmen.

> Vorgehen: Also du bist komplett gegen Branches? Tags/Branches sind nur zum festhalten eines bestimmten Standes?

Nein!!! Ich bin ein Verfechter von kurzlebigen Feature-Branches. Jedes Ticket (Bug, Verbesserung, neues Feature) hat einen eigenen Branch, der regelmäßig auf den Master rebased oder mit einem Merge vom Master aufgefrischt wird. Bedingung für die Übernahme in den Master ist, daß der Merge dorthin ein Fast Forward sein muß. Wenn Freetz eine Testsuite hätte, wäre eine weitere Bedingung, daß alle Tests vor dem Fast Forward in den Master grün sein müssen. Daß diese Vorstellung (die ich beruflich in allen meinen Projekten mit den von mir betreuten Teams aufsetze) damals nicht von allen Freetz-Mitentwicklern geteilt wurde und ein Wechsel zu Git nicht stattfand, war sogar einer meiner Gründe, nicht mehr aktiv mitzuarbeiten. Ich möchte einen stabilen Master haben, keinen Master als Entwickler-Branch.

> Entwicklungen sollen also in den Forks stattfinden? Es wird nur der "master" verwendet um zu mergen? (oder gleich committen?)

Entwicklungen _können_ in Forks stattfinden, z.B. wenn wie Du jemand nicht zum Kernteam gehört und (noch) keinen Schreibzugriff hat. Und daß momentan auf dem Master entwickelt wird, ist Realität, auch wenn ich sie nicht mag. Aber ich bin ja, wie gesagt, kein aktiver Entwickler mehr hier, habe also nicht meine Wunschvorstellung beschrieben wie oben sondern die aktuelle Situation. Ich kann und will hier keinem Aktiven Vorschriften machen.

> Soweit ich das mit git verstanden habe, ist das aber nicht das normale Vorgehen, sondern Branches wären das!

Vielleicht hast Du nur einen Teil verstanden, das macht aber nichts. Ging mir zu Beginn auch so. Technisch gesehen, ist jeder lokale Klon eines anderen Repositories ein Fork, weil er ja immer alles beinhaltet, also alle Commits und Branches. Es ist nicht wie bei SVN, wo man immer nur die Kopie eines einzelnen Branch [... cut ...] besseres SVN zu benutzen. Das wäre Schwachsinn. Man kauft ja auch keinen Ferrari, um damit mit Tempo 30 zum Brötchenholen zu fahren.

> Keine Ahnung wo du geschaut hast, ich seh davon nichts

Geschaut habe ich unter https://github.com/Freetz/freetz/network/members, dabei aber zwei kryptische Benutzernamen mit "f" am Anfang und mehrfach gleichen Ziffern dahinter verwechselt, nämlich @fda77 und @f-666. Ich finde solche Benutzernamen wenig hilfreich, habe aber nicht genau hingeschaut - mein Fehler.

fda77, 2019-03-05:
Hellau!
Hab nicht daran gedacht dass du meinen Nick hier nicht kennst. Auf freetz.org und im IPPF hab ich die Accounts nicht mehr. Hättest aber vor der GitHub Migration fragen können ;-)
Dass hier 3 Benutzer sind die mit "f" beginne ist blöd, kann man aber nicht ändern. Vielleidht hilft es wenn du mehr auf die Avatar-Bildchen achtest?

Man sollte systematischer vorghen. Es gibt die Fragen
1) ob ich hier mitmache
2) in welcher Form
3) wie das git von den Entwicklern hier benutzt wird

zu 3) sollte man besser ein anderes Issue reden (oder was es hier noch so gibt). Hätte da auch konkrete Fragen an jemanden der sich gut damit
auskennt, für ich mitnoch keine Lösung gefunden hab. Oder wie man am besten experimentelle Dinge handhabt, zb einen BusyBox bump um 3
Majorversionen, was meist irgendwelche Problem verursacht usw. 

zu 1) hast du jetzt nicht wirklich was geschrieben, ich warte auf deine Einladung

2) wie gesagt, hätte gern einen Branch (nicht master) für mich, in den ich Sachen "offiziell" pushen/mergen/committen und was es sonst noch so git kann.
Experimente würde ich dann im Fork in meinem Account machen - auf das was dort ist sollte sich niemand niemand verlassen, kann auch wieder rückgängig gemacht werden usw
Wie das in den master rüber kommt ist mir weitestgehend egal, soange nicht der Peter ran darf wegen dem ich mich schon aus dem IPPF verabschiedete.
Wenn sich sonst niemand findet würd ich es halt machen. Hab heut schonmal bissen herumprobiert, bis auf den commit der die .config der Kernel ändert kann man alle 7 Commits problemlos mergen: [https://imgur.com/blxnvKR](https://imgur.com/blxnvKR)

> Geschaut habe ich unter https://github.com/Freetz/freetz/network/members, dabei aber zwei kryptische Benutzernamen mit "f" am Anfang
Ich werde da nicht gelistet

> Wenn Du glaubst, Freetz weiterhin Konkurrenz machen zu müssen
Hatte ich nie, siehe meine beiden Posts hier. Wäre schön wenn du solche Erfindungen nicht als Tatsache hinstellst

> Nach eigener Aussage hast Du erfolglos versucht, Mitstreiter zu finden. Zusammen ist man stärker, also überleg Dir ob Du inzwischen teamfähig geworden bist und hier mitmachen willst. 
Ich wiederhole mich ungern, aber bitte Erfindungen nicht als Tatsache hinstellstellen

> Fork (der ja Git-technisch gar keiner ist, weil Du separat auf SVN arbeitest
Lass diese Erfundungen doch mal, du hast doch selbst zitiert dass ich auf github committe. Das svn syncronisiert sich einfach von github, sonst nichts

> Man sollte nicht direkt auf den Master committen 
Es ist mit als auch f666 schon passiert. Deshlab sollte man Vorkehrungen treffen

> Man kauft ja auch keinen Ferrari, um damit mit Tempo 30 zum Brötchenholen zu fahren.
:D

kriegaex, 2019-03-06:
> Hab nicht daran gedacht dass du meinen Nick hier nicht kennst.

Nicht Deine Schuld.

> Auf freetz.org und im IPPF hab ich die Accounts nicht mehr.

Davon weiß ich nichts, weil ich so lange inaktiv war.

> Hättest aber vor der GitHub Migration fragen können ;-)

Ich habe keine Migration durchgeführt. Das Repo hier gab es ja schon lange. Ich habe nur ein bißchen das Wiki repariert (ist aber immer noch sch****) und den alten SVN-Zopf abgeschnitten.

> Dass hier 3 Benutzer sind die mit "f" beginne ist blöd, kann man aber nicht ändern. Vielleicht hilft es wenn du mehr auf die Avatar-Bildchen achtest?

An alte Avatare aus meinen aktiven Freetz-Zeiten hätte ich mich sowieso nicht erinnert. Ich bevorzuge echte Namen und echte Fotos, siehe mein Foto. Aber es gibt ja anscheinend viele Entwickler, die glauben, wenn sie echte Namen und Fotos verwenden, verlören sie dadurch irgendetwas. Vielleicht schämen sie sie aber auch einfach nur, zu ihrer Arbeit zu stehen. Ja, das war jetzt polemisch und nicht an Dich persönlich gerichtet sondern mehr oder weniger in die ganze Welt.

> Es gibt die Fragen
>
> 1. ob ich hier mitmache
> 2. in welcher Form
> 3. wie das git von den Entwicklern hier benutzt wird
>
> zu 3) sollte man besser ein anderes Issue reden (oder was es hier noch so gibt). [...]

Ich habe zu 3 ein altes Angebot erneuert, das systematisch zu machen. Null Feedback bisher, wie üblich.

> zu 1) hast du jetzt nicht wirklich was geschrieben, ich warte auf deine Einladung

Doch, ich habe zwei Szenarien angeboten, entscheide Dich einfach für eines davon.

> zu 2) wie gesagt, hätte gern einen Branch (nicht master) für mich

Du hast bereits gesagt, was Du möchtest, ich habe gesagt, was ich anbiete. Siehe oben. Darüber diskutiere ich nicht weiter.

> > Geschaut habe ich unter https://github.com/Freetz/freetz/network/members, dabei aber zwei kryptische Benutzernamen mit "f" am Anfang
> 
> Ich werde da nicht gelistet

Habe ich nicht so gemeint. Den dort gelisteten Namen hatte ich mit Deinem verwechselt. Hätte es dort beide gegeben, wäre es mir aufgefallen. Aber zwischen zwei Tabs zu wechseln und sich dabei kryptische Namen zu merken, war genau das Problem. Ich hatte bereits eingeräumt, daß das mein Fehler war. Lassen wir es dabei.

Zu meinen vorgeblichen "Erfindungen": Ich erfinde nichts, ich beschreibe Tatsachen. Daß Du Mitstreiter für Freetz-NG erfolglos gesucht hast, hast Du selbst in einem anderen Thread geschrieben, ich finde ihn nur gerade nicht. Daß Du mit Deinem "Freetz-NG" Freetz Konkurrenz machst, ist auch offensichtlich: Eigenes Repo, eigene Veröffentlichungen. Dein alter Klon von Freetz hier auf GitHub enthält nur noch ein Readme, alles andere hast Du gelöscht. D.h. wir arbeiten zweigleisig. Wie ich mir eine Zusammenarbeit vorstelle, entweder als Entwickler mit Schreibrecht hier oder als Contributor, hatte ich beschrieben. Nochmals: Entscheide Dich und sage mir/uns, wie Du es gern hättest. Ein drittes Szenario wird es nicht geben. Ich hätte Dich gern im Kern-Team, aber erst, nachdem Freetz-NG abgewickelt und von der Bildfläche verschwunden ist. Mehr werde ich dazu nicht schreiben.

fda77, 2019-03-09:
> Davon weiß ich nichts, weil ich so lange inaktiv war.
Siehe 1. Post

> Ich habe keine Migration durchgeführt.
Sry, es ist ja nicht bekannt wer das machte

> Ich bevorzuge echte Namen und echte Fotos
Ich nicht :D

> Git-Workshops, s.o.
Ich bin nicht so der Workshop Typ (allgemein). Wenn ich aber Fragen hab ist es gut jemanden zu haben der sich auskennt

>  ich habe zwei Szenarien angeboten, entscheide Dich einfach für eines davon.
Wie ich im 1. Post schon schrieb (und weshalb ich immer noch hier schreib) ist dass ich es am besten finde 1 gemeinsames Repo zu haben

> Daß Du mit Deinem "Freetz-NG" Freetz Konkurrenz machst, ist auch offensichtlich
Nein. Ich fing damit an 2 Monate nach dem letzten Commit auf freetz.org bzw etwa 5 Jahre nachdem ich keine Schreibrechte auf Freetz.org mehr hatte.
Da anscheinen nun gar kein aktiver Entwickler mehr Schreibrechte hatte fing ich halt damit an und versuchte die ganzen Forks zusammenzubringen. Das ganze ging halt im Bashing von PeterPawn im IPPF unter. 
Hatte davon genug und es beteiligte sich eh niemand - und ich beendete den Versuch
Außerdem bekam dann plötzlich doch jemand hier Schreibrechte, was vorher jahrelang nicht möglich war.

Also deine/eure Bedinung dafür dass ich meine Sachen hier einbringen darf ist, dass ich alles zu Freetz-NG lösche?
Auf github.com ist nichts mehr in der Organisation, also kein Problem.
Und auf boxmatrix.info habe ich keinen Einfluss, bzw irgendwelche Berechtigungen.

Wenn ich alles richtig verstanden habe, kann ich also nichts daran ändern dass ich hier keine Schriebrechte bekomme und es bleibt wie es ist?

fda77, 2019-03-18:
Wie habt ihr euch nun geeinigt? Mir keine Schreibrechte geben, und einfach Commits übernehmen, wie in den 2 push-requests geschehen?

er13, 2019-03-19:
> Wie habt ihr euch nun geeinigt? Mir keine Schreibrechte geben

Ich weiß nicht, was Oliver entschieden hat. Ich habe mich jedenfalls dagegen ausgesprochen.

> und einfach Commits übernehmen, wie in den 2 push-requests geschehen?

Welcher Teil von GPLv2 verbietet das? Du hast schließlich in dem Moment, in dem du geforkt hast, gleich mehrere Freetz-Commits übernommen. Und wir haben nichts dagegen bzw. auch wenn wir etwas dagegen hätten, unterliegen wir auch den Bestimmungen von GPL.

p.s. Meine persönliche Sicht. Was der Rest des Teams darüber denkt, weiß ich nicht.

fda77, 2019-03-20:
Danke für deine öffentlich Stellungsnahem hier!

> Ich habe mich jedenfalls dagegen ausgesprochen. [..] p.s. Meine persönliche Sicht.

Schade, ich finde es sinnlos in Zukunft mit Patches zu hantieren. Zwischen Branches zu mergen wäre wesentlich einfacher (oder wie auch immer man es dann machen würde)

> Ich weiß nicht, was Oliver entschieden hat

Er hätte kein Problem mir Schreibrechte zu geben. Hat es aber nicht gemacht, da er die Befürchtung hat dass du sie wieder löschst und es ein hin und her gibt...
Ich seh das Enthaltung, bzw er gibt dir seine Stimme ab

> Du hast schließlich [..] geforkt [..] mehrere Freetz-Commits übernommen

freetz.org wurde doch auch auf github.com geforkt und meine Commits ohne zu fragen mitgenommen, sehen keinen Unterschied ausser in der Wahl der Reposoftware.
Wie gesagt hätte ich es nett empfunden wenn ihr meinen svn-Account mit dem github-Account verknüpft hättet

> Bestimmungen von GPL.

Wenn du schon GPL erwähnst, schau mal nach wie das mit dem Urheber&Quellen ist. Ich sehe keine Problem darin, zum original Commit einen Link und/oder git-hash in den Kommentar zu schreiben.
Wenn man einen Patch von jemand anderem für ein Package usw aufnimmt machst du das normal doch auch, und nicht nur "r1234 von xyz"

Davon abgesehen ging mir nicht um die GPL, sondern dass mir beim deaktivieren meine Accounts gesagt wurde, dass von mir nichts gebraucht + gewollt wird!
Und dann wird doch plötzlich kopiert, obwohl ich vorher angeboten hab meine Commits hier zur Verfügung zu stellen.
Wir können hier ja vereinbaren wie das in Zukunft gehandhabt wird, am besten so dass es wenig Aufwand ist. Mein Vorschlag war ein Brach in den ich commite.

Wie stellst du dir das vor? Mach doch mal einen Vorschlag (nur nicht wie kriegaex, dass ich hippies Webseite löschen soll)

er13, 2019-03-24:
> Wir können hier ja vereinbaren wie das in Zukunft gehandhabt wird, am besten so dass es wenig Aufwand ist. Mein Vorschlag war ein Brach in den ich commite.

Ich kann mich nur [diesem](https://github.com/Freetz/freetz/issues/89#issuecomment-466837862) und insbesondere [diesem](https://github.com/Freetz/freetz/issues/89#issuecomment-468876059) Beitrag von @kriegaex anschließen.

Was die Schreibrechte angeht, so verstehe ich offen gesprochen nicht, wozu diese notwendig wären und dann auch noch in der Form - beschränkt auf einen Branch. Was genau wird dieser Branch enthalten, alle Änderungen von freetz-ng oder nur die mit freetz kompatiblen und die sich ohne Konflikte mergen lassenden? Was soll dann mit diesem Branch geschehen? Wer soll es in den master mergen? Zu welchen Zeitpunkten, durch was angetriggert. Was tun, wenn wir aus bestimmten Gründen eine deiner Änderungen nicht übernehmen möchten? Und so weiter. Aus meiner Sicht versuchst du da einen komischen Prozess auszudenken, bei dem von Anfang an klar ist, dass er nicht funktionieren kann.

Ich zitiere nochmal den Beitrag von Alexander

> Es ist viel einfacher: Gehe auf die Hauptseite dieses Repositories, klicke auf "Fork" - und voilà! Du hast Dein eigenes, privates Repository, in dem Du beliebige Branches anlegen, mergen und rebasen [...]

Die Art Zusammenarbeit kann ich mir vorstellen.

---

> Ich sehe keine Problem darin, zum original Commit einen Link und/oder git-hash in den Kommentar zu schreiben.

Ich hoffe, dass "fda authored" (s. https://github.com/Freetz/freetz/commit/c956c462a2e51a5d7c1ff931e300861b537ce497 und https://github.com/Freetz/freetz/commit/dd85c531c1a634ab40ebef38d4ee55ba4a2073e4) deinen Vorstellungen von dem entspricht, wie deine Commits übernommen werden dürfen. Mir wäre es lieber, wenn es ganz normal von dir stammende (und nicht von @olistudent für dich erstellte) PRs gäbe, die man reviewen und mergen könnte.

Wenn du diese Arbeit nicht machst, ist die Wahrscheinlichkeit hoch, dass die meisten deiner Änderungen nie ihren Weg zurück in (non-ng) Freetz finden werden.

PeterPawn, 2019-03-24:
Da ich ja offensichtlich der "Stalker" aus dem IPPF sein soll, weil ich einige Merkwürdigkeiten im Zusammenhang mit dem "freetz-ng"-Fork aufzuzeigen wagte (https://www.ip-phone-forum.de/threads/freetz-ng.302327/) und auch Kritik an so manchem Punkt für angebracht hielt, habe ich mich bisher *hier* aus der Diskussion herausgehalten.

Trotzdem habe ich natürlich meine eigene Meinung - nur reicht jetzt ein einfaches "Thumbs-Up" für https://github.com/Freetz/freetz/issues/89#issuecomment-475950647 nicht - und ich will eigentlich darauf verzichten, das hier alles  noch einmal komplett niederzuschreiben - es steht im oben verlinkten Thread. Die dort von @fda77 (aka cuma) zwischenzeitlich mal gelöschten Beiträge, deren Fehlen den gesamten Thread *merkwürdig* aussehen ließe, wurden - in dem erwähnten Thread zumindest - zum größten Teil wiederhergestellt, nachdem sie vom Autoren entfernt wurden.

Ich könnte jedenfalls auch nicht verstehen, warum es eine *Extrawurst* für jemanden geben sollte, die sich ohnehin weder mit den "git"-Mechanismen, noch mit dem GitHub-GUI vertragen würde. Den "freetz-ng"-Fork gab es ja schon bis zum 15.02.2019 - in einer anderen Issue hatte ich den passenden Link auf den letzten Eintrag in dessen Historie auch schon mal angeführt: https://github.com/Freetz/freetz/compare/master...freetz-ng:978aea2a713c51d1864a558cc144cf9945b814af

Das "normale Vorgehen" unter Git-/GitHub-Bedingungen **war also bereits bekannt** und die "Abstammung" vom "freetz"-Projekt wurde erst im Nachhinein aufgegeben - welche andere Ursache als "pure Absicht" sollte das haben, wenn man auch nach entsprechenden Hinweisen (https://www.ip-phone-forum.de/threads/freetz-ng.302327/page-4#post-2316564) diesen Ansatz nicht etwa aufgibt, sondern den Graben sogar noch dahingehend erweitert, daß man den bisherigen "freetz-ng"-Inhalt entfernt und das nur noch als SVN-Repository irgendwoanders weiterführen will? Sogar das alte SVN-Repo von Freetz wurde nach GitHub gespiegelt - wer sich jetzt irgendetwas von "freetz-ng" ansehen will, muß auf den dortigen Trac-Server (den es inzwischen immerhin wieder gibt, lange blieb den interessierten Benutzern dort auch nur das SVN-Log, um die Änderungen überhaupt *kontrollieren* zu können) zurückgreifen.

Da finde ich die ständigen Beteuerungen, man würde sich eine Zusammenarbeit wünschen, schon etwas merkwürdig, wenn parallel dazu alles Erdenkliche getan wird, um genau diese zu erschweren bzw. sogar zu blockieren - sogar für diejenigen, denen man das Ganze ja zur Verfügung stellen will, nämlich die Freetz-**Benutzer**.

Für mich passen hier jedenfalls die Worte in dieser Issue und die flankierenden Taten nicht zusammen - ich bin aber auf jede plausible(!) Erklärung gespannt, habe dabei trotzdem immer "1. Johannes 2,1-6" im Auge.

---

JEDER kann zu Freetz etwas beitragen und sich dabei trotzdem an die etablierten Mechanismen (und für "git" bzw. GitHub gibt es die "best practices" ja auch nicht erst seit gestern) halten ... allerdings müßten die "Freetz-Macher" in meinen Augen auch noch ein paar "Hausaufgaben" machen, damit diejenigen, die ihren Beitrag leisten wollen, das auch "in geregelten Bahnen" tun können - auch dafür bietet GitHub (fast) alles, was man sich wünschen würde. 

Was dem eigenen Projekt an dieser Stelle vielleicht noch fehlt, kann man sich problemlos ansehen: https://github.com/Freetz/freetz/community - diese Punkte entstammen ja auch nicht nur den Hirnen irgendwelcher Theoretiker, sondern basieren auf den gesammelten Erfahrungen vieler anderer OpenSource-Projekte.

kriegaex, 2019-03-24:
> > Ich sehe keine Problem darin, zum original Commit einen Link und/oder git-hash in den Kommentar zu schreiben.
> 
> Ich hoffe, dass "fda authored" (...) deinen Vorstellungen von dem entspricht, wie deine Commits übernommen werden dürfen.
>
>Mir wäre es lieber, wenn es ganz normal von dir stammende (und nicht von @olistudent für dich erstellte) PRs gäbe, die man reviewen und mergen könnte.

Ich stimme zu: Committe selbst auf Deinem Freetz-Git-Fork, und Du stehst als Autor drin. Bei der Übernahme wird das automatisch berücksichtigt, keine muß irgendetwas beachten, weil das ein Kern-Feature von Git ist: Es gibt Committer und Autor. Als Autor würdest immer du bei jedem Commit drin stehen, als Committer derjenige, welcher den PR eingepflegt hat. Einen Benutzernamen in den Commit-Kommentar zu schreiben, ist sowas wie "Commit-Umweltverschmutzung". Zum 27. Mal: Das Git-Buch ist wirklich sehr empfehlenswert...

fda77, 2019-03-28:
@er13: Für alle meine commits einen extra PR zu machen ist mir zu zeitaufwändig. Ich weiss eh nicht was dir gefällt und was nicht, dann wäre es dazu noch überflüssig gewesen.
Wenn es einen Branch gäbe, könntest du von dort übernehmen was dir passt 

> Was die Schreibrechte angeht, so verstehe ich offen gesprochen nicht, wozu diese notwendig wären und dann auch noch in der Form

Ich bin auf Schreibrechte hier nicht angewiesen. Bin eh nicht sicher wie der aktuelle Status mir gegenüber ist ("wir wollen ncihts von dir" scheinbar nicht mehr)

> Was tun, wenn wir aus bestimmten Gründen eine deiner Änderungen nicht übernehmen möchten?

Dann lässt du halt commits die dir nicht gefallen
Was auf dauer schon blöd wäre, da die Forks wirklich auseinander laufen. Zeichnet sich schon bei Firmware bumps (MD5) und Labors (Automatik) ab
Besser wäre übernehemn und anschliessend ändern (wie es üblich war+ist)

> Aus meiner Sicht versuchst du da einen komischen Prozess auszudenken

Kannst gern besser Vorschläge machen (bitte konkreter wie "ein git Buch lesen").
Direk in den master was machen möchte ich vermeiden, das ergibt früher oder später sinnlose Streitereien

> Ich hoffe, dass "fda authored" deinen Vorstellungen von dem entspricht

Kann man ja eh nicht mehr ändern. Ich hätte erwartet dass die URL zu Quelle drin ist.
zB wie hier ccadbc5faaaeae1b0a35964ed51d6cfd794f371d
Im damit zusammenhängende Commit von mir fehlt die Quelle d813c268cbf77f11a3a4cbc521e370542bed4b36

> Ich zitiere nochmal den Beitrag von Alexander [..]

Welchen Vorteil hätte ich davon? Sehe da nur zusätzlichen Aufwand für mich, der evtl ohne Ergebnis ist

@kriegaex: Ich kenne deine Philosophie "irgendwas mit git machen". Ich bevorzuge "Freetz mit irgendwas"

kriegaex, 2019-03-28:
> @er13: Für alle meine commits einen extra PR zu machen ist mir zu zeitaufwändig. Ich weiss eh nicht was dir gefällt und was nicht, dann wäre es dazu noch überflüssig gewesen.
> Wenn es einen Branch gäbe, könntest du von dort übernehmen was dir passt

Du verstehst es immer noch nicht: Ein Branch auf Deinem Fork ist technisch genau das gleiche wie ein Branch direkt im Freetz-Repo. Es gibt keinen Unterschied darin, wie aufwendig die Übernahme ist. Aber was ganz sicher keinen Sinn macht, ist, alles in einen Branch zu packen und dann zu erwarten, daß wir Teile übernehmen, egal in welchem Repo der Branch sich befindet. Branches kann man nur ganz oder gar nicht mergen oder rebasen. Die Alternative wäre Cherrypicking (das heißt wirklich so!) einzelner Commits. Das wäre ein Alptraum und extrem aufwendig. Das wird es mit Sicherheit nicht geben. Wir wollen doch alle arbeiten und nicht frickeln. Also bitte ein PR pro Feature oder Bugfix.

> Ich bin auf Schreibrechte hier nicht angewiesen.

Eben! Und Du hast Dir das Szenario so ausgesucht durch Deine Weigerung, so wie alle anderen zu arbeiten. Sie meine Posts weiter oben dazu.

> Bin eh nicht sicher wie der aktuelle Status mir gegenüber ist ("wir wollen nichts von dir" scheinbar nicht mehr)

Das wurde doch jetzt wirklich oft genug gesagt von mehreren Beteiligten: Deine Beiträge sind willkommen.

> Kannst gern besser Vorschläge machen (bitte konkreter wie "ein git Buch lesen").

Ich habe es konkreter erklärt, diese Diskussion ist beendet. Und da Du das mit Branches, Forks etc. immer noch nicht verstanden hast: RTFM bzw. das Buch!

> Kann man ja eh nicht mehr ändern. Ich hätte erwartet dass die URL zu Quelle drin ist.

Zum letzten Mal, ich verliere die Geduld: Verweise im Commit-Kommentar sind Blödsinn, Du dtehst in selbst erstellten PRs automatisch als Autor drin und die Commits sind dann ja im Repo, Verweise sind sinnlos und überflüssig. Deine Commits werden Teil der Git-Historie des Projekts, wenn Du auf einem echten Fork committest.

> @kriegaex: Ich kenne deine Philosophie "irgendwas mit git machen". Ich bevorzuge "Freetz mit irgendwas"

Kommentare wie diesen solltest Du Dir sparen. Es geht hier um Freetz, nicht um Git als Selbstzweck. Ich erkläre nur, wie man Git richtig nutzt und wieso es SVN in allen Belangen überlegen ist.

fda77, 2019-03-28:
Hast du am frühen Morgen immer so schlechte Laune und bist so aggresiv? Dann poste doch lieber später.
Wenn ich hier 300 PR zusammenschubste, bekomme ich einen Tennisarm. Ist also keine Option

kriegaex, 2019-03-28:
Ich befinde mich seit Monaten in Asien, hier ist Nachmittag. An der Tageszeit liegt es also nicht. Mich nerven nur diese Endlos-Diskussionen mit Leuten, die für alles eine Extrawurst gebraten haben möchten. Wenn ich mir Deinen textuellen Output hier und in Foren so ansehe und extrapoliere, wie viele PRs Du in der Zeit hättest machen können ohne Tennisarm, frage ich mich, wieso wir hier Deine wertvolle und sicher begrenzte Zeit verschwenden und meine gleich mit. Das Angebot zur Zusammenarbeit via PR vom Freetz-Team liegt auf dem Tisch, geklappt hat es ja anscheinend auch schon mehrmals. Wenn es Dir zu aufwendig wird, laß es. Deine Beiträge kommen entweder freiwillig und in geeigneter Form (als PRs) oder eben nicht. Keiner wird Deine Arbeit für Dich erledigen und alles cherry-picken von Deinem Mega-Branch. Deine Entscheidung. Ich schlage vor, hier zu schließen und die Debatte zu beenden. Das ist ein Ticket-System, wir - mich eingeschlossen - mißbrauchen es als Diskussions-Forum.

closed 2019-03-28 by kriegaex