Freetz/freetz #102: Nach Firmware-Update INETD-Konfiguration zurückgesetzt.

WileC, 2019-03-04:
Nach einem Firmware-Update wird irgendwie die Konfiguration des INETD zurückgesetzt. Vor dem Firmware-Update sah meine Konfiguration folgend aus:
```
#:telnetd:23: AVM telnet daemon
23	stream	tcp6	nowait	root	/usr/sbin/telnetd	telnetd -i -l /sbin/ar7login

#:webcfg:81: Freetz web interface
81	stream	tcp6	nowait	root	/usr/bin/webcfg	webcfg -i

#:ftpd:21: AVM ftp daemon
#<off>#21	stream	tcp6	nowait	root	/bin/sh	sh /bin/inetdftp 

#:smbd:139: AVM smbd daemon
139	stream	tcp6	nowait	root	/bin/sh	sh /bin/inetdsamba 

#:smbd:445: AVM smbd daemon
445	stream	tcp6	nowait	root	/bin/sh	sh /bin/inetdsamba
```

Nach dem Firmware-Update so:
```
#:telnetd:23: AVM telnet daemon
23	stream	tcp6	nowait	root	/usr/sbin/telnetd	telnetd -i -l /sbin/ar7login

#:webcfg:81: Freetz web interface
81	stream	tcp6	nowait	root	/usr/bin/webcfg	webcfg -i
```

Einstellungen im WebIf habe ich nicht geändert. Der "NAS" ist lt WebIF "Heimnetz" weiter aktiv. Erst ein Abschalten des "NAS" -> übernehmen -> Anschalten bringt abhilfe. Dann werden die betreffenden Zeilen neu eingetragen.

WileC, 2019-03-07:
Ich habe nochmal getestet. Das gleiche Verhalten wird auch bei einem **reboot** gezeigt ... Nach einem **reboot** fehlen die Zeilen:

```
#:ftpd:21: AVM ftp daemon
#<off>#21	stream	tcp6	nowait	root	/bin/sh	sh /bin/inetdftp 

#:smbd:139: AVM smbd daemon
139	stream	tcp6	nowait	root	/bin/sh	sh /bin/inetdsamba 

#:smbd:445: AVM smbd daemon
445	stream	tcp6	nowait	root	/bin/sh	sh /bin/inetdsamba
```

f-666, 2019-03-08:
Bitte baue zum Testen ohne diesen Commit:
https://github.com/Freetz/freetz/commit/7344bf8384bff02b26f8498865a76f0244cb887e

WileC, 2019-03-08:
wie mach ich das genau ?!.. Dann habe ich aber auf meiner 3490 noch das Problem aus #109

f-666, 2019-03-08:
`wget -O - https://github.com/Freetz/freetz/commit/7344bf8384bff02b26f8498865a76f0244cb887e.patch | patch -p1 -R` im freetz Verzeichnis ausführen

JBBgameich, 2019-03-08:
Does this do the same as `git revert 7344bf8384bff02b26f8498865a76f0244cb887e`? (Just that git creates a new revert commit)

f-666, 2019-03-08:
`git revert` ist der dafür vorgesehene Befehl, richtig.

WileC, 2019-03-09:
siehe #109

Nach einem AVM-Recover, einer Neuinstallation von Freetz und wiederherstellen der Einstellungen, getrennt für AVM und Freetz, läufts wie es soll. Es werden keine Einstellungen mehr bei einem Reboot verschluckt.

f-666, 2019-03-13:
Dem letzten Kommentar nach kann dieses Issue geschlossen werden.

WileC, 2019-03-18:
ich konnte das Verhalten reproduzieren: meine Boxen (3490, 6820) verhalten sich so, wenn als erstes AVM 6.8x + Freetz drauf lief und dann auf AVM 7.01 + Freetz per WebIF geupdatet wird.

wenn man ein Recover direkt auf AVM 7.01 macht und dann Freetz über den Bootloader nachinstalliert, klappt alles reibungslos. Vielleicht kann das jemand reproduzieren?

WileC, 2019-03-18:
bitte dieses Issue wieder öffnen...

olistudent, 2019-03-18:
Reopen.

WileC, 2019-03-18:
danke.