Freetz/freetz #109: LAN-Device startet nicht, FritzBox nicht mehr erreichbar

WileC, 2019-03-06:
Ich habe heute ein neues Image für eine 3490 mit dem aktuellen Stand gebaut. Die Box startet zwar, aber das LAN wird nicht gestartet.. somit ist die Box nicht mehr erreichbar.

Mit einem Image vom Stand des 28.02.2019 startet die Box normal und ist auch per LAN zu erreichen.

[config_new.txt](https://github.com/Freetz/freetz/files/2937034/config_new.txt)
[config_old.txt](https://github.com/Freetz/freetz/files/2937040/config_old.txt)

f-666, 2019-03-08:
Hat das vielleicht was mit https://github.com/Freetz/freetz/issues/87 zu tun? Bitte die Kommandos von dem Ticket ausprobieren.

WileC, 2019-03-08:
nein gar nicht, da die Box sofort beim booten das LAN (nicht WLAN) deaktiviert...

WileC, 2019-03-09:
habe das jetzt mal durchgetestet:

habe meine 3490 mittels AVM-Recover komplett auf die V7.01 zurückgesetzt, dann das aktuellste freetz.image.in-memory mittels den EVA-Tools von @PeterPawn aufgespielt, dann als erstes lediglich die AVM-Einstellungen übers AVM-WebIF wiederhergestellt und zum Schluss NUR die freetz-Einstellungen übers Freetz-WebIf... Und schon läuft alles wie geschmiert. Die Box startet wie sie soll... und auch das Problem #102 hat sich erledigt.

Allerdings bleibt #87 bestehen.. Ohne **ifconfig wlan up** werden keine IP-Adressen an WLAN-Geräte vergeben. Lediglich APIPA.

WileC, 2019-03-18:
ich konnte das Verhalten reproduzieren: meine Boxen (3490, 6820) verhalten sich so, wenn als erstes AVM 6.8x + Freetz drauf lief und dann auf AVM 7.01 + Freetz per WebIF geupdatet wird.

wenn man ein Recover direkt auf AVM 7.01 macht und dann Freetz über den Bootloader nachinstalliert, klappt alles reibungslos. Vielleicht kann das jemand reproduzieren?

vgl. #102