Freetz/freetz #120: USB Disk unmounted after some days

donacarr, 2019-03-10:
I am running on  freetz-devel-15014  7.01 on  7490
I see in mod_mount.log:

[INFO] starting hd-idle with 600 seconds
[INFO] Mounting device /dev/sdb5 at /var/media/ftp/MyPassport ... 
[INFO] Partition MyPassport (/dev/sdb5) was mounted successfully (ext3)
[INFO] Mounting device /dev/sda5 at /var/media/ftp/SandiskDrive ... 
[INFO] Partition SandiskDrive (/dev/sda5) was mounted successfully (ext2)
[INFO] /var/media/ftp/MyPassport (/dev/sdb5) - unmounted successfully
[INFO] /var/media/ftp/SandiskDrive (/dev/sda5) - mounting read-only
[INFO] /var/media/ftp/SandiskDrive (/dev/sda5) - forcing unmount

olistudent, 2019-03-17:
I can't confirm that. But my setup is a bit different. I've attached an USB-Stick to the 7490. But this runs for months without issues. So you should check if it depends on your package selection or your usb-case?

f-666, 2019-03-19:
I don't have this problem, too. Maybe flaky USB connection / case?

donacarr, 2019-03-19:
I was running 7490_06.84-freetz-devel-14887 with same configuration and I did not face the issue....
Not sure if after the upgrade to 7.01 is mandatory a reset of the box... honestly I did not reset it.

olistudent, 2019-03-19:
I don't think that a reset will solve your issue...

donacarr, 2020-06-05:
Updating to 07.12  the issue disappears.

closed 2020-06-05 by donacarr