Freetz/freetz #121: LDAP Client missing

MartinX3, 2019-03-11:
Dear Freetz-Team,

maybe I missed something, but wasn't there support for a LDAP Client?
In my scenario I plan to run a ldap server as a central address book for several services and use the fritzbox 7490 to send the contacts to my custom DECT phones.

er13, 2019-04-03:
"Official" Freetz has never provided an LDAP client. One of the forks might however provide it. There were also a few attempts to add the support for an LDAP client in [IPPF](https://www.ip-phone-forum.de/threads/patch-openldap-funktioniert-und-avm-telefonbuch-von-aussen-auslesbar.207395/) (note the patch is very old, not sure if it still applies, works and secure).

As of now there are no plans to work on this feature. But feel free to implement it yourself and provide a PR here.

closed 2019-04-03 by er13

er13, 2019-04-03:
Buildroot provides an [LDAP package](https://github.com/buildroot/buildroot/tree/master/package/openldap). It might be a good idea to use it as the source of inspiration.