Freetz/freetz #129: 7490 Labor 7.08-66736 USB Treiber starten nicht mehr.

gradius-v, 2019-03-19:
Ich hab mal versucht ein 7490 Image mit der aktuellen Labor 66736 zu kompilieren. Leider starten die USB Treiber nicht mehr. Mit Labor 66226 gings noch.

olistudent, 2019-03-19:
Steht was im dmesg dazu? Sind die nötigen  Module geladen? Was geht konkret nicht?

gradius-v, 2019-03-19:
can't load module usbserial (kernel/drivers/usb/serial/usbserial.ko): unknown symbol in module, or unknown parameter

gradius-v, 2019-03-19:
[   81.970000] [module-alloc-by-name] give 0x9000 bytes at 0x817f1000 to module 'usbserial' (0x117000 total bytes left)
[   81.980000] module_alloc_size_list_alloc: error: module 'usbserial' reserved size 32768 too small for demand size 36864 - need 4096 more (module_alloc_waste=-4096)
[   81.990000] usbserial: Unknown symbol __dynamic_dev_dbg (err 0)
[   81.990000] usbserial: Unknown symbol __dynamic_pr_debug (err 0)
[   82.000000] usbserial: Unknown symbol __dynamic_dev_dbg (err 0)
[   82.000000] usbserial: Unknown symbol __dynamic_pr_debug (err 0)

f-666, 2019-03-19:
Da AVM die Quellen des Linux Kernels bei Laborversionen nicht zur Verfügung stellt, überrascht dieser Fehler nicht. AVM scheint die Konfiguration geändert zu haben und jetzt fehlen für das Modul _usbserial_ notwendige Funktionen. Abwarten bis die endgültige Version kommt und AVM die Quellen freigibt.

closed 2019-03-19 by f-666