Freetz/freetz #134: Möglichkeit zum Abschalten des "masquerading" vom DSL/WAN-Interface

WileC, 2019-03-23:
Ich habe mal über die ar7.cfg einen "hack" realisiert, in der ich dort das "no_masquerading=no" auf "yes" geändert hatte. So wird das NAT der Box abgeschaltet und die Box verhält sich wie ein "normaler" Router und kann als Eierlegendewollmilchsau dazu benutzt werden, Subnetze miteinander zu verbinden und bidirektional arbeiten zu lassen. Normalerweise prallen ja sämtliche Pakete am DSL/WAN-Interface von Außen ab.

Vielleicht hat jemand eine Idee, wie man das ganze schaltbar/auswählbar über das WebIF realisieren könnte?

Hier mal mein alter Post im IPPF:
[https://www.ip-phone-forum.de/threads/fritzboxen-kaskadieren-f%C3%BCr-subneting.291985/#post-2211936]

PeterPawn, 2019-03-23:
Es gibt Rudimente in der AVM-Firmware, was diese Betriebsart angeht.

Mit den folgenden Änderungen (an einer deutschen Version, bei der internationalen muß man die `guiflags.lua` nicht anfassen) kann man zumindest die Einstellung für "NAT deaktivieren" im AVM-GUI (bei den IPv4-LAN-Einstellungen) freischalten.

```
root@FB7490:/var/media/ftp $ diff -u /usr/www/avm/net/boxnet.lua boxnet.lua
--- /usr/www/avm/net/boxnet.lua
+++ boxnet.lua
@@ -8,7 +8,7 @@
 if next(box.post) and box.post.cancel then
 http.redirect(href.get(g_back_to_page))
 end
-g_my_nat=false
+g_my_nat=true
 g_isDsl = config.DSL
 g_isVdsl = config.VDSL
 g_viewmain=(g_isDsl or g_isVdsl) and (config.WLAN or config.ETH_COUNT>1)
root@FB7490:/var/media/ftp $ diff -u /usr/lua/guiflags.lua guiflags.lua
--- /usr/lua/guiflags.lua
+++ guiflags.lua
@@ -45,6 +45,8 @@
 flags.sip_packetsize = true
 flags.static_net = true
 end
+flags.static_net = true
+flags.use_nat = true
 flags.language_is_de = config.language == "de"
 local interface = {}
 function interface.import(dest)
root@FB7490:/var/media/ftp $
```

Ob das Speichern funktioniert und der Rest dann auch wie erwartet arbeitet, habe ich nie getestet. 

Sollte das der Fall sein, könnte man aus den drei Änderungen sicherlich einen optionalen Patch zusammenklöppeln - zwei davon sind ja ohnehin nur bei der deutschen Version erforderlich. Aber das "g_my_nat" wird (soweit ich das gesucht und gefunden habe) nirgendwo aus irgendeiner Einstellung gelesen und ich habe schon länger keine Version mehr gesehen, wo dieses Deaktivieren der Firewall freigeschaltet war ... denn auch in der internationalen Firmware ist dieses "g_my_nat" normalerweise fix gesetzt.

Ich gehe mal davon aus, daß die Umschaltung funktionieren könnte ... meines Erachtens ist das ein Modus, den AVM für irgendwelche Provider in einer speziellen Firmware als Option anbietet, wobei die Abhängigkeit von "static_net" wohl daher kommt, daß dieser "Routermodus" (den man ja auch von anderen Anbietern/Geräten kennt) nur mit einem öffentlichen Sub-Netz Sinn macht, wenn die Box tatsächlich ein Edge-Router ist.

WileC, 2019-03-26:
Wo genau greift denn die Einstellung aus der **ar7.cfg**? Das müsste ja irgendwie aufs DSL-Device greifen? oder auf die Firewall?

er13, 2019-04-03:
So, @WileC, das Ganze hat jetzt nichts mit Freetz direkt zu tun. Finde raus, was alles / wie zu machen ist, setze es um und `feel free to provide a patch` for Freetz.

closed 2019-04-03 by er13