Freetz/freetz #136: LTE issue with FritzBox 6890

Greg57070, 2019-03-25:
Hello,

I'm using Freetz since some weeks on my 7490 and I'm very happy with it !
I bought a FritzBox LTE 6890 to get ADSL redundancy.

No matter I use Firmware 7.0x ou 6.8x, when Freetz is installed, I've no more the LTE connection.
(Even with no added packages, just choosing Hardware type in menuconfig)

I've see on dmesg that the LTE modem is reset every 60 secs (see attached files).
Can't find any explanation on that.

If i return to official firmware, everything work.

Do you have an idea ?

[dmesgDeviceUSBLoop.txt](https://github.com/Freetz/freetz/files/3003432/dmesgDeviceUSBLoop.txt)

Thanks a lot for the good job on Freetz !
Keep up with it !

olistudent, 2019-03-25:
Can you please attach your .config?

Greg57070, 2019-03-26:
Hello,

Please find attached my .config file.
[config.txt](https://github.com/Freetz/freetz/files/3007922/config.txt)

As you can see, with the latest git version (I try again with yesterday version), in menuconfig I choose : 
- Hardware Type : 6890
- Firmware Language : (en) international
- Firmware version : FRITZ!OS 06.8x (put FRITZ.Box_6890_LTE.en-de-es-it-fr-pl-nl.162.06.87.image file in dl/fw/ downloaded from an unofficial repository) or 07.0x (tried both)

Thanks in advance for you help.

Best regards,
Gregory

Greg57070, 2019-04-27:
Hello all,

I seen some users reports their Fritzbox 6890 works good with Fritz. But they buy their one long time ago.
Is it possible, as mine is a 6890 **v2** (released by AVM on 2018-10-03) the source of my problem ?

On boxmatrix.info, they report only a LTE modem hardware modification.
The modem is a FIBOCOM L830-EB-11

Thanks in advance for your help !

BR,
Grégory

WileC, 2020-06-26:
Hallo,

ich habe nun auch das Problem, das meine "neue" 6890 mit Freetz nix vom LTE-Modem wissen will. Anbei ein ScreenShot, das SysLog und die .config.

[image]

[config.txt](https://github.com/Freetz/freetz/files/4838464/config.txt)

[syslog_freetz_noLTE.txt](https://github.com/Freetz/freetz/files/4838459/syslog_freetz_noLTE.txt)

closed 2020-07-13 by f-666

WileC, 2020-07-31:
kurzes Feedback:
aktuell läuft auf meiner 6890LTE der freetz-master-20200713-bfbcfca09 mit WLAN-Bridge (Hotspot) am WAN-Port und einer Telekom-SIM-Karte über LTE im Dual-Modus ohne Probleme.

Internet über WAN funktioniert, über LTE auch und der Fallback funktioniert auch so, wie er soll...

Somit -> produktiv getestet.

PR #303 (2020-07-13): busybox: enable some features used by avm
enable some features used by avm, espacially to use the LTE-Modems against freetz

(inspired by PeterPawn and @fda77 )

refs: 
[https://www.ip-phone-forum.de/threads/fb-6890-kein-lte-mehr-nach-freetz.307287/page-2](https://www.ip-phone-forum.de/threads/fb-6890-kein-lte-mehr-nach-freetz.307287/page-2)
Fixes #136