Freetz/freetz #139: Integrate modfs_boot_manager in freetz web-if, i.e. system page

WileC, 2019-03-28:


PeterPawn, 2019-03-28:
Why do you want it necessarily in the Freetz GUI?

There's __always__ the original GUI from AVM - no Freetz image removes it all. But there's not always a Freetz GUI - if you (or someone else) use(s) the "no-freetz" mode of `fwmod`: https://freetz.github.io/wiki/help/howtos/development/repack_fw.html

Don't get me wrong ... it's possible to integrate it and there are already some provisions for integration into other interfaces (e.g. the `get_values` call: https://github.com/PeterPawn/YourFritz/blob/master/bootmanager/gui_bootmanager#L63). 

But the idea behind it was, that there could be a more sophisticated (or in some other aspects better) solution, which can take the same values and provide an _own_ GUI with these info. 

The new code for AVM versions starting from 07.08 was more or less the first "client" of this part of implementation, because it ignores the `html_display` call and provides JSON data to the browser via the `get_values` function.

But I do not see any gain of usability yet, if it's integrated into the Freetz GUI, too - as long as this integration doesn't provide any other advantage. 

In my opinion, the Freetz GUI is not the "daily-used interface" of a FRITZ!Box - after you've set-up your system, there's only seldom any need, to use the Freetz GUI. And my arguments, that the Freetz GUI is meanwhile much more insecure than the original interface, are still valid ... compare this document from BSI: https://www.bsi.bund.de/DE/Publikationen/TechnischeRichtlinien/tr03148/tr03148_node.html with the real implementation of Freetz GUI (especially points 4.1 and 4.2). 

There's no TLS for this GUI (even not as an option), password rules do not exist (https://github.com/Freetz/freetz/blob/master/make/mod/files/root/usr/mww/cgi-bin/pwchange_page.sh) and are therefore such rules are not "enforcable", the default password uses a fixed value and it's change is 'recommended', but even not enforced ... password entropy (and the visualization of it) was never a topic, too. 
The whole update process is unprotected - see pg. 19, 2nd paragraph from top for recommendations for a SECURE handling. 

And this technical guideline wasn't written in any dark backrooms by people without the needed knowledge ... most claims and recommendations from "hackers" (e.g. CCC representives) are not even included there. All of this leads to my opinion, that the current Freetz GUI is more than outdated and would need some - quite important - updates first.

But there was an earlier discussion of this topic at IPPF board: https://www.ip-phone-forum.de/threads/fehler-bei-systemauswahl-im-freetz-im-no-freetz-modus-f%C3%BCr-7590.301079/#post-2308026 - some posts by `cuma` have been deleted meanwhile (by cuma himself), that makes it harder to read and to follow.

WileC, 2019-03-28:
Also ich nutze fast ausschließlich das Freetz-WebIF, da hier meine Dienste laufen, die ich benötige und eben auch teste.

Es ist natürlich kein großer Aufwand, aber im Moment muss ich mich zusätzlich noch auf dem AVM-GUI einloggen, um dort einen Partitionswechsel durchzuführen... 

Es wäre halt "handlicher" wenn's diese Option (evtl zusätzlich) im Freetz-WebIF geben würde/könnte.

PrinzVonBillAir, 2019-03-28:
Ich würde die **zusätzliche** Möglichkeit im Freetz-Webinterface (beispielsweise auf der system.cgi-Seite) begrüßen. Falls dafür aber dann der Bootmanager aus dem AVM Webinterface fliegt, bin ich dagegen.

er13, 2019-03-28:
I don't see any need/reason to integrate it also into the Freetz web-if or (in other words) to provide a feature already available in AVM web-if (it won't be better in any way) and thus do not plan to put any effort into it.

But feel free to implement it and to provide a "tested and ready to be reviewed" pull-request. Please check if there are corner cases (e.g. parallel calls from both AVM and Freetz web-ifs) and consider them if necessary.

er13, 2019-03-28:
Und bitte keinen cuma Code hier als eigenen kommunizieren. Wenn es eine Kopie des cuma Codes bzw. stark darauf basierend bitte explizite Hinweise darüber.

closed 2019-03-28 by er13

PeterPawn, 2019-03-28:
> Es ist natürlich kein großer Aufwand, aber im Moment muss ich mich zusätzlich noch auf dem AVM-GUI einloggen, um dort einen Partitionswechsel durchzuführen...

DAS verstehe ich als Problem ... aber ich würde hier eher zu einer anderen Lösung greifen und mir überlegen, wie man die Anmeldung an den beiden Interfaces so weit vereinheitlichen kann, daß auch das Freetz-GUI von den (unbestreitbaren) Verbesserungen bei AVM (die sich seit der Implementierung des Freetz-GUI ergeben haben) profitieren kann. Das muß ja noch nicht gleich darauf hinauslaufen, daß beide dieselbe Oberfläche verwenden (auch wenn Freetz durchaus ein Facelift im Hinblick auf mobile Geräte vertragen könnte, aber da bin ich auch Realist genug, um das als Wolkenschloß zu erkennen) ... aber dieselben Credentials sollten schon machbar sein, wenn man den Zopf der Rückwärtskompatibilität mit früheren Versionen abschneidet.

Dann könnte das Freetz-GUI von einigem profitieren, was bei AVM inzwischen Einzug gehalten hat ... vom Brute-Force-Schutz bis zur Protokollierung von Zugriffen bzw. Zugriffsversuchen. Ob auch die 2FA verwendbar wäre, müßte man mal schauen ... aber die höhere Sicherheit des AVM-GUI ist eben nicht zu übersehen.