Freetz/freetz #142: Enable external freetz gui when external https access to avm gui is enabled

WileC, 2019-03-29:


f-666, 2019-04-01:
Ich verstehe aus dem Titel dieses Issues leider nicht, was gemeint ist.
Wenn man das Freetz Webinterface per https zugänglich machen will, kann man zum Beispiel _stunnel_ verwenden.

WileC, 2019-04-01:
Ich hatte die Idee, wenn ich im AVM-WebIf die FritzBox per MyFritz oder auf herkömmlichen Wege per https freigebe, dass dann auch automatisch das Freetz-WebIf mit freigegeben werden kann ...

> Wenn man das Freetz Webinterface per https zugänglich machen will, kann man zum Beispiel stunnel verwenden.

Das kannte ich nicht und muss mir das mal näher ansehen.

PeterPawn, 2019-04-01:
Wie wäre es denn mit einem CGI-Wrapper, der unter einem anderen Pfad in der URL aufgerufen wird und damit das Freetz-GUI auch extern zur Verfügung stellen kann, ohne daß man dafür "stunnel" verwenden müßte und damit auch einen anderen Port?

Ich stelle mir da etwas in der Art vor, wie es AVM für TR-064 von der WAN-Seite verwendet - das hätte (richtig implementiert) dann auch noch den Vorteil, daß man die AAA-Mechanismen von AVM verwenden könnte  und damit eine modernere und sicherere Authentifizierung, als Freetz sie derzeit bietet. 

Das Freetz-GUI ist am Ende ohnehin immer CGI-basiert, oder? So ein Wrapper bräuchte also nach der Berechtigungsprüfung nur mit "execve()" (o.ä.) das passende Freetz-CGI aufrufen (wie das der "httpd" der BusyBox auch macht) - der Rest (Environment, STDIN) ist bereits vorbereitet. Wenn man aus STDIN liest, um die SID bei POST-Requests zu prüfen, muß man vielleicht zuvor noch den Pointer wieder auf Beginn setzen - aber ansonsten kriegt so ein eigenes CGI-Modul das alles schon auf dem Silbertablett. Da noch den passenden Header für die Authentifizierung hinzugefügt zum Environment und der Aufruf kann starten - praktisch ohne jede Änderung am Freetz-GUI, egal wie renovierungsbedürftig das ansonsten noch sein mag.

Das spart auch noch das Problem der zusätzlichen Portfreigabe auf die Box selbst, dem man sich ansonsten (bei der Benutzung von "stunnel") parallel widmen müßte - das ist bekanntlich auch nicht mehr so simpel, wie es vor PCP mal war.

Alternative wäre halt die VPN-Verbindung ... die setzt aber immer das passende Gerät voraus, auf dem dieser VPN-Zugang auch installiert ist oder zumindest (sicher) installiert werden kann. Das ist sicherlich auch nicht immer der Fall bzw. beschränkt den Zugang dann auf solche Geräte und man kann die Benutzung eines fremdem Computers (die ohnehin Risiken in sich birgt) per se schon vergessen.

WileC, 2019-04-01:
Das klingt gut. Derzeit wird beim HTTPS-Zugriff auf das AVM-WebIf (von Außen) nach einem "klick" auf Freetz die Seite `https://host.vpntome.de:450/cgi-bin/freetz_status` aufgerufen und dann auf Port 81 weitergeleitet. Dieser ist ja natürlich nicht offen, bzw. kann nicht von außen auf die Box selber zeigen.

Daher finde ich den Ansatz von @PeterPawn gar nicht so schlecht, eine Lösung mittels CGI-Wrapper zu realisieren.

PeterPawn, 2019-04-01:
Das war aber nur die "Prinzipskizze" ... bitte nicht dahingehend falsch verstehen, daß ich das nun auch implementieren würde.

Erst mal sollte man klären, wie "dringend" so etwas tatsächlich gebraucht wird - wobei es eben nach meiner Ansicht (habe ich woanders bestimmt auch schon das eine oder andere Mal argumentiert) durchaus parallel auch notwendig wäre, das Interface "auf Aktualität zu prüfen" - zumindest in puncto Security, aber vielleicht ja auch mal im Hinblick auf Darstellungen, die [kleiner als 600px in der Breite](https://github.com/Freetz/freetz/blob/master/make/mod/files/root/usr/lib/mod/cgi/width.sh#L10) sind, was bei einen Smartphone (mit geringer Auflösung) im Hochkant-Format auch schon mal der Fall sein kann oder wo man dann (beim Zoomen, damit man den Inhalt in dem 600px-Fenster noch lesen kann) einen passenden Viewport als Lupe über die gesamte HTML-Seite schieben muß, wenn man von links bis rechts und von oben bis unten den Seiteninhalt lesen will.