Freetz/freetz #147: Kernel "config-puma6.x86-6490_07.02" lässt sich nicht übersetzen

smischke, 2019-04-02:
Zunächst einmal Daumen hoch für dieses Projekt! freetz-master läuft auf meiner 6490 in der Default-Konfiguration für FritzOS 07.0x ohne Probleme. 

Bislang benutze ich sie allerdings im Bridge-Mode mit einem Linux-Router, der Firewall (iptables) und Traffic Shaping (tc, IFB, cake) übernimmt. Dies würde ich gerne auf die Fritzbox verlagern. Leider scheitern meine Versuche, IFB als Modul mit aufzunehmen. In "make kernel-menuconfig" habe ich lediglich "CONFIG_NET_CLS_ACT=y" und "CONFIG_IFB=m" geändert. Ersteres wird gebraucht, damit zweiteres sichtbar und auswählbar wird. Bei einem "make" kommt am Ende folgende Meldung:

```
make[2]: Verzeichnis »/home/freetz/freetz/source/kernel/ref-puma6.x86-6490_07.02/linux-2.6.39« wird verlassen
make[1]: *** [vmlinux.eva_pad] Fehler 2
make[1]: Verzeichnis »/home/freetz/freetz/source/kernel/ref-puma6.x86-6490_07.02/linux-2.6.39« wird verlassen

ERROR: Build failed.
make: *** [source/kernel/ref-puma6.x86-6490_07.02/linux-2.6.39/vmlinux.eva_pad] Fehler 1

```
Dies scheint ein Folgefehler von diesem hier zu sein:

```
  i686-pc-linux-gnu-gcc -Wp,-MD,drivers/net/avm_cpu_connect/.avm_cpu_connect.o.d  -nostdinc -isystem /home/freetz/freetz/toolchain/build/i686_gcc-5.5.0/i686-pc-linux-gnu/lib/gcc/i686-pc-linux-gnu/5.5.0/include -I/home/freetz/freetz/source/kernel/ref-puma6.x86-6490_07.02/linux-2.6.39/arch/x86/include -Iinclude  -include include/linux/kconfig.h  -D__KERNEL__ -I/home/freetz/freetz/source/kernel/ref-puma6.x86-6490_07.02/linux-2.6.39/arch/x86/include -Iinclude -include include/linux/kconfig.h -DNEW_CONFIG -I/home/freetz/freetz/source/kernel/ref-puma6.x86-6490_07.02/linux-2.6.39/include -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -fno-delete-null-pointer-checks -Wno-missing-field-initializers -O2 -m32 -msoft-float -mregparm=3 -freg-struct-return -mpreferred-stack-boundary=2 -march=atom -mtune=atom -mtune=generic -Wa,-mtune=generic32 -ffreestanding -DCONFIG_AS_CFI=1 -DCONFIG_AS_CFI_SIGNAL_FRAME=1 -DCONFIG_AS_CFI_SECTIONS=1 -pipe -Wno-sign-compare -fno-asynchronous-unwind-tables -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -fno-stack-protector -fno-omit-frame-pointer -fno-optimize-sibling-calls -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fconserve-stack -DCC_HAVE_ASM_GOTO -Wall -Wextra -Werror    -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(avm_cpu_connect)"  -D"KBUILD_MODNAME=KBUILD_STR(avm_cpu_connect_driver)" -c -o drivers/net/avm_cpu_connect/avm_cpu_connect.o drivers/net/avm_cpu_connect/avm_cpu_connect.c
make -f scripts/Makefile.build obj=net/ipv6/netfilter
make -f scripts/Makefile.build obj=net/l2tp
make -f scripts/Makefile.build obj=net/llc
make -f scripts/Makefile.build obj=net/netfilter
make -f scripts/Makefile.build obj=net/netlink
make -f scripts/Makefile.build obj=net/packet
make -f scripts/Makefile.build obj=net/sched
In file included from include/net/pkt_sched.h:6:0,
                 from drivers/net/avm_cpu_connect/avm_cpu_connect.c:45:
include/net/sch_generic.h: In function 'skb_act_clone':
include/net/sch_generic.h:648:14: error: unused parameter 'action' [-Werror=unused-parameter]
          int action)
```

Offenbar müsste man irgendwo `-Wno-error=unused-parameter` einfügen, aber ich habe keine Ahnung wo. Alternativ kann man natürlich den bislang unbenutzten Parameter dummy-mäßig benutzen, aber das wäre eine Frickellösung. Kann mir jemand helfen?

f-666, 2019-04-03:
Schau im entpackten Kerneltree, wo der Paramenter `-Werror=unused-parameter` gesetzt wird und nimm ihn raus. Falls der Kerneltree überschrieben wird, musst du das Archiv neu packen und in der `freetz/config/mod/download.in` die MD5 Prüfsumme löschen.

smischke, 2019-04-03:
Wenn ich in der Datei `freetz/source/kernel/ref-puma6.x86-6490_07.02/linux-2.6.39/drivers/net/avm_cpu_connect/Makefile` in der Zeile `EXTRA_CFLAGS += -Wall -Wextra -Werror` das `-Werror` entferne oder auskommentiere, dann lässt es sich übersetzen. Die Datei ist aber nicht im git index. Vermutlich wird sie generiert oder entpackt. Wo müsste man es im Repository ändern, damit es richtig gefixt ist?

f-666, 2019-04-04:
Die Kernelquellen werden von den Freetz Mirrorservern heruntergeladen und unter `dl/fw` abgelegt. Somit sind sie nicht im _Repository_. Falls du das dauerhaft und sauber in Freetz haben willst, musst du einen Patch erstellen und im passenden Verzeichnis unter `make/linux/patches` ablegen.

smischke, 2019-04-04:
Danke für's in die richtige Richtung lenken. Ich werde versuchen, einen Patch beizusteuern.

smischke, 2019-04-05:
Komischerweise wird ifb.ko nicht mit ins Image gepackt:
```
root@FB6490:/var/mod/root# modprobe ifb
modprobe: module ifb not found in modules.dep
root@FB6490:/var/mod/root# find / -name ifb.ko
root@FB6490:/var/mod/root#
```
Woran könnte das liegen?

PeterPawn, 2019-04-05:
https://github.com/Freetz/freetz/blob/master/config/ui/modules.in

smischke, 2019-04-05:
Danke. Bei FREETZ_MODULES_ALL kommen die von mir gewünschten Module rein. Allerdings auch viele andere. Da muss ich wohl jeweils Punkte für anlegen (ifb, hfsc, ...). Ist ein Pull Request gewünscht?

PeterPawn, 2019-04-05:
> Ist ein Pull Request gewünscht?

Ich würde an Deiner Stelle erst mal prüfen, ob die Ideen, die Du mit dem IFB realisieren willst (sicherlich irgendwas mit eigenem QoS) auch tatsächlich funktionieren. Durch den AVM-eigenen IP-Stack auf der WAN-Seite ist das nicht unbedingt gesagt. 

Wenn das dann tatsächlich läuft, besteht sicherlich auch Interesse bei anderen - aber da gehört dann m.E. neben dem Erstellen des LKM "ifb.ko" noch die Beschreibung dazu, was der Freetz-Benutzer damit anfangen kann und wie er das genau anstellt.

smischke, 2019-04-05:
Ich folge im Grunde den Ideen in diesem Thread:
[7390 + Downstream QOS](https://www.ip-phone-forum.de/threads/7390-downstream-qos.256939/)

Wenn es klappt, dann melde ich mich. Wenn nicht vermutlich auch mit Hilfeschreien. :)

PeterPawn, 2019-04-05:
Das wird dann vermutlich doch eine sehr, sehr spezielle Lösung.

Mal abgesehen davon, daß bei einer Kabel-Box das QoS ohnehin "geteilt" läuft, denn das DOCSIS-Modem (das läuft auf dem ARM-Core des Puma6) wird - in Teilen - ohnehin vom Provider per Config-File gefüttert. Auch hat sich beim AVM-QoS in den letzten sechs Jahren einiges getan ... das muß man wohl alles berücksichtigen.

Ich weiß ja nicht, was konkret Deine "Schmerzen" mit dem QoS der Box sind, aber auch die 6490 ist ohne Beschleunigung für die Pakete eher langsam, was ihren Netzwerkdurchsatz angeht und wenn Du nicht gerade an irgendeinem muchtigen 32 MBit-Anschluß Deines KNB hängst, ist das Abschalten des PA (und ohne das siehst Du eben nicht alle Pakete auf der x86-CPU) vermutlich eher eine schlechte Idee.

Aber ich bin mal gespannt, was Du da "regeln" willst per QoS - vielleicht hast Du ja übertriebene Latenzprobleme. Wobei die Diskussion dazu dann ohnehin wieder ins IPPF gehört ...

closed 2019-04-06 by er13