Freetz/freetz #151: Wie mache ich ein neues Image?

Conan179, 2019-04-07:
Hallo erst mal.
Ich hab da mal eine blöde frage, wie erstelle ich ein neues Image?
Ich hab hier zuhause eine 7272 und eine 7390 mit schon älteren Freetz.
Jetzt möchte ich sie erneuern und weil meine Image noch auf der alten Seite waren, möchte ich neu anfangen, also wie mache ich das?

Endschuldigung für die Noob frage.

JBBgameich, 2019-04-07:
Du checkst am beaten das neue git repository per `gut clone https://github.com/freetz/freetz` aus. Ab diesem schritt ist alles genau wie früher mit svn.

Conan179, 2019-04-07:
Ah super, Ich danke dir!

Conan179, 2019-04-07:
Oh noch eine kleine frage, wie wird ein update des geklonten Datenstand erstellt?

JBBgameich, 2019-04-07:
`git fetch origin && git merge origin/master` oder alternativ `git pull` sollten reichen.

Conan179, 2019-04-07:
Ok, Danke!

closed 2019-04-07 by Conan179

er13, 2019-04-07:
So, Leute, das ist hier ein Bug-Tracking-System und kein Support-Forum!