Freetz/freetz #152: Fritzbox 7170 als Alien für 7112 macht dauerschleife

Sebijk, 2019-04-07:
siehe auch diese Diskussion: https://www.ip-phone-forum.de/threads/fritz-box-7112-7170-alien-annex-a.265519/

Die Erstellung des 7170 Alien Firmware für 7112 klappt, aber nach dem Flashen macht die Box ein Dauerreboot.

Das Draufspielen eines 7113 Firmware auf 7112 funktioniert hingegen einwandfrei. Hier sollte man jedoch ein Patch anbieten, dass die Festnetztelefonie verschwinden lässt, da die 7112 nur Internettelefonnie kann.

er13, 2019-04-08:
Die .config fehlt. Aber auch mit der wird dir vermutlich keiner der Entwickler helfen können, weil keiner so eine uralte Hardware hat. Musst du selbst rausfinden, woran es liegt. Dies zum einen durch den Vergleich der original-Firmware mit der alien-Firmware - vielleicht wurde was übersehen. Und zum anderen anhand der Einschätzung wie weit die Box beim Booten kommt, dies anhand der Zeit bzw. LED-Verhalten.

Sebijk, 2019-04-13:
Bootet ca. 65 Sekunden, bevor es neu startet und WLAN-LED ist auch an.
```
#
# Automatically generated file; DO NOT EDIT.
# Freetz Configuration
#
FREETZ_HAVE_DOT_CONFIG=y

#
# User competence ------------------------
#
# FREETZ_USER_LEVEL_BEGINNER is not set
# FREETZ_USER_LEVEL_ADVANCED is not set
FREETZ_USER_LEVEL_EXPERT=y
FREETZ_SHOW_ADVANCED=y
FREETZ_SHOW_EXPERT=y

#
# Hardware/Firmware ----------------------
#

#
# WLAN Repeater
#
# FREETZ_TYPE_1750 is not set

#
# LAN
#
# FREETZ_TYPE_2170 is not set

#
# WLAN
#
# FREETZ_TYPE_3020 is not set
# FREETZ_TYPE_3030 is not set
# FREETZ_TYPE_3130 is not set
# FREETZ_TYPE_3131 is not set
# FREETZ_TYPE_3170 is not set
# FREETZ_TYPE_3270_V2 is not set
# FREETZ_TYPE_3270_V3 is not set
# FREETZ_TYPE_3272 is not set
# FREETZ_TYPE_3370 is not set
# FREETZ_TYPE_3390 is not set
# FREETZ_TYPE_3490 is not set

#
# WAN
#
# FREETZ_TYPE_4020 is not set
# FREETZ_TYPE_4040 is not set

#
# Fon
#
# FREETZ_TYPE_FON is not set
# FREETZ_TYPE_5010 is not set
# FREETZ_TYPE_5050 is not set
# FREETZ_TYPE_5124 is not set
# FREETZ_TYPE_5140 is not set

#
# Fiber
#
# FREETZ_TYPE_5491 is not set

#
# Cable
#
# FREETZ_TYPE_6490 is not set
[... 2121 lines cut ...]
FREETZ_KERNEL_VERSION="2.6.13.1"
FREETZ_KERNEL_VERSION_MAJOR="2.6.13"
FREETZ_KERNEL_VERSION_MODULES_SUBDIR="2.6.13.1-ohio"
FREETZ_KERNEL_VERSION_2_6_13_MIN=y
FREETZ_KERNEL_VERSION_2_6_13_MAX=y
FREETZ_KERNEL_VERSION_2_6_19_MAX=y
FREETZ_KERNEL_VERSION_2_6_28_MAX=y
FREETZ_KERNEL_VERSION_2_6_32_MAX=y
FREETZ_KERNEL_VERSION_2_6_39_MAX=y
FREETZ_KERNEL_VERSION_3_10_MAX=y
FREETZ_KERNEL_VERSION_4_4_MAX=y
FREETZ_AVM_UCLIBC_0_9_29=y
FREETZ_AVM_SOURCE_04_80=y
FREETZ_AVM_SOURCE_ID="04.80"
FREETZ_REPLACE_KERNEL_AVAILABLE=y
FREETZ_REPLACE_MODULE_AVAILABLE=y
FREETZ_TYPE_PREFIX="7170"
FREETZ_TYPE_PREFIX_SERIES_SUBDIR="04_XX"
FREETZ_TYPE_PREFIX_ALIEN_HARDWARE="7112_"
FREETZ_INSTALL_BASE=y
FREETZ_REPLACE_BUSYBOX=y
FREETZ_CIFS_SUPPORT_AVAILABLE=y
FREETZ_NFS_SUPPORT_AVAILABLE_AS_MODULE=y
FREETZ_NFS_SUPPORT_AVAILABLE=y
```

PeterPawn, 2019-04-14:
@Sebijk:
I'm not sure, what's your native language, if you're really from NZL ... that's why I'll write in English to avoid misunderstandings.

If you try to read https://github.com/Freetz/freetz/issues/152#issuecomment-482894456 yourself, do you notice something? 

If not, a little hint: No-one wants to READ the content of your `.config` file in a comment, at least that's not the primary intention / usage. 

What do you think, is the usual way to deal with a foreign `.config` file? Correct ... it gets downloaded to the own system and gets viewed with some kind of text viewer program or even with `mconf`,  which is the whole magic behind `make menuconfig`.

But how could anybody download YOUR `.config` file from the comment above? Do you really think, someone would use copy & paste to get the content from your comment to an own text file?

Short: Please consider to edit the comment above once again ... remove the "inline configuration" and attach a file version of this data. 

To attach a file, you may drag & drop it to your input area or click on "selecting" on the footer above the "Comment" button to get a "select a file to attach" dialog. 

But the file can't be "hidden" - that means, it's name is not allowed to start with a dot and therefore you have to rename your `.config` file first - best use `config.txt` as name.

Sebijk, 2019-04-23:
Hello,

i put the content in a file.

I found out, that problem is only when i choosed the international version. If i choose the german version, it works. But the german version support only Annex B.

PeterPawn, 2019-04-23:
> I found out, that problem is only when i choosed the international version. If i choose the german version, it works. But the german version support only Annex B.

Are you sure, that your device was set to the correct branding (usually it's `avme` for the international version) prior to next reboot, after you've installed the new image?

If the device starts without problems using the german version, it usually has a "branding" (value of `firmware_version` from the "urlader environment" - may be read with `GETENV` command from FTP session) of `avm` or `1und1`. The latter is more expectable, 'cause the 7112 was built (afaik) only as an OEM version for the german provider 1&1.

If you want to install a firmware with a different branding, you have to set the right value in this environment accordingly.

If you're steady switching between `avme` and `avm` during your experiments and you did not mention it in the problem report, simply forget this hint.

Sebijk, 2019-04-24:
> Are you sure, that your device was set to the correct branding (usually it's `avme` for the international version) prior to next reboot, after you've installed the new image?

Yes, i changed that also to avme.