Freetz/freetz #154: Q: Why doesn't Freetz use a symlink in kernel patch files in target's filename?

PeterPawn, 2019-04-10:
While looking/reading through some kernel patch files (from `make/linux/patches`) of older versions I found, that practically all of these files use the real path name in their content to address the file(s) to change.

Am I really wrong, if I would expect, that each build system always contains a symlink called `linux`, pointing to the right directory containing the kernel sources? And that, if this symlink doesn't exist yet, it even has to get created? Best also, before the patch files are applied?

This assumption would make some things much easier ... even while someone with a new kernel version still has to check such patches with the source (are they still needed, need them new/refreshed anchor code, etc.), the patch files must not be changed at a very first step, to point to the new target files. And if some patch files may cover more than a single kernel version (maybe with acceptable offsets for their "hunks"), they don't need to be changed anymore.

I was pointed to this question while looking into this simple one-liner, where line no. 3 could/should look slightly different to cover more than only a kernel version of 3.10.xyz: https://github.com/PeterPawn/YourFritz/blob/master/patch_kernel/src2patch.sh 

It's only a question yet ... and a request for other opinions (what are the current (dis-)advantages?), not a demand to change something immediately.

PeterPawn, 2019-04-10:
Ach so ... ich würde auch dafür plädieren, das ggf. in Deutsch fortzusetzen. Wenn jemand den Sinn der Sache verstanden hat und weitergehendes Interesse am Thema und der Diskussion hat, aber kein Deutsch lesen kann, könnte er sich folgende Beiträge ja von irgendeinem Translator-Service übersetzen lassen. 

Ich finde das jedenfalls besser, als wenn sich jemand durch die Mühen des Formulierens der eigenen Ansicht in Englisch davon abhalten läßt, sich dazu auch zu äußern. Jenseits des einleitenden Beitrags sollte jeder in der Sprache schreiben, die ihm am besten liegt - damit meine ich allerdings auch nur die Auswahl zwischen Deutsch und Englisch, damit jetzt niemand mit Русский язык oder Español anfängt.

er13, 2019-04-11:
> Am I really wrong, if I would expect, that each build system always contains a symlink called `linux`, pointing to the right directory containing the kernel sources?

No, you're not wrong. It's created [here](https://github.com/Freetz/freetz/blob/d6d29a1c498e171170be1341d38caa02ed6b943b/make/linux/kernel.mk#L111)

> And that, if this symlink doesn't exist yet, it even has to get created? Best also, before the patch files are applied?

and if necessary could be moved [here](https://github.com/Freetz/freetz/blob/d6d29a1c498e171170be1341d38caa02ed6b943b/make/linux/kernel.mk#L42)

> This assumption would make some things much easier ... even while someone with a new kernel version still has to check such patches with the source (are they still needed, need them new/refreshed [...]

Even though we have a lot of identical patches (s. the statement below), it is still much easier if in context of one kernel-sources-package one have to deal only with patches related to this package without the need to check/think how would his/her changes affect the other kernel-sources-packages.

```
$ find make/linux/patches -type f -name "*.patch" -exec md5sum -b \{\} \+ | sort
```

> I was pointed to this question while looking into this simple one-liner, where line no. 3 could/should look slightly different to cover more than only a kernel version of 3.10.xyz: [...]

I think for special cases like yf_patchkernel it's ok to use `linux` instead of `linux-$(KERNEL_VERSION_MAJOR)`. Should I move the `ln -s` statement as written above?

---

> Ach so ... ich würde auch dafür plädieren, das ggf. in Deutsch fortzusetzen. Wenn jemand den Sinn der Sache verstanden hat und weitergehendes Interesse am Thema und der Diskussion hat, aber kein Deutsch lesen kann, könnte er sich folgende Beiträge ja von irgendeinem Translator-Service übersetzen lassen. ... Jenseits des einleitenden Beitrags sollte jeder in der Sprache schreiben, die ihm am besten liegt - damit meine ich allerdings auch nur die Auswahl zwischen Deutsch und Englisch, damit jetzt niemand mit Русский язык oder Español anfängt.

Interessanter Gedankengang ;-) Die, die kein Deutsch können, sollen "die Beiträge von irgendeinem Translator-Service übersetzen lassen". In wieweit unterscheiden sich Russisch oder Spanisch an der Stelle von Deutsch? Die, die kein Russisch/Spanisch/andere Sprache können, können doch genauso "die Beiträge von irgendeinem Translator-Service übersetzen lassen" ;-)

er13, 2019-04-11:
Btw. after reading the [code](https://github.com/Freetz/freetz/blob/d6d29a1c498e171170be1341d38caa02ed6b943b/make/linux/kernel.mk#L45) (notice `$(KERNEL_DIR)` instead of `$(KERNEL_BUILD_ROOT_DIR)`) I'm a bit confused why our patches contain the prefixes like `linux-VERSION` at all. All our package patches do not contain any `packagename-packageversion` prefixes.

So I would say a better solution would be to change
```
#kernel version specific patches
@$(call APPLY_PATCHES,$(KERNEL_PATCHES_DIR),$(KERNEL_DIR))
#firmware version specific patches
@$(call APPLY_PATCHES,$(KERNEL_PATCHES_DIR)/$(AVM_SOURCE_ID),$(KERNEL_DIR))
```
to
```
#kernel version specific patches
@$(call APPLY_PATCHES,$(KERNEL_PATCHES_DIR),$(KERNEL_BUILD_ROOT_DIR))
#firmware version specific patches
@$(call APPLY_PATCHES,$(KERNEL_PATCHES_DIR)/$(AVM_SOURCE_ID),$(KERNEL_BUILD_ROOT_DIR))
```
and simply remove all `linux-VERSION` prefixes from all kernel patches.

PeterPawn, 2019-04-12:
> Die, die kein Russisch/Spanisch/andere Sprache können, können doch genauso "die Beiträge von irgendeinem Translator-Service übersetzen lassen"

OK, ich ging hier schon davon aus, daß Deutsch und Englisch die am häufigsten bei FRITZ!Box-Besitzern eingesetzten Sprachen sind. Meinetwegen kann jeder auch in Russisch schreiben oder sogar in Klingonisch, solange die Übersetzungsangebote das wieder in eine für mich verständliche Sprache wandeln. :sunglasses: 

Ich habe nur schon gehört, daß sich Leute ohne Englisch-Kenntnisse (oder mit geringen) einfach nicht trauen, hier dann zu schreiben (und die sprachen dann in erster Linie Deutsch) bzw. ihre eigenen Beiträge schon mittels eines Übersetzungsprogramms ins Englische bringen, bevor sie sie hier einzustellen.

Da wäre dann das "Gemisch", das auch im IPPF benutzt wird, schon ein Angebot mit einer deutlich niedrigeren Einstiegshürde und wir wollen ja eigentlich, daß die Leute Fehler melden und andere Vorschläge machen, oder?

Insofern kannst Du den Gedankengang bzgl. weiterer Sprachen auch so interpretieren, daß ich dabei "das Wohl vieler" im Auge hatte, das mehr zählen sollte als "das Wohl weniger" (ja, bin halt Cineast). Wenn also die kleine Schar deren, die flüssig Russisch oder Spanisch sprechen (und schreiben) können, ihre Beiträge selbst ins Englische oder ins Deutsche übersetzen, können es wieder die vielen lesen, die sich ansonsten noch mit Freetz befassen und eine der beiden Sprachen beherrschen. 

Wenn tatsächlich die Mehrzahl der hier Lesenden jedesmal erst die Übersetzungshilfen bemühen muß, um einen Kommentar in Russisch zu verstehen, ist das vermutlich deutlich mehr Aufwand (über alle gerechnet), als wenn so ein Autor dann seinerseits eine der "Projektsprachen" benutzt, können wir uns darauf einigen? 

Ich wollte/will gar keine Sprachen diskriminieren ... nur ist ausschließlich Englisch eine Hürde für so manchen Freetz-Benutzer, während andererseits jeder (Freetz-Benutzer) - schon wg. der in Freetz derzeit unterstützten Sprachen - entweder mit Deutsch oder Englisch oder beidem klarkommen sollte. 

Das ist auch der Grund, warum ich ausschließlich diese beiden Sprachen als "Projektsprachen" ansehen würde - wenn jemand zusätzlich Französisch in Freetz einbaut (bis in die Packages), stehe ich (opportunistischerweise) auch am Rande und winke mit der Trikolore, pendant que j'applaudis. Liberté, égalité, fraternité - également pour les langues étrangères.

Ich hoffe, daß das Deine Frage, warum ich die Sprachen so unterschiedlich "wichten" (oder behandeln) würde, beantwortet? :grin:

Bei Gelegenheit will ich mich dann auch mal ans Werk machen und ein Template für PRs und Issues vorschlagen, das genau diese Informationen (welche Sprache sollte man verwenden, welche Infos werden benötigt, usw.) beeinhaltet, denn GitHub kann dann auch solche "templates" bereitstellen und die "Beitragenden" (auch bei Fragen/Problemen) haben dann gleich die passenden Infos, was sie alles angeben müssen. Da kann/soll dann auch noch mal darauf hingewiesen werden, daß es hier im GitHub ums Bug-Tracking bzw. um Erweiterungs- und Änderungswünsche geht und nicht darum, wie man was macht und was man wo einstellen soll.

Zwar steht das auch irgendwo schon, aber wenn's beim Druck auf "New issue" auch angezeigt wird, kann sich niemand mehr herausreden, er hätte es nicht gefunden - speziell was die "Wie mache ich ..."-Fragen angeht, die nun im IPPF ja deutlich besser zu stellen sind ... das stand ja auch schon im Trac-Wiki.

---

> and simply remove all linux-VERSION prefixes from all kernel patches.

Eine mögliche Lösung ... wobei man dann ja auch mit "-p1" arbeiten könnte und das erste Level eben ignoriert. Ich hatte noch nicht verglichen, wie weit sich die Patches für verschiedene Kernel-Versionen am Ende tatsächlich unterscheiden (jenseits von Offsets) und ich weiß auch nicht, wie gut der ja deutlich unschärfer arbeitende Patch-Code in Freetz dann damit klarkommt.

Stellt sich das als problemlos heraus, könnte man ja wirklich das "ln -s" nach oben ziehen, denn "logisch" kann es sicherlich sowohl am Beginn als auch am Ende der Vorbereitung der Kernel-Quellen stehen.

---

Abhängig vom zusätzlichen Aufwand, irgendwelche "Developer-Scripts" auch noch anzupassen, wenn man nicht länger mit "linux" (oder auch dem Verzeichnis mit der Versionsnummer) als erste Verzeichnisebene  patchen will (wenn es solche Werkzeuge gibt), würde ich das erste Level in den Patch-Files lassen.

Andererseits gibt es wohl tatsächlich in den Kernel-Quellen interessanterweise nicht ein einziges Patch-File, an dem man sich orientieren könnte, wie andere es machen. Schaut man dann in andere Projekte (z.B. OpenWRT), setzen die Patches dort auch erst ab dem "linux"-Verzeichnis an (auch wenn da als erstes i.d.R. "a" und "b" stehen, weil vieles Git-Diffs sind).

Aber wie schon geschrieben ... alles ohne jede Priorität in meinen Augen. Ich war nur auf der Suche nach einer Erklärung, wie es dazu kam bzw. warum es immer noch so ist. 

Einiges hat Freetz ja auch von OpenWRT "geerbt" bzw. übernommen ... aber solange da inzwischen Patch-Files mit Git-Diffs verwendet werden, kann man die ja ohnehin nicht 1:1 übernehmen, wenn man keine Erkennung einbaut, die dann automatisch die passende Anzahl von Pfadkomponenten beim "patch"-Aufruf ignorieren läßt oder "a" und "b" gegen "linux-something" austauscht.

Wenn ich das richtig sehe, hat man ja drei Pfadvariablen zur Verfügung beim "make" für den Kernel:

```
[... cut ...]
wobei die beiden letzten sich nur dadurch unterscheiden, daß die eine über den Symlink geht und die andere nicht.

Wenn man das nun "streng nach Name" betrachtet, würde ich als (unvoreingenommener) Betrachter jetzt erwarten, daß beim Patchen von Kernel-Sourcen auch der Pfad verwendet wird, dessen Name am ehesten auf diese Quelldateien verweist - sprich: Ich würde - wenn man das ändert - sogar anstelle von "KERNEL_BUILD_ROOT_DIR" das "KERNEL_SOURCE_DIR" verwenden, selbst wenn das (derzeit zumindest) eben dieselbe Stelle im Dateisystem (wenn auch mit anderem Pfad) ist.

PeterPawn, 2019-05-07:
Ich mache hier mal zu (falls sich niemand das ansonsten traut), weil m.E. alles gesagt ist und wenn man weitere Konsequenzen daraus ableiten wollte (z.B. das "Verschieben" des "ln -s"-Aufrufs nach vorne), geht das ja auch bei geschlossenem Eintrag weiterhin - aber die Liste der "offenen Fragen" wird wieder kürzer.

er13, 2019-05-07:
Ehrlich gesagt wollte ich das Ticket noch bewusst im Status `offen` halten. Ich habe mich noch entschieden, was ich mit dem Patch-Level mache. Ob ich nun
1. `linux-$(KERNEL_MAJOR_VERSION)` komplett entferne, oder
2. durch `linux` ersetze, oder
3. gar nichts mache.

Aktuell tendiere ich zu 1, denn es wäre konsistent zu dem, was wir bei den Packages machen. Das einerseits und andererseits würde das Dein (zugegebenermaßen kleines) Problem aus https://github.com/PeterPawn/YourFritz/blob/master/patch_kernel/src2patch.sh lösen, welches der Anlaß für diese Issue gewesen ist.

PeterPawn, 2019-05-07:
Nach Deinem Belieben ... ich wollte nur meinen Standpunkt dazu "kundtun" und übergebe hiermit die "Verantwortung" für das spätere Schließen an Dich. :sunglasses: 

Wie Du das auch umsetzt, ich werde es mitbekommen und dann ggf. "patch_kernel" entsprechend anpassen.

PeterPawn, 2019-06-06:
Ich würde jetzt darum bitten, daß hier IRGENDETWAS passiert - es sind weitere 4 Wochen vergangen, seitdem ich diesen Fall hier schließen wollte, da meine Frage (s. Titel) beantwortet ist.

Angesichts von 19 anderen, offenen Problemen muß etwas wie diese Frage nicht die "issues queue" zusätzlich verstopfen ... zumal ja in den vergangenen vier Wochen weder neue Infos dazugekommen wären, noch jemand anderes eine Meinung zur ventilierten Frage geäußert hat.

Wenn das hier bei irgendwem als Task/ToDo enden sollte, dann sollte er/sie das auch im eigenen Repo entsprechend verwalten können - hier vergrößert es nur "die Oberfläche", die ein anderer absuchen muß, wenn er ergründen will, ob ein von ihm selbst zu meldendes Problem bereits anderweitig erwähnt/beschrieben wurde ... auch wenn ich es schon mit dem Titel so weit wie möglich abgrenzen wollte.

Ich stehe zwar noch als "author" hier drin (daher finde ich das bei entsprechender Recherche auch als "open"), aber ich hatte das meinerseits bereits geschlossen und (heute) vor 8 Wochen eröffnet (also 2x die "offene Zeit" nach meiner 4-Wochen-Policy). 

Also möge bitte jemand anderes das Problem hier übernehmen (und zwar so, daß **ersichtlich** ist, daß sich jemand darum kümmert und ich damit nichts mehr zu tun habe - dafür gäbe es z.B. "assigned to", was ich dann bei meiner Abfrage berücksichtigen könnte) oder das doch wieder schließen - ich hatte die Verantwortung dafür ja abgegeben. 

Ich kann zwar hier ebenfalls wieder "zu machen", aber ich will auch nicht, daß das auf irgendwelche "Spielchen", wer schneller mit der Mouse ist, hinausläuft.

Und "nein" ... ich verlange von niemandem, daß er irgendwelche weiteren Arbeiten auf der Basis des hier Geschriebenen ausführt - egal, ob er das freiwillig macht oder ob er dazu gezwungen wird. 

Nur ist das hier eben - zumindest in meinen Augen - auch kein privater Notizzettel, was man noch alles machen wollte ... das Argument mit der Demotivation anderer und dem zusätzlichen Aufwand für andere, wenn sie beim Erstellen eigener Issues "verantwortungsvoll" vorgehen wollen und selbst nachsehen, ob ein Problem schon beschrieben wurde, sollte nachvollziehbar sein (und stammt auch ursprünglich nicht von mir, ich finde es nur sehr nachvollziehbar und mache es mir daher zu eigen - gerne verlinke ich noch mal auf die "best practices for maintainers": https://opensource.guide/best-practices/).

er13, 2019-06-06:
Ok, vorerst 3.

closed 2019-06-06 by er13