Freetz/freetz #161: Remove WLAN - Zugangsdaten site broken

Sebijk, 2019-04-24:
After i removed wlan, the Zugangsdaten site seems to broken. I cant fill any input field and if i try to save, it log me out from the webinterface. The javascript console give me following errors:

Loading failed "http://fritz.box/js/wlanscan.js". fritz.box:1
SyntaxError: unexpected token: ':' fritz.box:266:4

er13, 2019-05-03:
What box, what firmware version, .config is missing => useless/invalid bug report

Sebijk, 2019-05-05:
Its 7360 firmware 6.85

Sebijk, 2019-05-05:
config file:

[config_7360.txt](https://github.com/Freetz/freetz/files/3144995/config_7360.txt)

PeterPawn, 2019-05-05:
You could try to remove lines 32 to 38 from the `360-remove_wlan.sh` file (https://github.com/Freetz/freetz/blob/master/patches/scripts/360-remove_wlan.sh#L33).

The code, which depends on existing WLAN support, is properly enclosed (by AVM) with the needed check for presence of this option.

That's possibly not the last problem from this remove patch - please take note of the fact, that all remove patches are EXPERIMENTAL starting with FRITZ!OS version 06.5x, where AVM switched to the "responsive interface design".

Therefore it may be necessary to change a piece of code and immediately test the result yourself - the chances, that other people use the same old device (with the same old firmware) and the same request (to remove all WLAN related functions from their Freetz image) are really tiny.

These patch files were constructed to change as much as possible versions from one single file ... meanwhile the removal of the line starting with `wlanscanOnload` (line 37 of the original patch file) leads to a syntax error in the Lua file, because only the first line of four gets removed as it was necessary in earlier versions of this file.

```
1037 wlanscanOnload("<?lua box.js(box.glob.sid) ?>", <?lua box.out(wlanscan.getstate()) ?>);
```

vs. 

```
 959 if (hasWLAN) {
 960 wlanscanOnload({
 961 sid: "<?lua box.js(box.glob.sid) ?>",
 962 scan: <?lua if config.WLAN then box.out(wlanscan.getstate()) else box.out([[{}]]) end ?>
 963 });
 964 }
```

as seen from version 06.5x on.

As mentioned above ... that's only ONE possible problem - with newer versions the patches are mostly untested and are sometimes not working, too. 

That's explicitly given as warning:

https://github.com/Freetz/freetz/blob/master/config/ui/firmware.in#L416

With newer devices there's no need to remove parts of the original firmware to get additional free space - therefore these patches are more and more superfluous and are not maintained any longer.

Sebijk, 2019-05-09:
> You could try to remove lines 32 to 38 from the `360-remove_wlan.sh` file (https://github.com/Freetz/freetz/blob/master/patches/scripts/360-remove_wlan.sh#L33).

Thanks, that fixed this problem. 

Now only the Setup assistant if the fritzbox is use first time have still this problem. It produces a blank page  and can only bypass by visiting specific url (example by including ?lp=sys).

[image]

> With newer devices there's no need to remove parts of the original firmware to get additional free space - therefore these patches are more and more superfluous and are not maintained any longer.

I agree with you. But the problem on 7360 is, there are exists two versions. V1 have only 16 mb and get FritzOS only until version 06.33. For 6.85 it is too big. It only fits by removing some big modules like Samba and WLAN. On 7390 which have also 16 MB, i saw that AVM fixed this problem by put the WLAN module as a plugin in the internal NAS space. Maybe that could be an option for 7360 v1 by using an usb stick.