Freetz/freetz #165: FRITZ!Box 4040 - Swap-Dienst kann nicht gestartet werden

daxfox2172, 2019-05-01:
Beim Starten des Swap-Dienstes nach dem Erstellen einer Swap-Datei über das Freetz Webinterface erscheint der folgende Fehler:

sed: /proc/swaps: No such file or directory
Starting swap ... failed.

Hardware-Informationen
Boxname: FRITZ!Box 4040 ANNEX: Ohne
HWRevision: 227.1.0.0 RAM: 256 MB
NAND: 0 MB TFFS: 32 MB
CPU-Kerne: 4
Taktfrequenzen
CPU MHz SYSTEM MHz BogoMIPS 109.80 / 109.80 / 109.80 / 109.80
Betriebsstundenzähler 14 hours 0 days 0 months 0 years (19 starts)

Firmware-Informationen

Firmwareversion: 155.07.10 AVM-Revision: 67848 Sprache: de
Erstellungsdatum (AVM)
Bootloaderversion: 1.3243

Freetz-Informationen

Boxtyp: 4040 AVM-Firmwareversion: 07.10 Sprache: de
Kernelversion: 4.4.60 (jenkins@ldxmesh18xcortexa9xfullxMi1004x1) (gcc version 5.4.0 (Buildroot 2016.05-gc3ca4dd) )
Freetz-Version: freetz-master-c616c7bf2-190428
Erstellungsdatum: 01.05.2019 14:39:25
Ursprünglicher Dateiname: freetz-master-c616c7bf2-190428.de_20190501-143925.image

Irgendwelche Ideen?

PeterPawn, 2019-05-02:
https://github.com/Freetz/freetz/blob/master/make/linux/configs/freetz/config-ipq40xx-4040_07.01#L46

PeterPawn, 2019-05-02:
Die Behandlung der Situation: "kein Swap-Support im Kernel" sollte durch #166 nachgerüstet werden (es gibt ja auch noch Modelle, die keine USB-Ports haben) - nur hilft Dir das nicht beim Aktivieren von Swap-Space, sofern Dein Kernel mit der Standardkonfiguration erzeugt wurde. 

Auch der AVM-Kernel wurde offenbar ohne Swap-Unterstützung kompiliert. 

Meines Wissens ist bisher noch keine Vorkehrung in Freetz vorhanden, um dem Benutzer die Wahl beim Swap-Support zu lassen - das bräuchte weitere (durchaus umfangreichere) Änderungen, weil alle Änderungen an der Kernel-Konfiguration, die über die Frage "LKM ins Freetz-Image kopieren oder nicht" hinausgehen, bisher nicht so richtig gut umgesetzt werden können (soweit ich das verfolgt habe).

Durchaus denkbar, daß man daher für die 4040 in der Freetz-eigenen Kernel-Konfiguration das "CONFIG_SWAP" von Beginn an einschaltet - dann braucht's aber auch für den Swap-Support zwingend das "replace kernel" ... was auch wieder nicht jeder Freetz-Benutzer möchte und was m.W. bisher auch nicht als "Freetz-Symbol" mit entsprechenden Abhängigkeiten existiert.

Daher wird die Abwägung hier vermutlich noch spannende Diskussionen hervorbringen ... bis dahin hast Du die Wahl, die Änderungen auf #166 für Dich zu testen und auf die Verwendung von Swap-Space zu verzichten oder Du änderst Dir selbst die (Kernel-)Konfiguration (hinsichtlich "CONFIG_SWAP") und aktivierst das "replace kernel".

daxfox2172, 2019-05-02:
Scheint so als würde die Option "replace kernel" zur Zeit ausfallen, momentan sagt mir make menuconfig nur, das replace kernel nicht verfügbar ist (not available, no sources by AVM). Schätze ich muss dann wohl erstmal auf die Funktion verzichten.

PeterPawn, 2019-05-02:
Ohne das jetzt geprüft zu haben ... ich bezweifle einfach mal, daß AVM ausgerechnet bei der 4040 (im Gegensatz zu praktisch allen anderen Boxen) Entscheidendes am Kernel geändert hat.

Damit bleiben zwei Möglichkeiten:

- entweder man verwendet die Version 07.01 - immerhin könnte man dort dann Swap-Space benutzen oder
- man ändert selbst dahingehend, daß für die 07.10 derselbe Kernel gebaut wird, wie für die 07.01

Ich hätte jetzt, basierend auf dieser Zeile: https://github.com/Freetz/freetz/blob/master/config/avm/source.in#L237 auch erwartet (ebenfalls ohne eigenen Test), daß für die 07.10 bei der 4040 dieselben Quellen wie für die 07.01 herangezogen werden.

Soweit ich weiß, gibt es bisher gar keine Quellen für die 07.10 von AVM - nicht einmal für die 7590 (bzw. 7580), die als erstes Modell die 07.10 erhalten hat: osp.avm.de

Man muß also selbst (kreativ) tätig werden oder tatsächlich verzichten bzw. warten, bis sich jemand die Mühe macht, (a) die Quellen für 07.10 bei AVM anzufordern und (b) die notwendigen Änderungen "offiziell" am Freetz-Master vorzunehmen.

Das geht - naturgemäß - schneller (bzw. vermutlich würde "replace kernel" dann auch mit den 07.01-Quellen für die 07.10-Version freigeschaltet), wenn es jemand testet.

daxfox2172, 2019-05-02:
Gut, dahingehend reichen dann meine Kenntnisse nicht aus, das überlass ich dann jemandem der da mehr in der Materie ist. Zudem ich ohnehin noch selber keinen eigenen Kernel kompiliert habe, selbst unter Linux auf einem normalen Desktop nicht... bisher hatte ich nie wirklich einen Grund dazu.