Freetz/freetz #177: Labor-Versionen werden beim MAKE nicht mehr entpackt

WileC, 2019-05-09:
Derzeit werden bei einem **make** die Laborversionen nicht mehr aus den ZIP-Containern entpackt.

Fehlermeldung:
```
ERROR: Could not extract firmware image.
make/Makefile.image.in:75: recipe for target 'dl/fw/FRITZ.Box_6890_LTE-07.08-67492-LabBETA.image' failed
make: *** [dl/fw/FRITZ.Box_6890_LTE-07.08-67492-LabBETA.image] Error 3
```

PeterPawn, 2019-05-09:
Ursache ist diese Zeile: https://github.com/Freetz/freetz/blob/master/make/Makefile.image.in#L20

Sie muß korrekt lauten:

```
DL_SOURCE$(1)_CONTAINER_SUFFIX:=$(suffix $(FREETZ_DL_SOURCE$(1)_CONTAINER))
```

Zwar wird in der Zeile darüber der Variablen "$(DL_SOURCE$(1)_CONTAINER)" ein Wert zugewiesen, diese Zuweisung wird aber erst beim `$(eval ...)` in Zeile 24 bis 26 (für die verschiedenen Indizes) ausgeführt. Daher hat "$(DL_SOURCE$(1)_CONTAINER)" zum Zeitpunkt der Zusammenstellung im Makro in Zeile 20 noch keinen Wert und das "case"-Statement im Makefile hat am Ende keinen Suffix zur Auswahl des Programms fürs Entpacken.

er13, 2019-05-09:
@WileC: bitte testen, habe die Anpassung über die Github Edit-Funktion im Browser gemacht, kann selbst leider nicht testen. Danke!

WileC, 2019-05-10:
@er13 läuft mit mehreren Labors durch, für eine 6890 und 6820 auch lauffähig (compile + run tested)

closed 2019-05-10 by WileC