Freetz/freetz #182: Login to web interface

AndyHee, 2019-05-14:
Hello,

This issues is a support question about logging in via the web interface on port 81.

In the past, I successfully built several images with branch 2.x of this code for FB 7360 v1 that were fully functional.

Now I've cloned the master branch and built two images that seem to have compiled and flashed OK, but I can't log in. I assume the logon details are: `admin` and `freetz`.

What could be the issue? How to login via the web interface?

PeterPawn, 2019-05-14:
If your device contained already a Freetz image in the past and you haven't removed all old settings, any earlier changes to the credentials are still active - did you try to login with any other/earlier (hopefully changed) credentials yet?

AndyHee, 2019-05-14:
Thanks for the swift reply!

Yes, that was indeed the case. I worked - great! Thanks.

closed 2019-05-14 by AndyHee