Freetz/freetz #186: 7560 + Fritz!OS-7.1x, GRX5 boxes: integrate kernel sources from opensrc pack...

feedzapper, 2019-05-26:
Seit der Integration der Kernel Sources kommt es auf meiner 7560 sehr oft zur Kernel panic mit "out of memory" und anschließendem reboot.
Die 7560 hat ja nur 256MB die anderen Boxen 512MB , besteht da ein evtl. ein Zusammenhang ? 
Due Quellen sind ja von einer 7583 wie ich lese ...

feedzapper, 2019-05-27:
Problem gefunden, liegt definitiv an irgendwelchen inkompatiblen DNSSEC Servern für DNSMASQ obwohl sie laut Liste DNSSEC tauglich sind. Mit 1.1.1.1 oder 1.0.0.1 als Server läuft die Box tagelang ohne den oben beschriebenen Fehler.

closed 2019-05-27 by feedzapper

PeterPawn, 2019-05-27:
> wäre sicher eine fundamentale Bereicherung für Freetz als Package

https://github.com/Freetz/freetz/tree/master/make/getdns

https://www.ip-phone-forum.de/threads/getdns-paket.297283/

feedzapper, 2019-05-27:
Danke ! -> läuft 
Unter der Bezeichnung "getdns" hatte ich es NICHT auf dem Schirm.
Könnte es mal jemand updaten ?
stubby -V sagt -> 0.2.3
Ich habe in Tomato -> 0.2.6

PeterPawn, 2019-05-27:
Ich bin mir nicht sicher, daß jemand die Aufforderung hier drin liest - oder sie gar zum Anlaß nimmt, da tätig zu werden.

Wie wäre es denn mit einem neuen Eintrag?