Freetz/freetz #187: Bug: unable to downgrade via Freetz GUI, ...

PeterPawn, 2019-05-29:
if a FRITZ!OS version is running, where the file `/etc/version` had get the latest changes (regarding the new way to store version values in `/etc/init.d/rc.conf`).

Only for the records, 'cause no-one else has mentioned it yet.

First analyses may be found here: https://www.ip-phone-forum.de/threads/openvpn-server-bridge-auf-7560-ipclient-fritz-os-7-01-freetz15014.302100/page-5#post-2328721

I'm personally unable to test any changes, I'm not using Freetz mod anywhere. Therefore I'm the wrong contact/consignee for further questions or tests, if anyone would start to solve the problem - sorry.

WileC, 2019-05-29:
Ich bekomme bezüglich der /etc/version folgenden Fehler:
`May 29 11:27:05 fritz daemon.err dnsmasq[2874]: failed to update mtime on /etc/version: Read-only file system`

PeterPawn, 2019-05-29:
Das mag zwar auch mit der "/etc/version" zu tun haben, aber definitiv nichts mit dem Downgrade -> falsche Issue.

Ansonsten gehört das Thema (wenn man nichts Neues aufmachen will) hier hin: https://github.com/Freetz/freetz/issues/163 - weiter also dort.

er13, 2019-06-06:
> Only for the records, 'cause no-one else has mentioned it yet.
Auch 4-Wochen-Frist oder darf es "for records" auch länger offen bleiben?

PeterPawn, 2019-06-06:
Gerne noch einmal deutlich ... wenn es (a) jemand zur Kenntnis nimmt (was hiermit vermutlich geschehen ist) und (b) es eine Perspektive für eine Lösung gibt, dann ist mir das egal.

Es ist nur nicht sinnvoll (die Gründe wiederhole ich nicht, auch andere Punkte aus den "best practices" sind so interessant/wichtig, daß man das schon selbst mal lesen sollte und dann kann man sich immer noch darüber austauschen, was davon auch für Freetz gelten sollte), solche Sachen ohne Reaktion stehen zu lassen.

Denkbare Reaktionen reichen von der Zuweisung von Labels (weil man bei einem Bug die Issues mit Tag "Enhancement" gar nicht erst ansehen muß) bis zur Zuweisung an einen Verantwortlichen - bei dem kann man dann nämlich auch mal nachfragen, ob/wann es weiter geht. Afaik ist das Zuweisen von Labels direkt beim Erstellen einer Issue nicht möglich (für den "Bearbeiter" weiß ich es sicher) - gegenteilige Infos nehme ich gerne und kennzeichne das dann auch selbst, wenn es funktionieren sollte.

---

Ich betone es gerne noch einmal: Diese Policy hat überhaupt nichts mit Dir zu tun - wer das tatsächlich verfolgt hat (und ich hatte meinerseits dorthin verlinkt: https://github.com/Freetz/freetz/pull/66#issuecomment-464047370 - ein "habe ich nicht mitbekommen" ist also auch keine nachvollziehbare Begründung), konnte auch kaum auf diese Idee kommen.

Denn der erste "Vorfall", bei dem ich das so gehandhabt und auch schon so begründet habe, liegt vom Datum her (15.02.2019) in einer Zeit, wo hier (und im IPPF) nur noch die ganz Unverdrossenen damit gerechnet haben, daß Du je wieder auftauchen würdest. 

Als "Beweis" für diese "Erwartungshaltung" kannst Du gerne auch noch einmal in diesen PR schauen: https://github.com/Freetz/freetz/pull/116 - meine Aufforderung zur Diskussion (inkl. Verweis auf den Thread, wo wir unsere Positionen "ausgetauscht" haben - ich habe da nicht ein Argument versucht zu verschweigen) kannst Du dort im ersten Beitrag sehen.

---

Ich weiß jedenfalls nicht genau, was Dich jetzt wieder gebissen hatte (https://github.com/Freetz/freetz/pull/166#issuecomment-497272768), daß Du dieses für eine angemessene Reaktion hieltest ... aber ich bin inzwischen auch wieder auf dem Rückzug. 

Ich werde in Zukunft wieder meinen Stiefel an meinem eigenen Fork durchziehen und auf PRs künftig verzichten. Sollte mal etwas dabei sein bei meinen (Freetz-)Branches und anderen Projekten, was irgendjemand für Freetz als "sinnvoll" erachtet, steht - sofern die Lizenz paßt, wenn das nicht von mir selbst schon in einem Freetz-Fork veröffentlicht wird - die Übernahme frei.

Aber meine Hoffnungen, daß es mit dem Übergang zu GitHub auch eine "neue Kultur" beim Umgang mit Vorschlägen von anderen geben könnte bzw. würde, habe ich inzwischen auch wieder begraben - und da meine ich auch nicht primär mich und meine Vorschläge (partiell schon, komme ich noch mal drauf zurück), damit das nicht wieder als "nu' isser beleidigt" fehlinterpretiert werden kann.

Es ist ganz einfach: Höflichkeit und Freundlichkeit tut normalerweise nicht weh ... schaut man sich so manches hier in diesem Repo an PRs und Issues an und die Reaktionen darauf, gewinnt man (zumindest geht es mir so, das MUSS man selbstverständlich nicht teilen als Ansicht) eher den Eindruck, daß solche "Ansinnen" eher dahingehend interpretiert werden, daß sie "die Friedhofsruhe", die ansonsten herrscht, stören, was in D obendrein strafbar ist (§ 168 StGB). Wobei das Zurechtschneiden der Hecken bei mir noch Bestandteil dieser Friedhofsruhe ist ... das läuft bei mir unter "Wartung" und "bloß nichts wirklich ändern/verbessern".

---

Wie es - bei der Freundlichkeit und beim Umgang mit den Leuten, die hier etwas beitragen wollen - besser ginge, zeigt(e) spaßigerweise schon @olistudent für den ersten PR, den jemand hier im GitHub eingebracht hatte: https://github.com/Freetz/freetz/pull/3#issuecomment-272713199

> Danke für die Pakete. 

Und nun behaupte bitte niemand, daß ein "Thanks!" nach einer "Aufgabenliste", die der "Störer" zunächst mal abzuarbeiten hat, dasselbe ist - das ist jedenfalls garantiert kein Dank dafür, daß sich da jemand beteiligen will. 

Ich betone auch gerne noch einmal, daß es mir hier gar nicht um irgendwelche Threads geht, in denen sich @er13 und meine Wenigkeit direkt "beharkt" haben ... es ist der generelle Ton im Umgang mit anderen, der mir hier (sehr sauer) aufstößt und der mich - nach einer zwischenzeitlichen Phase der Hoffnung - inzwischen wieder zur Überzeugung gebracht hat, daß ich mich besser komplett ausklinke.

Ob das auch für Hilfen für Fragesteller im IPPF gilt oder nicht, habe ich noch nicht abschließend entschieden.

---

Sollte sich hier der Umgang miteinander (sichtbar) wandeln und dazu gehört es für mich auch, daß deutlich "selbstherrliche Entscheidungen" zurückgehen und zumindest ein Austausch (gerne auch zwischen den "members" - daß der im Sinne der Transparenz auch öffentlich sein sollte, steht wieder in den "best practices":

> Don’t forget to document your interactions, too. Wherever you can, keep communication about your project public. If somebody tries to contact you privately to discuss a feature request or support need, politely direct them to a public communication channel, such as a mailing list or issue tracker.

) stattfindet, würde ich das neu durchdenken ... aber so, wie er derzeit aussieht, ist mir der Umgang miteinander hier einfach zuwider. Punkt. 

[... cut ...] ächlich "verpflichtet" gefühlt, hier noch darauf hinzuweisen ... heute kann ich das tatsächlich auch nicht mehr nachvollziehen, was mich da geritten haben mag.

---

Ich habe mit Stand heute jedenfalls alles das "abgearbeitet", wofür ich mich (zumindest moralisch) zuständig fühlte ... damit ist die Sache für mich gegessen und ich habe künftig keinen Bedarf mehr, einen eigenen Überblick (im Freetz-Projekt) über den Fortgang von Arbeiten an von mir eröffneten Issues oder Pull-Requests zu gewinnen.