Freetz/freetz #190: remove of dsld not possible

wschleifer, 2019-05-30:
Hello all,
i want to set the option "remove dsld".  as an automatism the option "remove kids" will also be checked.

Then I start the make process and I get the following message:
menulua_remove: no matches for '\["lua"\] = "[^"]*kids.lua"" found

My settings are:
fb7390 
git repo: may 29 / 23:17 / commit 02c06654cd972e127b4b00457dc28da5d9499e09

Thanks a lot for your help.
Werner.

programminghoch10, 2019-06-17:
Removal patches generally don't work reliably, we need to wait for somebody to get working on it

Djfe, 2019-09-28:
I just realized the same thing
just for reference:
https://www.ip-phone-forum.de/threads/make-firmware-nocompile-fb-7390.304573/
https://www.ip-phone-forum.de/threads/7390-trunk-13529-error-menulua_remove-no-matches-for.283357/

Djfe, 2019-09-28:
How do I compile for my 7390, without any removal patches?