Freetz/freetz #201: DNSMasq doesn't recognize all nameservers at boot?

WileC, 2019-06-21:
Werden beim starten der Box vom DNSMasq nicht alle Nameserver eingelesen? Oder nur nicht richtig im Syslog ausgegeben?

Hier das Syslog direkt nach dem Start der Box:
```
Jan  1 01:00:30 fritz daemon.info dnsmasq[2887]: started, version 2.80 cachesize 150
Jan  1 01:00:30 fritz daemon.info dnsmasq[2887]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset auth DNSSEC loop-detect no-inotify dumpfile
Jan  1 01:00:30 fritz daemon.info dnsmasq[2887]: DNSSEC validation enabled
Jan  1 01:00:30 fritz daemon.info dnsmasq[2887]: DNSSEC signature timestamps not checked until system time valid
Jan  1 01:00:30 fritz daemon.info dnsmasq[2887]: configured with trust anchor for <root> keytag 20326
Jan  1 01:00:30 fritz daemon.info dnsmasq[2887]: configured with trust anchor for <root> keytag 19036
Jan  1 01:00:30 fritz daemon.info dnsmasq[2887]: asynchronous logging enabled, queue limit is 10 messages
Jan  1 01:00:30 fritz daemon.info dnsmasq-dhcp[2887]: DHCP, IP range 192.168.10.200 -- 192.168.10.239, lease time 12h
Jan  1 01:00:30 fritz daemon.info dnsmasq[2887]: using nameserver 192.168.251.1#53 for domain 251.168.192.in-addr.arpa (no DNSSEC)
Jan  1 01:00:30 fritz daemon.info dnsmasq[2887]: using nameserver 192.168.251.1#53 for domain rmk-hotspot.lan (no DNSSEC)
Jun 21 11:38:07 fritz daemon.warn dnsmasq[2887]: overflow: 19 log entries lost
Jun 21 11:38:07 fritz daemon.info dnsmasq[2887]: read /etc/hosts - 19 addresses
Jun 21 11:38:07 fritz daemon.info dnsmasq-dhcp[2887]: read /etc/ethers - 13 addresses
```
Es wird nur der letzte Eintrag "server=..." aus der Extra-Konfiguration ins Log geschrieben.

Und nach einen Restart des DNSMasq:
```
Jun 21 13:54:06 fritz daemon.info dnsmasq[2887]: exiting on receipt of SIGTERM
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: started, version 2.80 cachesize 150
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: compile time options: IPv6 GNU-getopt no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset auth DNSSEC loop-detect no-inotify dumpfile
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: DNSSEC validation enabled
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: system time considered valid, checking DNSSEC signature timestamps.
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: configured with trust anchor for <root> keytag 20326
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: configured with trust anchor for <root> keytag 19036
Jun 21 13:54:07 fritz daemon.warn dnsmasq[5327]: warning: using interface lan instead
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: asynchronous logging enabled, queue limit is 10 messages
Jun 21 13:54:07 fritz daemon.info dnsmasq-dhcp[5327]: DHCP, IP range 192.168.10.200 -- 192.168.10.239, lease time 12h
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 192.168.251.1#53 for domain 251.168.192.in-addr.arpa (no DNSSEC)
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 192.168.251.1#53 for domain rmk-hotspot.lan (no DNSSEC)
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 192.168.250.1#53 for domain 250.168.192.in-addr.arpa (no DNSSEC)
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 192.168.250.1#53 for domain kh-hotspot.lan (no DNSSEC)
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 192.168.1.1#53 for domain 1.168.192.in-addr.arpa (no DNSSEC)
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 192.168.1.1#53 for domain tmk-audio.lan (no DNSSEC)
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 192.168.1.1#53 for domain kurekhome.lan (no DNSSEC)
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 192.168.1.1#53 for domain d-jay.lan (no DNSSEC)
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using local addresses only for domain 10.168.192.in-addr.arpa
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using local addresses only for domain rmkurek.lan
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using local addresses only for domain powerline
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using local addresses only for domain box
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 2606:4700:4700::1001#53
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 2606:4700:4700::1111#53
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 1.0.0.1#53
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: using nameserver 1.1.1.1#53
Jun 21 13:54:07 fritz daemon.info dnsmasq[5327]: read /etc/hosts - 19 addresses
Jun 21 13:54:07 fritz daemon.info dnsmasq-dhcp[5327]: read /etc/ethers - 13 addresses
```
Hier werden nun alle Nameserver (Upstream/VPN) im log korrekt angezeigt.

er13, 2019-06-21:
> 
> ```
> ...
> Jan  1 01:00:30 fritz daemon.info dnsmasq[2887]: asynchronous logging enabled, queue limit is 10 messages
> ...
> Jun 21 11:38:07 fritz daemon.warn dnsmasq[2887]: overflow: 19 log entries lost
> ...
> ```

From http://www.thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html

```
--log-async[<lines>]

Enable asynchronous logging and optionally set the limit on the
number of lines
which will be queued by dnsmasq when writing to the syslog is slow. 
Dnsmasq can log asynchronously: this
allows it to continue functioning without being blocked by syslog, and
allows syslog to use dnsmasq for DNS queries without risking deadlock.
If the queue of log-lines becomes full, dnsmasq will log the
overflow, and the number of messages  lost. The default queue length is
5, a sane value would be 5-25, and a maximum limit of 100 is imposed.
```

WileC, 2019-06-21:
Aber ich verstehe nicht ganz. Es sind doch die selben Einstellungen beim Boot als auch beim Restart?

Und bei beiden Logs ist „asynchronous logging enabled“ mit 10 Zeilen, also Werkseinstellung?

er13, 2019-06-21:
Zu Bootzeit ist syslog vermutlich zusätzlich mit etwas anderem beschäftigt und nimmt weniger Nachrichten von dnsmasq entgegen.

Setze den Wert in der extra-Section höher und teste, ob es hilft. Wenn ja, könnten wir entweder den [hartkodierten](https://github.com/Freetz/freetz/blob/41062db62e878b69c9b613f5acab682becb2eedf/make/dnsmasq/files/root/etc/default.dnsmasq/dnsmasq_conf#L7) Wert erhöhen oder gar konfigurierbar machen.

WileC, 2019-06-21:
Man darf leider lt. Fehlermeldung keine zwei gleichen Keywords in der Konfiguration angeben. 

Ich werde das mal über die Konsole testen.

er13, 2019-06-21:
Dann editiere die generierte config händisch und schick das SIGHUP an dnsmasq (dnsmasq sollte dann die Konfiguration neu einlesen).

```
kill -HUP <pid of dnsmasq>

oder

killall -HUP dnsmasq
```

Hmm. Habe ich geschrieben und erst dann realisiert, dass das Problem ja beim Booten auftritt.

Am einfachten wäre es den Quellcode zu ändern und dann neu zu flashen.

WileC, 2019-06-21:
Ich werde mal was basteln.. so dass ein Standardwert gesetzt wird, welcher im WebIf angezeigt und ggf geändert werden kann.

Braucht man diesen Config-Tag überhaupt ?? .. wenn man ihn gar nicht setzt? .. also komplett aus der Konfiguration rauslässt??

er13, 2019-06-21:
> Braucht man diesen Config-Tag überhaupt ?? .. wenn man ihn gar nicht setzt? .. also komplett aus der Konfiguration rauslässt??

Weiß ich nicht. Vermutlich wird dann synchron geloggt. Ob blockierend oder nicht, kann ich nicht sagen. Vermutlich eher nicht, die Log-Nachrichten gehen dann einfach verloren (zumindest wäre das meine Erwartung bzw. ich würde es so umsetzen).

er13, 2019-06-26:
Fixed in https://github.com/Freetz/freetz/commit/b71ec415684f94df79e7c16d869f76f8ca084053

closed 2019-06-26 by er13