Freetz/freetz #204: [Segmentation fault] meshd

Helmut1972, 2019-06-22:
Hallo,

hab heute die freetz-master-20190620-934181fe0 auf einer 7490 eingespielt. Augenscheinlich funktioniert alles, aber im Log kommt:

1970-01-01 01:01:09(1) [Segmentation fault] meshd([3184]) CRASHED at 00000000 (?) accessing 00000000
SIGNO 11 ERRNO 0 CODE 1
Version: 07.11
No bugmsg
ze: 00000000 at: 00000001 v0: 00000000 v1: 778dd090
a0: 76ea102c a1: 00000001 a2: 7f9b67e4 a3: 7787071d
t0: 00000001 t1: 00000001 t2: 00000001 t3: 00000400
t4: 00000800 t5: 00000807 t6: 00000030 t7: 0000006d
s0: 76ea3e90 s1: 7f9b67e4 s2: 004a11fc s3: 004a1218
s4: ffffffff s5: 004a1218 s6: 004a11fc s7: 00489a56
t8: 00000001 t9: 00000000 k0: 7f9b65fa k1: 00000000
gp: 778dd090 sp: 7f9b6720 fp: 0047ad14 ra: 7787074b
FA 00000000
PC 00000000
PC Code: <illegal function pointer>
RA 7787074b std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&)+0x2e (/sbin/meshd at 0006474b)
RA Code: 6d01 ea40 653a <1003> 9a61 4b01 da61 6444 e8a0
[bt] *7787074a std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&)+0x2e (/sbin/meshd at 0006471d/0x6474b)
                        Code: 6d01 ea40 653a <1003> 9a61 4b01 da61 6444 e8a0
[bt] *777799c8 [777799a7] <avm::Aicmd::Aicmd(std::shared_ptr<avmipc> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)+0xe>+0x22 (//lib/libmesh_shared.so.0 at 000629a7/0x629c9)
                        Code: 675c f012 9af8 <ef40> 653f 9608 4047 4a05 659e
[bt] *777b6736 [777b6531] <avm::daemon::Mainloop::Mainloop()+0x10>+0x206 (//lib/libmesh_shared.so.0 at 0009f531/0x9f737)
                        Code: 0617 6782 653f <ef40> 6722 9604 659e f011 98b0
[bt] *777b68c8 [777b6895] <avm::daemon::Mainloop::get()+0x10>+0x34 (//lib/libmesh_shared.so.0 at 0009f895/0x9f8c9)
                        Code: f3b0 98a4 9407 <ed40> 653d 9604 659e f651 98b8
[bt] *77861e72 [77861c95] <main+0x10>+0x1de (/sbin/meshd at 00055c95/0x55e73)
                        Code: 1023 f193 994c <ea40> 653a 6782 9604 f272 9948
[bt]  7713e89c __uClibc_main+0x1fc (//lib/libc.so.1 at 000636a0/0x6389c)
                        Code: 27a30018 ac438c20 8f828af4 <0320f809> 8c460000 1000001c 8fbc0010 8f908a00 8e1900c8
[bt] finished.

er13, 2019-06-22:
Hmm, angesichts
```
FREETZ_STDCXXLIB_USE_UCLIBCXX=y
```
(sprich libstdc++ wird von Freetz NICHT ersetzt) würde ich sagen, ist AVM der richtige Ansprechpartner für das Problem. Seit wann tritt das Problem auf? Erst seit 7.1x oder auch schon davor?

Hast Du irgendeine andere Box außer 7490 mit 7.1x drauf? Tritt das Problem auf dieser ebenso auf?

Helmut1972, 2019-06-22:
Ich hab leider keine andere Box und bin von der 6.93 gekommen da die 7 nie funktioniert hatte (kein WLAN). Hab eine mögliche Lösung dafür gefunden ("ifconfig wlan up" in der rc.custom) und daher den Schritt gewagt. In dem Thread mit der Lösung wird der Segfault jedoch auch erwähnt:

https://www.digital-eliteboard.com/threads/7490-freetz-macht-probleme-bei-mesh.476120/

Wenn ich was testen soll mach ich das gerne, ansonsten geht ja wie gescherieben eh alles soweit.

WileC, 2019-06-23:
Also ich muss ja auf meinen 3490 auch mit "ifconfig wlan up" arbeiten, damit WLAN richtig funktioniert, aber den [Segmentation fault] meshd habe ich nicht in meinem Log.

er13, 2019-06-25:
Der Stack-Trace deutet auf einen Fehler in `avm::Aicmd::Aicmd` hin. Ob jetzt der Constructor mit falschen Parametern aufgerufen wird (eher nicht) oder im Constructor selbst was schief läuft, kann ich nicht sagen. Von den von Freetz ersetzten Komponenten ist lediglich `__uClibc_main` involviert und zwar als Aufrufer/Starter von `/sbin/meshd`.

Wenn man nach den letzten beiden (den beiden oberen) Stack-Trace-Einträgen googelt, findet man viele Treffer, wo am Ende alles auf irgendeinen Programmierfehler hinausläuft. S. z.B. [diesen angeblichen gcc-Bug](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=58822). Am Ende ([Comment:26](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=58822#c26) und [Comment:27](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=58822#c27)) hat sich herausgestellt, dass ein Konstrukt verwendet wurde, welches den C++-Standard verletzt und damit UB (undefined behavior) ist.

In jedem Fall hat es mit Freetz nichts zu tun. Freetz macht es nur leichter den Fehler zu sehen. Auf einer unmodifizierten Box müsste der Fehler in der crash.log genauso zu sehen sein - vorausgesetzt die Mesh-Konfiguration ist dieselbe, denn vermutlich tritt der Fehler nicht immer auf, denn sonst hätte AVM ihn gesehen.

closed 2019-06-25 by er13

brandsimon, 2019-11-11:
Ich hatte den Fehler auch, da in dem Link von @Helmut1972 einige schrieben, dass es bei ihnen funktioniert, bin ich dir Sache mal auf meiner Box etwas nachgegangen.
Ich habe meine Box von 6.30 auf 7.12 geflasht. Der Fehler ließ sich bei mir dadurch beheben, die Box danach auf Werkseinstellungen im Fritz Menu zurückzusetzen. Anschließend war der Fehler im Freetz AVM Log verschwunden.
Ich konnte dann auch erfolgreich ein Mesh aufsetzen (7430 als Master, 7490 als Repeater).
Eventuell hilft das ja bei dem ein oder anderem auch.